MarlinFirmware/Marlin #6930: Calibrating my nozzle

jamesjfc123, 2017-06-01:
How can I calibrate my nozzle with marlin firmware.

Roxy-3D, 2017-06-01:
Typically....  the size of the nozzle's hole is a specified number.  Where ever the nozzle came from would tell you that number.    But you can raise the nozzle up 100mm above the bed and extrude 10mm of filament and then measure the thickness of the extruded material.  It won't be super accurate, but you can use a micrometer to decide if it is closer to .3mm or .4mm for example.

jamesjfc123, 2017-06-01:
How would I tell it to extrude 10mm

Maxqia, 2017-06-02:
http://lmgtfy.com/?q=extruder+calibration
:D (sorry couldn't help myself)

Tannoo, 2017-06-02:
Yes. There is no "extruder calibration" in Marlin. Turn to your choice of host sofware for that.

jamesjfc123, 2017-06-02:
Thank you. Any suggestion on what host software to use.

Tannoo, 2017-06-02:
Pronterface, Repetier, Octoprint, Matter Control, Cura... bash prompt, Putty, etc, etc...

Tannoo, 2017-06-02:
Google agrees:

https://www.google.com/search?q=3d+printer+host+software

Tannoo, 2017-06-02:
Not trying to be mean.... but this is not a firmware issue. Please close.

closed 2017-06-02 by jamesjfc123