MarlinFirmware/Marlin #6934: leds and 3d printer

Radiu, 2017-06-02:
hi sorry to make this question sense this isn't a issue
from what i understand we can use case light leds and heatbed leds, i have been trying to find how it works and how to wire up schematics, i see that there is this options on marlin code but i couldn't figure out what it does and how to wire up.
im current using ramps 1.4

thanks

p3p, 2017-06-02:
As you correctly identified this is not the place to ask generic questions about machine setup there are too many variations of printers that use this firmware and issues need to be focused on firmware bugs. saying that, as far as I'm aware you would need to identify a free pin, (PWM compatible for adjustable brightness), attach a Mosfet or relay to power the leds, then control it with M355, for more information you will need to ask this question on an appropriate forum. Please close this issue.

Radiu, 2017-06-02:
i felt why not ask sense firmware provides this options and there is here so many people that have done this and could share info.
 why not have a wiki explaining how works and to how wire up examples, ii have try look into schematics and i couldnt find a source with real details, last thing i wanted to do is to just plug things up with out having a clue how its done proper.
from what i understand so far i would need to have 3 PWM pins atatch to 3 transistors    to control a rgb led strip for the heat bed,
and my guessing is for the case light would need 1 pin sense is like a switch on and off for 1 color

closed 2017-06-02 by Radiu

Bob-the-Kuhn, 2017-06-02:
The best way to identify an available hardware PWM is to look at [this page in the Wiki](http://marlinfw.org/docs/configuration/laser_spindle.html).

M355 and the rgb LEDs don't require a hardware PWM.  If you want just on/off then any available pin will do.