MarlinFirmware/Marlin #6935: Issue : stuttering movement when activating UBL on coreXY with bugfix 1.1.x

joris57, 2017-06-02:
Even with a zero-ed bed (G29 P0) simple movements (G1 X10 in relative mode) display stuttering. When UBL is de-acivated, all is fine again.

oldmcg, 2017-06-03:
I don't know much about coreXY but 10 doesn't seem like it would cross more than one or two mesh cells.  
What kind of feedrate, and does it stutter with lower feedrate?

Roxy-3D, 2017-06-03:
There is one 'stutter' that is known and just something we live with.    If the printer is not moving and you issue a large move that crosses multiple mesh lines, you will get a 'stutter'.    The reason is the mesh software needs to break up the move into smaller moves that align with mesh boundaries in order to do the Z-Correction to track the bed.

The very first segment that gets generated gets sent down to the stepper motors and they start working on it.   But...  when the next segment (crossing the next mesh line) gets sent down the first one is already gone and being worked on.   So there is nothing to append that move to.   And you get a slow down and speed up for the next segment.    Subsequent moves will always have the previous move to append their continuations to and you won't see a stutter.

joris57, 2017-06-03:
This describes indeed more precisely what I am seeing. At least I know now that this is 'as expected' and not a specific defect on coreXY. Thanks for he concise explanation. I'm closing the issue.

closed 2017-06-03 by joris57