MarlinFirmware/Marlin #6936: Mesh Bed Leveling (1.1.2)

rvieirax, 2017-06-02:
Even when calibrating with the printer with manual leveling, the two z-axis motors become immobile when each layer is printed.
Printer: prusa i3

Roxy-3D, 2017-06-02:
I'm guessing from what you write that when you are not printing a layer the two z-motors are mobile.

rvieirax, 2017-06-02:
Dear, thank you for your interest in helping. I will describe the problem:
If the table is not perfectly flat, the z-axis motors must compensate for this irregularity by moving the extruder away from the table so that the extruder-table distance is constant.
Problem: The firmware is configured for Mesh Bed Leveling; The calibration of the table-extruder distance is done following the steps of the LCD; During printing the z-axis motors do not move to compensate for table irregularities.

Roxy-3D, 2017-06-02:
Did you activate the system?    Did you  activate the: MIN_SOFTWARE_ENDSTOPS ?
```cpp
// If enabled, axes won't move below MIN_POS in response to movement commands.
#define MIN_SOFTWARE_ENDSTOPS
// If enabled, axes won't move above MAX_POS in response to movement commands.
#define MAX_SOFTWARE_ENDSTOPS
```
That will keep the nozzle from going below 0.000 to track the bed if it dips.

In order to use the original mesh leveling, the machine needs to be homed, and then a G29 performed.
Did you get the 
```cpp
          SERIAL_PROTOCOLLNPGM("Mesh probing done.");
          BUZZ(100, 659);
          BUZZ(100, 698);
          mesh_probing_done();
```
message at the end of the probing?  If not...  The mesh will not be active.

rvieirax, 2017-06-02:
The endstops are enabled and I receive the following message after manual calibration:
"leveling done"

`//===========================================================================
//=============================== Bed Leveling ==============================
//===========================================================================
// @section bedlevel

/**
 * Choose one of the options below to enable G29 Bed Leveling. The parameters
 * and behavior of G29 will change depending on your selection.
 *
 *  If using a Probe for Z Homing, enable Z_SAFE_HOMING also!
 *
 * - AUTO_BED_LEVELING_3POINT
 *   Probe 3 arbitrary points on the bed (that aren't collinear)
 *   You specify the XY coordinates of all 3 points.
 *   The result is a single tilted plane. Best for a flat bed.
 *
 * - AUTO_BED_LEVELING_LINEAR
 *   Probe several points in a grid.
 *   You specify the rectangle and the density of sample points.
 *   The result is a single tilted plane. Best for a flat bed.
 *
 * - AUTO_BED_LEVELING_BILINEAR
 *   Probe several points in a grid.
 *   You specify the rectangle and the density of sample points.
 *   The result is a mesh, best for large or uneven beds.
 *
 * - AUTO_BED_LEVELING_UBL (Unified Bed Leveling)
 *   A comprehensive bed leveling system combining the features and benefits
 *   of other systems. UBL also includes integrated Mesh Generation, Mesh
 *   Validation and Mesh Editing systems. Currently, UBL is only checked out
 *   for Cartesian Printers. That said, it was primarily designed to correct
 *   poor quality Delta Printers. If you feel adventurous and have a Delta,
 *   please post an issue if something doesn't work correctly. Initially,
 *   you will need to set a reduced bed size so you have a rectangular area
 *   to test on.
 *
 * - MESH_BED_LEVELING
 *   Probe a grid manually
 *   The result is a mesh, suitable for large or uneven beds. (See BILINEAR.)
 *   For machines without a probe, Mesh Bed Leveling provides a method to perform
 *   leveling in steps so you can manually adjust the Z height at each grid-point.
 *   With an LCD controller the process is guided step-by-step.
 */
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
#define MESH_BED_LEVELING

/**
 * Enable detailed logging of G28, G29, M48, etc.
 * Turn on with the command 'M111 S32'.
 * NOTE: Requires a lot of PROGMEM!
 */
//#define DEBUG_LEVELING_FEATURE

#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_BILINEAR) || ENABLED(AUTO_BED_LEVELING_UBL)
  // Gradually reduce leveling correction until a set height is reached,
  // at which point movement will be level to the machine's XY plane.
  // The height can be set with M420 Z<height>
  #define ENABLE_LEVELING_FADE_HEIGHT
#endif

#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)

  // Set the number of grid points per dimension.
  #define GRID_MAX_POINTS_X 3
[... 97 lines cut ...]
// Manually set the home position. Leave these undefined for automatic settings.
// For DELTA this is the top-center of the Cartesian print volume.
//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
//#define MANUAL_Z_HOME_POS 0

// Use "Z Safe Homing" to avoid homing with a Z probe outside the bed area.
//
// With this feature enabled:
//
// - Allow Z homing only after X and Y homing AND stepper drivers still enabled.
// - If stepper drivers time out, it will need X and Y homing again before Z homing.
// - Move the Z probe (or nozzle) to a defined XY point before Z Homing when homing all axes (G28).
// - Prevent Z homing when the Z probe is outside bed area.
//#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    // Y point for Z homing when homing all axis (G28).
#endif

// Homing speeds (mm/m)
#define HOMING_FEEDRATE_XY (50*60)
#define HOMING_FEEDRATE_Z  (4*60)
`

rvieirax, 2017-06-02:
The EndStops are enabled.
After manual table calibration the printer beeps and the message **"Leveling Done!"**

The code:

```
//===========================================================================
//=============================== Bed Leveling ==============================
//===========================================================================
// @section bedlevel

/**
 * Choose one of the options below to enable G29 Bed Leveling. The parameters
 * and behavior of G29 will change depending on your selection.
 *
 *  If using a Probe for Z Homing, enable Z_SAFE_HOMING also!
 *
 * - AUTO_BED_LEVELING_3POINT
 *   Probe 3 arbitrary points on the bed (that aren't collinear)
 *   You specify the XY coordinates of all 3 points.
 *   The result is a single tilted plane. Best for a flat bed.
 *
 * - AUTO_BED_LEVELING_LINEAR
 *   Probe several points in a grid.
 *   You specify the rectangle and the density of sample points.
 *   The result is a single tilted plane. Best for a flat bed.
 *
 * - AUTO_BED_LEVELING_BILINEAR
 *   Probe several points in a grid.
 *   You specify the rectangle and the density of sample points.
 *   The result is a mesh, best for large or uneven beds.
 *
 * - AUTO_BED_LEVELING_UBL (Unified Bed Leveling)
 *   A comprehensive bed leveling system combining the features and benefits
 *   of other systems. UBL also includes integrated Mesh Generation, Mesh
 *   Validation and Mesh Editing systems. Currently, UBL is only checked out
 *   for Cartesian Printers. That said, it was primarily designed to correct
 *   poor quality Delta Printers. If you feel adventurous and have a Delta,
 *   please post an issue if something doesn't work correctly. Initially,
 *   you will need to set a reduced bed size so you have a rectangular area
 *   to test on.
 *
 * - MESH_BED_LEVELING
 *   Probe a grid manually
 *   The result is a mesh, suitable for large or uneven beds. (See BILINEAR.)
 *   For machines without a probe, Mesh Bed Leveling provides a method to perform
 *   leveling in steps so you can manually adjust the Z height at each grid-point.
 *   With an LCD controller the process is guided step-by-step.
 */
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
#define MESH_BED_LEVELING

/**
 * Enable detailed logging of G28, G29, M48, etc.
 * Turn on with the command 'M111 S32'.
 * NOTE: Requires a lot of PROGMEM!
 */
//#define DEBUG_LEVELING_FEATURE

#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_BILINEAR) || ENABLED(AUTO_BED_LEVELING_UBL)
  // Gradually reduce leveling correction until a set height is reached,
  // at which point movement will be level to the machine's XY plane.
  // The height can be set with M420 Z<height>
  #define ENABLE_LEVELING_FADE_HEIGHT
#endif

#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)
[... 101 lines cut ...]
// For DELTA this is the top-center of the Cartesian print volume.
//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
//#define MANUAL_Z_HOME_POS 0

// Use "Z Safe Homing" to avoid homing with a Z probe outside the bed area.
//
// With this feature enabled:
//
// - Allow Z homing only after X and Y homing AND stepper drivers still enabled.
// - If stepper drivers time out, it will need X and Y homing again before Z homing.
// - Move the Z probe (or nozzle) to a defined XY point before Z Homing when homing all axes (G28).
// - Prevent Z homing when the Z probe is outside bed area.
//#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    // Y point for Z homing when homing all axis (G28).
#endif

// Homing speeds (mm/m)
#define HOMING_FEEDRATE_XY (50*60)
#define HOMING_FEEDRATE_Z  (4*60)

```

gapx15, 2017-06-04:
Hello, any news. I have exactly the same problem. Mesh is saved (and G29 always return right matrix) #define MIN_SOFTWARE_ENDSTOPS are disabled, but z motors are not moving. Thanks for help in advance.

My configuration file:
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/1050564/Configuration.h.zip)

BlackLightUK, 2017-06-05:
`M420 V1` should print out the state of the mesh that's currently in use. If it's all zeroes, then you didn't store the mesh to EEPROM before you started printing. 

If the mesh has all the offsets listed, then you will probably need an `M420 S1` in your startup gcode (which turns on the bed levelling), but put this after a `G28` (home XYZ) as a home will turn off the levelling routine.

rvieirax, 2017-06-06:
I inserted the M420 S1 code and the leveling function is working fine.
Thanks for the help.

closed 2017-06-06 by rvieirax

carlonb, 2017-06-06:
Yes, in this way it's working but why in previous release (i do not remember wich one) was not necessary manually put M420 S1 in the object g code?
I think enable the BED leveling automatically when we launch the print without add the g code M420 S1 is simpler.

fiveangle, 2017-06-08:
@carlonb - it needs to be enabled at least once to get confirmation from user that they calibrated mesh.  The state of bed levelling is stored in EEPROM.  This has always been the case as far as I know.

Roxy-3D, 2017-06-08:
> I think enable the BED leveling automatically when we launch the print without add the g code M420 S1 is simpler.

Historically... G29 always is disabled by a G28.   And there is controversy about the fact that G28 preserves the Active/Disabled state of UBL.   But with UBL, you can do a M500 and save the state as 'Active', and never even put a G29 in your Slic3r's Startup GCode.   That is how I operate.

carlonb, 2017-06-09:
@Roxy-3D  - Thanks for the clarify, can you give me an hint where, in the code, the G28 received command from SD card will disable the G29 Mesh/UBL ? I want to try to mod by myself the code to experimenting and avoid to add  M420 S1 to the startup gcode of  slic3er.
Or better the diff with one of previous code where G28 do not deactivate the MESH enable.
Bye
carlo

BlackLightUK, 2017-06-09:
All that springs to mind here is that I really need to invest in a decent bed sensor...... G28-and-go sounds really quite appealing at this point in time :)

Roxy-3D, 2017-06-09:
> Thanks for the clarify, can you give me an hint where, in the code, the G28 received command from SD card will disable the G29 Mesh/UBL ?

It doesn't matter if it is from SD-Memory card or from a serial command.  G28 disables leveling here: https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Marlin_main.cpp#L3726