MarlinFirmware/Marlin #6937: Code to reset time?

mperdue, 2017-06-02:
Is there a code that can be sent to the printer to reset the time on the LCD to zero? It seems that when I have to about a print and restart it the elapsed time display continues from where it was. If there was a reset code I could add it to my start-up script and not have to mess with it any more.

p3p, 2017-06-02:
I don't think there is a reset specifically, but M77 should stop the timer, and when started again it is reset

mperdue, 2017-06-02:
I just use the stop button in Simplify 3D so I'm not really sure what it does.

p3p, 2017-06-02:
Ah, I jumped to the conclusion you were printing from SD and stopping the print via lcd, pretty much everything is up to the host application when printing over serial, I'm not familiar with Simplify 3D they may not be sending an M77 when you click stop.

mperdue, 2017-06-02:
I'll figure something out.

closed 2017-06-03 by mperdue