MarlinFirmware/Marlin #6944: BABYSTEP_ZPROBE_OFFSET Didn't do Babystepping Properly

sarf2k4, 2017-06-03:
Running the bugfix 1.1.x downloaded on 30th May, I decided to enable the `BABYSTEP_ZPROBE_OFFSET` and test the printer today. Apparently the printer didn't adjust the z axis accordingly and more likely only adjust the z-offset. 

So I had to restart the print to see whether or not the z-offset has been adjusted properly.

I do know that this option is to integrate the babystepping z with m851 but there were no babystepping were done in the process except adjusting the z-offset, got to restart the print process to see the changes.

Was this intentional to change only z-offset instead of babystepping and offset at the same time?
If I enable `BABYSTEP_ZPROBE_OFFSET`, do I have to use "store memory" under control section?

tcm0116, 2017-06-04:
Had you recently executed a G29? The Z axis will only be babystepped when adjusting the Z Offset if the machine has been leveled and the motors have not timed out (note that homing after leveling resets the leveling).

sarf2k4, 2017-06-04:
I usually do babystepping during printing, i always have g28 followed by
g29, after both heaters reached theor target. Once the machine starts, i
check on the first layer if i need to use babystepping or not

dot-bob, 2017-06-06:
You by chance aren't using UBL?  If so see #6488

RalphSch, 2017-07-25:
I also got the issue with not working z-baby-stepping when using bilinear abl and activated BABYSTEP_ZPROBE_OFFSET. I can babystep as much as I want but the nozzle stays at the same height. Using UBL to avoid this behavior is not an option ;)

closed 2017-11-01 by thinkyhead

S1CAR1US, 2018-02-19:
why is this closed?  what is the solution?  I have same issue.

thinkyhead, 2018-02-19:
@S1CAR1US — Please test `bugfix-1.1.x` and file a new report if you have this problem.