MarlinFirmware/Marlin #6945: Shifting in the y axis

jamesjfc123, 2017-06-03:
My printers are shifting significantly in the y axis. Is this a slicer or firmware issue.

Bob-the-Kuhn, 2017-06-03:
Can you be more specific on the problem description?  

Are you loosing steps (Y keeps shifting slowly) or is the print at a location you're not expecting it to be?

If you're loosing steps the first thing to do is crank the stepper current way up to be 100% sure this isn't part of the problem.  Probably not the issue since this is occurring on multiple printers.

Next try lowing these settings:
- DEFAULT_MAX_FEEDRATE
- DEFAULT_MAX_ACCELERATION
- DEFAULT_ACCELERATION
- DEFAULT_TRAVEL_ACCELERATION
- DEFAULT_YJERK

jamesjfc123, 2017-06-03:
The print just shifts at one point during the print. It is not gradual.

Bob-the-Kuhn, 2017-06-03:
I'm coming up with nothing obvious.  I'll start going through the usual problem isolation questions.

The "take two aspirin and call me in the morning" response is to ask you to try the latest [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x). That resolves a fair number of problems.

I remember something about a similar issue a couple months ago but I can't find the thread to see how it was resolved.

Multiple printers implies a software issue.  

How big is the shift?

Does it happen at the same height on all the printers?

Does it happen with multiple objects?

Have you tried a different slicer?

Could the cables to the print head be binding/pulling after a certain height.  Not likely with multiple printers.

jamesjfc123, 2017-06-04:
It was an issue with the acceleration and skipping. Thank you.

closed 2017-06-04 by jamesjfc123