MarlinFirmware/Marlin #6949: E0_AUTO_FAN_PIN

Bergerac56, 2017-06-03:
This is not an issue but a question. 
I have a second "Extruder fan" activated automatically (E0_AUTO_FAN_PIN in section extruder of configuration_adv.h). I would like to be able to activate/desactivate it via a M command. Is it feasible?

Thanks

Bob-the-Kuhn, 2017-06-04:
If I read the code correctly you need:
- FAN1_PIN defined 
- Set E0_AUTO_FAN_PIN to the same pin.  Use this to do it  `#define E0_AUTO_FAN_PIN   FAN1_PIN `
- To control it use M105/M106 P1 Sxx where xx is the fan speed 0-255

Bergerac56, 2017-06-04:
Thanks Bob.

But I could not get it working.
The additionnal fan is supposed to improve the cooling of the point just extruded. I want to use it for PLA but not for ABS (for ex.). I am looking for a "soft" way to toggle it. My logical guess was to use the extruder cooling fan section (Fan activated by temperature like on another printer I have) and to be able to command it by a magic M command... 

When I use only the section "extruder cooling fan", the fan is activated by the temperature of the nozzle (with the correct pin defined for my printer: D11). But I cannot "command" it by soft.

When I follow your suggestion (see part of the code below, as far as I understood correctly ;) ), this fan starts when a stepper motor is activated. But not when the temperature exceeds the AUTO_FAN_TEMPERATURE. M106 does not have any effect on this specific fan. It stops only after all motors are stopped and the delay is over (for test purpose, set to 1 second).
M106 does not seem to take the P parameter in account. I can only command the extruder original fan (M106 P0 S255 or M106 S255 for ex.). When I use P1, nothing happens (and also no impact on extruder fan). (M105 you suggested is to get temperatures, am I right?. M107 does not stop the second fan)

I certainly did something the wrong way... but what ?

Here is the small section adapted in configuration_adv.h. I did not modify other files for this purpose (pins,...):
/**
 * Controller Fan
 * To cool down the stepper drivers and MOSFETs.
 *
 * The fan will turn on automatically whenever any stepper is enabled
 * and turn off after a set period after all steppers are turned off.
 */
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN 11         // Set a custom pin for the controller fan. Port D11 Jacbot
  #define CONTROLLERFAN_SECS 1          // Duration in seconds for the fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 255       // 255 == full speed
#endif

// When first starting the main fan, run it at full speed for the
// given number of milliseconds.  This gets the fan spinning reliably
// before setting a PWM value. (Does not work with software PWM for fan on Sanguinololu)
//#define FAN_KICKSTART_TIME 100

// This defines the minimal speed for the main fan, run in PWM mode
// to enable uncomment and set minimal PWM speed for reliable running (1-255)
// if fan speed is [1 - (FAN_MIN_PWM-1)] it is set to FAN_MIN_PWM
//#define FAN_MIN_PWM 50

// @section extruder

/**
 * Extruder cooling fans
 *
 * Extruder auto fans automatically turn on when their extruders'
 * temperatures go above EXTRUDER_AUTO_FAN_TEMPERATURE.
 *
 * Your board's pins file specifies the recommended pins. Override those here
 * or set to -1 to disable completely.
 *
 * Multiple extruders can be assigned to the same pin in which case
 * the fan will turn on when any selected extruder is above the threshold.
 */
#define E0_AUTO_FAN_PIN FAN1_PIN // Ventilateur supplémentaire Jacbot PIN 11
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed

dbian, 2017-06-04:
Firstly, define your second FAN PIN id in your pins_xxx.h,
[image]
Then add some codes to M106&M107 like below
[image]

Bergerac56, 2017-06-05:
Hello dbian. 

Thanks for your help. But... it does not work ...
I tried different things and finally I defined the pin in "pins_RAMPS.h" this way:
// Heaters / Fans
//
#ifndef MOSFET_D_PIN
  #define MOSFET_D_PIN  -1
#endif
#ifndef RAMPS_D11_PIN        \\ specific for my printer
  #define RAMPS_D11_PIN 11
#endif
#ifndef RAMPS_D8_PIN
  #define RAMPS_D8_PIN   8
#endif
#ifndef RAMPS_D9_PIN
  #define RAMPS_D9_PIN   9
#endif
#ifndef RAMPS_D10_PIN
  #define RAMPS_D10_PIN 10
#endif

#define HEATER_0_PIN     RAMPS_D10_PIN

#if ENABLED(IS_RAMPS_EFB)                      // Hotend, Fan, Bed
  #define FAN_PIN        RAMPS_D9_PIN
  #define HEATER_BED_PIN RAMPS_D8_PIN
  #define FAN1_PIN       RAMPS_D11_PIN    // Specific for my printer

By doing that, the options "automatic cooling fan" or "use controler fan" functions correctly (but not the 2 together as they cannot share the same pin). But no way to control this fan through m106.

Modifications to marlin_main.cpp is as follow:
  inline void gcode_M106() {
    uint16_t s = parser.seen('S') ? parser.value_ushort() : 255,
                 p = parser.seen('P') ? parser.value_ushort() : 0;
    NOMORE(s, 255);
    if (p < FAN_COUNT) fanSpeeds[p] = s;
    if (1 < FAN_COUNT) fanSpeeds[1] = s;
  }

  /**
   * M107: Fan Off
   */
  inline void gcode_M107() {
    uint16_t p = parser.seen('P') ? parser.value_ushort() : 0;
      if (p < FAN_COUNT) fanSpeeds[p] = 0;
      if (1 < FAN_COUNT) fanSpeeds[1] = 0;
  }

I am not sure I understand correctly what those 2 added instructions are doing but I followed your suggestion.
The original fan is still working as expected, but I am unable to control the second one.

Bergerac56, 2017-06-05:
Finally I got it :)

Just replacing the "1" by a "2" in the 2 modified lines of marlin_main.cpp:
if (1 < FAN_COUNT) fanSpeeds[1] = s; by if (2 < FAN_COUNT) fanSpeeds[2] = s;

All is as expected. Thanks abain a lot dbian.

closed 2017-06-07 by Bergerac56