MarlinFirmware/Marlin #6955: font outside of screen when change font from SSD1306, SH1106 ?

fred2088, 2017-06-05:
on 0.96" and 1.3" OLED Screen

#define FONT_MENU_NAME u8g_font_7x13  
 
   the font from all the following function will out of screen ....

    void lcd_info_board_menu();
    void lcd_info_printer_menu();
    void lcd_advanced_pause_toocold_menu();
    void lcd_advanced_pause_option_menu();
    void lcd_advanced_pause_init_message();
    void lcd_advanced_pause_unload_message();
    void lcd_advanced_pause_insert_message();
    void lcd_advanced_pause_load_message();
    void lcd_advanced_pause_heat_nozzle();
    void lcd_advanced_pause_extrude_message();
    void lcd_advanced_pause_resume_message();
--------------------------------------------------------------
    setfont() are added between START_MENU() and END_MENU()

      START_MENU();
      lcd_setFont(FONT_STATUSMENU);
      .....
      lcd_setFont(FONT_MENU);
      END_MENU();

is possible to update into the marlin ?

Thanks

thinkyhead, 2018-02-07:
Is this still an issue in the current `bugfix-1.1.x`? I'm guessing yes.

boelle, 2019-02-18:
@fred2088 s this still an issue in the current `bugfix-2.0`?

closed 2019-02-19 by fred2088