MarlinFirmware/Marlin #6960: Marlin bugfix-1.1.x hangs with LIN_ADVANCE enabled and large-ish K values

joris57, 2017-06-05:
With a K=300 or larger following gcode script makes Marlin 'hang' (LCD is still responsive but no more code gets executed). The extrusion in below code does not `occur.

```
M190 S45 ; wait for bed temp
M109 S190 ; wait for extruder temp
G1 X50 Y50 Z0.35 F3000.0
G91
G1 X60.0 E9.0  F1000.0 ; intro line 1
G1 X40.0 E8.0  F1000.0 ; intro line 2 was E12.5
G90
```

psavva, 2017-06-17:
Please check https://github.com/MarlinFirmware/Marlin/issues/5699#issuecomment-309223303

thinkyhead, 2017-06-19:
@psavva Are you saying a lower K value is required, and that's all that can be done?

psavva, 2017-06-19:
Issue is caused with too high K value, too high microstepping, and finally causes the freeze.  If you check the last comments on 5699 which also has a workaround for this. 

I still think a permanent fix is needed.

Lowering K value together with dropping Microstepping should resolve it (workaround)

nallar, 2017-10-02:
This was probably the same issue as #5699, if so it can be closed.

joris57, 2017-10-06:
Not inclined to do further testing at the moment, so closing as requested.

closed 2017-10-06 by joris57