MarlinFirmware/Marlin #6961: Usage Question: How to use LCD only for manual UBL mesh creation?

WheresWaldo, 2017-06-05:
I think this question is directed at @Tannoo since the UBL menu is his creation. So I got UBL up and running as as I suspected if I auto-create the mesh I have to go back and tweak about 80 points of the 100. For me it is not worth the time and effort to figure out how to move the head to the correct spot and then edit the information for each of those points. So I am trying to figure out how to use the "business card" method to create the entire mesh from scratch. To say the UBL menu system is a bit less than intuitive is being kind. No disrespect to all the work you did, I wouldn't have gotten as far as I have without the UBL menu system. Is there a way to outline from start to finish what buttons must be pressed in what order to go from no UBL mesh to a complete and enabled UBL mesh, using just the LCD and manually measuring each point?

Sorry for being insistent on the user experience, but no matter how much better UBL is than MESH, one thing that can be said about Ed Patel's implementation is that it was easy to do, with very few steps. So at least, at this stage since better isn't really easy, it becomes a barrier to implement.

Tannoo, 2017-06-06:
Okay, I'm not the UBL expert. This is what I was given and went with that.
>Just do the commands
G28
G29 P1 O
G29 P3 R C .5
G26 P C O3.5
G29 P4 R O To edit....

That sequence will get you very close to a perfect mesh.
You just repeat the last two commands and you will have everything perfect.

1) G29 P1 O -- Generate the mesh.
`Prepare - Unified Bed Leveling - Tools - Build Mesh -`

`Build PLA Mesh`
or
`Build ABS Mesh`
or
`Build Custom Mesh`
or
`Build Cold Mesh`

2) G29 P3 R C .5 -- Fill-in the spots the probe couldn't reach.
`Prepare - Unified Bed Leveling - Tools - Build Mesh - Fill-in Mesh -`

`Fill-in Mesh`
or
`Smart Fill-in`
or
`Manual Fill-in`

Note: `Smart Fill-in` is recommended.

3) G26 P C O3.5 -- Validate Mesh
`Prepare - Unified Bed Leveling - Tools - Validate Mesh -`

`Validate PLA Mesh`
or
`Validate ABS Mesh`
or
`Validate Custom Mesh` <-- It will be "preset" to the temps used in `Build Custom Mesh`.

4) G29 P4 R O To edit....
`Prepare - Unified Bed Leveling - Tools - Edit Mesh - Fine Tune All`

Tannoo, 2017-06-06:
I was thinking of a `Step-by-Step` mesh generation based on those placed here:
`Prepare - Unified Bed Leveling - Step-by-Step -`
`1 - Generate Mesh`
`2 - Fill-in Mesh`
`3 - Validate Mesh`
`4 - Edit Mesh`

Thoughts?

WheresWaldo, 2017-06-06:
If I read the instructions above correctly, UBL needs an auto-generated mesh before you can do a Fill-In Mesh. And, I am not sure why I would validate the mesh if I am just going to fine tune the whole mesh.

The results of the above sequence are: Selecting Fine Tune All just pauses for a minute then drops you back at the Info screen with the status message Doing G29 UBL! but nothing actually happens. There is no hotend movement, no temps increasing, no up and down movement. Turning the encoder does nothing. Pressing the encoder exits the Doing G29 UBL! but nothing changed. 

It appears that within UBL currently there is no equivalent functionality to the old MESH bed leveling or if the functionality is there it is broken. This is what should be considered a show stopping bug.

Tannoo, 2017-06-06:
Open a serial monitor and then `Prepare - Unified Bed Leveling - Output Mesh Map - Output for Host`

Post here the output of that.

Tannoo, 2017-06-06:
Validating the mesh, prints a pattern to see what needs to be adjusted.
Without that, you are just editing what you think needs editing.

Tannoo, 2017-06-06:
I am working on a way to see the mesh point values on the lcd and then edit that point (and some points around it).

Tannoo, 2017-06-06:
>If I read the instructions above correctly, UBL needs an auto-generated mesh before you can do a Fill-In Mesh. And, I am not sure why I would validate the mesh if I am just going to fine tune the whole mesh.

Well, you asked:
> Is there a way to outline from start to finish what buttons must be pressed in what order to go from no UBL mesh to a complete and enabled UBL mesh, using just the LCD and manually measuring each point?

So, I did that.

Roxy-3D, 2017-06-06:
@WheresWaldo    Tannoo and I are exploring ideas to make the user interface very intuitive from the LCD.   We don't have all the answers.   But just as a side effect of exploring the ideas discussed in https://github.com/Tannoo/Marlin/issues/4#issuecomment-306090327   this fell out: 
 https://github.com/MarlinFirmware/Marlin/pull/6964 to solve some of the user intuitive issues from the LCD Panel interface.     And now Thinky is wondering if he can get another 1KB out of RAM here: https://github.com/MarlinFirmware/Marlin/pull/6965#issue-233732803  I might be side tracked for a few weeks.

A while ago you wanted Z-Movement of the probe while editing mesh points.  I put it back in the code with this comment:
```cpp
 #define UBL_MESH_EDIT_MOVES_Z     // Sophisticated users prefer no movement of nozzle
```
Tannoo and I are mostly focused on getting the foundation pieces in place.  In 6 weeks, everything will be polished.   

Here is something that would be helpful and can be implemented in a couple of days:    Find three or four command strings of various G29 commands that build a perfect mesh and works for your users.  We can give you a 'Quick Start' menu sequence that does everything you need.   One set of enqueued commands after another....   

If you can put 3 or 4 sequences of commands together to build a perfect mesh, we can make it so the user just clicks through them.

But right now....  until we get Tannoo's stuff working on 20x4 displays...  We aren't ready to polish things.

WheresWaldo, 2017-06-06:
As I mentioned I am not intentionally trying to be dense, it is just hard to justify a wholesale change that removes an easy to use function. I am trying Tannoo's method again, and it simply doesn't work, When you get to the Fill In All it stops and does nothing. The instructions don't work so really was my question answered?

If this is something that will be looked at in the near future, then I will tell all my users to stick with 1.1.1 and MESH until it is done. I know you are working on this and I know that it is near and dear to you. I also know that you are working to get all the basic functions working in a predictable manner. Just please do not forget that 3D printing is no longer solely used by enthusiasts who actually know what a g-code is.

Also note that if this was only a request for myself, I wouldn't care as much about simplicity, I am fully capable of creating a mesh, validating a mesh and editing a mesh from the serial terminal. What I cannot state with any degree of certainty is whether or not any of the potential users are capable of the same.

Tannoo, 2017-06-06:
>Just please do not forget that 3D printing is no longer solely used by enthusiasts who actually know what a g-code is.

That is exactly why I created the UBL menu system.

Tannoo, 2017-06-06:
There have been some code changes and the UBL menu doesn't always get updated with it. I will have to look at it again and see if everything is still kosher.
Working on other things keeps me from maintaining merged code all the time.

Roxy-3D, 2017-06-06:
@WheresWaldo    Please read (and spend a few minutes improving) Bob's UBL guide here:  https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md

But very early...  In the first page of that document is a quick start guide of commands that are all available from Tannoo's menu system....   How about you try going through those steps using the LCD Interface and let us know if some step is failing.

And we will even put what you want on a 'Quick Start' LCD Menu Page.   But...   right now we are trying to get the foundation in place.

Tannoo, 2017-06-06:
>When you get to the Fill In All it stops and does nothing.
It does do something. It fills in the remaining 0.000's with something other than 0's. It doesn't need to move anything to do that.

Tannoo, 2017-06-06:
I am now creating a new mesh using the new Step-by-Step. I will see if there is something broken or not.

WheresWaldo, 2017-06-06:
@Tannoo There appears to be a miscommunication between what I am asking about and what you are hearing I am asking about. I don't want to edit the mesh, I want to MANUALLY create the mesh. Whether the mesh has values (as shown in Prepare - Unified Bed Leveling - Output Mesh Map - Output for Host) or are all zeros, when you get to the step of Fine Tune All it stops working. 

I will see what I can do about looking over the wiki page and ideas about a Quick Start Menu later on in the week,

Tannoo, 2017-06-06:
Okay, I have never created a full mesh that way. With a probe, but not auto-generating any of it?

WheresWaldo, 2017-06-06:
@Tannoo Thank you for being so patient with me. That is what I meant by duplicating the MESH method using UBL.

Tannoo, 2017-06-06:
Why do UBL then? Why not MESH_BED_LEVELING?

I will see about creating it like you said and see where the issue might be.

Because in my playing with the LCD map for UBL, I am able to edit points even with everything being 0's.
Roxy has had issues with editing anything that way, I don't know why because mine works fine.

WheresWaldo, 2017-06-06:
Why use UBL over MESH, because everyone is telling me that MESH is deprecated!

Tannoo, 2017-06-06:
This is what I got when trying to edit the points with the entire mesh being all 0's:
```
echo:enqueueing "G29 P4 R T"
echo:Home XYZ first
echo:busy: processing
X:0.00 Y:0.00 Z:6.05 E:0.00 Count X:0 Y:0 Z:23232
?(R)epetition count invalid (1+).
```

WheresWaldo, 2017-06-06:
You get the same if the mesh actually has numbers in it.

Roxy-3D, 2017-06-06:
People...   I'm not sure if there are compile and link problems if you do this...   But there is no reason you can't do a G29 P0 and follow it with a G26.    You will know (without even probing a single location) where you need to edit the mesh...

You should be able (with one extra iteration) to bring up UBL with a perfect mesh even if you probe nothing.

Don't even bother manual probing the bed.  Just do a G26 and **_SEE_** where you need to edit things.

Tannoo, 2017-06-06:
Okay, I remember somewhere that the R needs to be R999 to do all points. I will change that.
That works for me.

WheresWaldo, 2017-06-06:
@Roxy-3D I understand what you are saying, but question whether that is actually easier, feel can be much more accurate that sight especially for people like me as old as dirt.. I don't think we are going to solve this tonight, maybe once you are further along with code optimization and core functionality we can revisit this and figure out a way to make this work

Tannoo, 2017-06-06:
So, a manual input of `G29 P4 R999 T` will work.

Can also change the edit menu item in `ultralcd.cpp`:
from
`MENU_ITEM(gcode, MSG_UBL_FINE_TUNE_ALL, PSTR("G29 P4 R T"));`
to
`MENU_ITEM(gcode, MSG_UBL_FINE_TUNE_ALL, PSTR("G29 P4 R999 T"));`

Tannoo, 2017-06-06:
I did this and it is now starting to edit points with no issues.

WheresWaldo, 2017-06-06:
Just curious @Tannoo did you ever use MESH yourself, or have you mostly used auto-leveling then moved on to UBL? If you have used MESH then you should recall how the mesh was created. 
I can't test this fix at the moment since my printer is loud and everyone here but me is sleeping.

Roxy-3D, 2017-06-06:
> I understand what you are saying, but question whether that is actually easier, feel can be much more accurate that sight especially for people like me as old as dirt.. 

Yeah...  I get it...  I got distracted today because I saw a way to save 100 bytes of RAM.   But I did say:  https://github.com/Tannoo/Marlin/issues/4#issuecomment-306111408

> Probably figuring out how to get the nozzle to track the edit location nicely is more important right now.

We are trying hard to get the big pieces in place...

WheresWaldo, 2017-06-06:
As I suggested, we can hash this out when the core work is done. I just wanted to make sure that the both of you understood the request and the necessity for it.

Tannoo, 2017-06-06:
I used MESH back when auto bed leveling didn't work for me.
I would try the auto bed leveling every so often to see if it got fixed and one day it did. So, I started using that. It was WAAAY easier and automated.

Then UBL came about. Soooo many options. I couldn't remember what to use to make the mesh work. So, I made a menu system to get my "cheat sheet" in the menu.

WheresWaldo, 2017-06-06:
Agree, Automated is the way it should be used. however I am stuck with a poor implementation of the endstop probe, that is inconsistent across the bed, It is close, but not close enough, so auto-leveling of any kind ends up with less than stellar results. MESH was the answer, since you knew by feel if you nozzle was touching the bed. I know this functionality is stuck in UBL somewhere, I just asked if it can be bubbled up to the top.

Tannoo, 2017-06-06:
I still have a few small issues with UBL that are hard to explain, and one day it will be sorted out, or I will understand what is happening.

Tannoo, 2017-06-06:
>I am stuck with a poor implementation of the endstop probe, that is inconsistent across the bed, It is close, but not close enough

That needs to be addressed. No need to be "stuck with it".

Roxy-3D, 2017-06-06:
And my history is I added some much needed options to the early Auto Bed Leveling:  http://3dprintboard.com/showthread.php?3105-Auto_Bed_Leveling-Enhanced-G29-command   That thread is 3 years old.   And with one exception...  all of the extra options there made it into the main branch of Marlin.   

Now...  I'm trying to get the best features of all of the various bed leveling systems into the same system and working together and I'm getting top notch help on everything from the LCD display to making the system work on Delta's.   

> >I am stuck with a poor implementation of the endstop probe, that is inconsistent across the bed, It is close, but not close enough
> That needs to be addressed. No need to be "stuck with it".

Even without fixing that bed slop.... You can do a G29 P0 and do a G26 and start editing your mesh.

Please see if that will work for your users.

WheresWaldo, 2017-06-06:
@Roxy-3D I have asked in the past about editing mesh points, and as a courtesy I will ask again, I will also try it with no mesh enabled to see if it is workable, I wonder is simply doing a G29 P0 followed by a G29 P4 R999 T would do exactly what I am expecting. That would save the time of printing the validation grid.

WheresWaldo, 2017-06-06:
@Tannoo beleive me I am not STUCK with any part of my printers. But when you are dealing with users, some of whom had no business buying a 3D Printer in the first place, you can't expect them to "upgrade" anything.

Roxy-3D, 2017-06-06:
You need to go up to a higher level...   You want to end up with a perfect mesh that describes your print surface.   How you get there doesn't matter.

If you have a good machine...  One that with not auto bed leveling correction can print a calibration cube in the center....

Then...  Doing a G29 P0 followed with a G26 mesh validation pattern will let you in 2 or 3 iterations have a perfect mesh.

If you have a horrible machine with a tilted bed with lots of high and low spots...   You might be better off leveling the bed as much as you can and then doing a G29 P1 followed by a G26 G29 P4 editing cycle.

> But when you are dealing with users, some of whom had no business buying a 3D Printer in the first place, you can't expect them to "upgrade" anything.

Well, hopefully they are willing to try the steps in the Bob's "Quick Start Guide" :   

> 
> The following command sequence can be used to home, level, and then fine-tune the results:
> 
> M502          ; Reset settings to configuration defaults...
> M500          ; ...and Save to EEPROM. Use this on a new install.
> 
> M190 S65      ; Not required, but having the printer at temperature helps accuracy
> M104 S210     ; Not required, but having the printer at temperature helps accuracy
> 
> G28           ; Home XYZ.
> G29 P1        ; Do automated probing of the bed.
> G29 P2 B T    ; Do manual probing of unprobed points. Requires LCD.
> G29 P3 T      ; Repeat until all mesh points are filled in.
> 
> G29 T         ; View the Z compensation values.
> M420 S1       ; Activate leveling compensation.
> G29 S1        ; Save UBL mesh points to EEPROM.
> M500          ; Save current setup. WARNING: UBL will be active at power up, before any `G28`.
> 
> G26 C P T3.0  ; Produce mesh validation pattern with primed nozzle
> G29 P4 T      ; Move nozzle to 'bad' areas and fine tune the values if needed
>               ; Repeat G26 and G29 P4 O commands as needed.
> 
> G29 S1        ; Save UBL mesh values to EEPROM.

That can now all be done from the LCD Panel.  But doing that from a host interface **_DOES_** work.  I bet if you try it from PronterFace you are going to say "Oh...  This kind of makes sense...."

WheresWaldo, 2017-06-06:
@Roxy-3D I have done it that way, not an issue, but I am smarter than the average bear, or at least I like to think so!

I will try out all these methods tomorrow (actually later today after some sleep). Making note of which ones are actually intuitive for someone like BooBoo to use, easy and intuitive are not equivalent expressions. If there is a way to make as simple as MESH actually was, the beauty was in its simplicity and the fact that it actually worked, with a minimum of steps it will be a big win for everyone.

Tannoo, 2017-06-06:
>But when you are dealing with users, some of whom had no business buying a 3D Printer in the first place, you can't expect them to "upgrade" anything.

Believe me, I feel your pain. I have that in my profession also.

WheresWaldo, 2017-06-06:
Okay, Fine Tune All is not the answer! Fine tuning still requires you to GUESS what is the correct adjustment. Yes @Roxy-3D I know I can print a validation grid and "look" at it, but there is simply no feedback on Fine Tune to assure the user that it is correct. The functionality was there, now it is not. Another note to remember post code optimization. No noob want to input a bunch of numbers, they want to turn a knob and have it move. Yes UBL_MESH_EDIT_MOVES_Z is enabled.

Look at it from a complete idiots perspective. I had Auto-Level that never really worked good, some spots would be too close some spots too far, then MESH shows up and while it required more steps than including G29 in my slicer start up script, I only needed to do it once and the machine helped me do it by providing tactile feedback if equipped with an LCD controller. Now UBL is replacing MESH and you tell me I can do exactly the same thing, but it I need to use 7 more steps and I will get no tactile feedback from the printer and every number has to be figured out and entered by me. Why would idiot me want that, I barely understood what I needed to do with MESH and the machine helped, Now I have to do more and I don't get any help from the printer to boot.

That is exactly how this has to be viewed. You can talk until your blue in the face about how great and easy it is, but you are not seeing that it really isn't easy. It is more comprehensive and maybe even better, but you need to strike EASY from your vocabulary.

Sorry I am not in a great mood today as I was just informed that a closer friend has only a short time left as a result of cancer. Regardless the essence of this post stands. 

**tldr:** I had tactile feedback before, now I don't, I could do everything in a few steps, now I can't.

EddieFerg, 2017-06-06:
As an outsider with a “probeless” printer with LCD controller, please let me echo Waldo’s thoughts. I’m upgrading my son’s printer firmware and he needs to use it untethered from a PC.  Once the firmware is flashed, he needs to operate and calibrate it as a stand-alone machine. He has no GCode experience.

Currently we’re using the manual MESH process, which I understand is going away, but a Manual UBL method would still be an option.  MESH was quick, easy to understand, and it worked very well: Simply build a height map by using a paper feeler gauge, turn the knob for resistance, push the knob to go to the next point in the grid, repeat until done. Save.  Takes about 5 minutes to do a 7x7 grid. No tethering, no entering numbers.

I can understand automating the bed leveling process for more sophisticated machines and users, but please also include a MESH-like basic, simple, manual, mechanical-feedback, untethered process.  I’m happy to test and report back.

Roxy-3D, 2017-06-06:
> I can understand automating the bed leveling process for more sophisticated machines and users, but please also include a MESH-like basic, simple, manual, mechanical-feedback, untethered process. I’m happy to test and report back.

Yes...  The need is understood.   But seriously...   It is worth trying something and seeing if it works for you.   - First, get your Z-Endstop set so you can print a small object in the center of your bed with no bed leveling active.  This is mandatory.
- Set the entire mesh to 0.0000's with G29 P0.    
- Then do a G26 Mesh Validation Pattern and see what it looks like.  If you even get half of the pattern to print, you can move the nozzle to various problem areas and start editing the mesh.
- You **_WILL_** see an improvement after each iteration, and pretty soon you will be able to use the entire print bed with no adhesion issues.

EddieFerg, 2017-06-06:
Roxy - Thank you for your reply.

I'm willing to try, but issuing GCode means being tethered to a PC.  The premise of MESH was that a printer with an LCD controller could be calibrated as a stand-alone machine with no knowledge of GCode.

Or am I missing something?

Roxy-3D, 2017-06-06:
> I'm willing to try, but issuing GCode means being tethered to a PC. 

No...  This is not true.  The LCD Panel code (if you have a Graphics controller) will do all the commands you need in that list.  We are working to get the 20x4 LCD Panel's going.   Those steps up above may be working on 20x4 right now.  I don't know.

> The premise of MESH was that a printer with an LCD controller could be calibrated as a stand-alone machine with no knowledge of GCode.

No.  That was never the premise of Mesh Bed Leveling.  Mesh Bed Leveling happened because epatel could not get acceptable results using the grid based leveling that tilted an assumed perfectly flat plane.  

As it turns out...  very few pieces of glass are 'flat'.   They look flat, but they are not flat.   That is the premise for Mesh Bed Leveling.    But with that said...  The steps up above should work from a Graphical LCD Panel and if they don't work for a 20x4 LCD Panel, they will soon enough.

WheresWaldo, 2017-06-06:
@Roxy-3D I don't think the issue is that no one wants to try it your way, it is just that your way adds more steps that involve human interpretation of the results, each time you add that step you introduce another point where errors can creep into the process. 

Very simple, (I actually did this already today)

1. I set up an auto generated mesh, 
2. then print a mesh validation and visually interpret the results
3. then figure out where the mesh needs adjusting and adjust them
4. then reprint the validation mesh
5. then figure out where it still needs adjusting or I adjusted it incorrectly and adjust them, now I am closer
6. then print a third mesh validation print and adjust it now I am satisfied.

1. I ask MESH to create a mesh and measure all the points
Done

So a process that used to take 15 minutes now takes at the least an hour maybe more It actually took me about 1:40 minutes after all the prints and looking at the results and figuring out what needed changes and where. Regardless the intent of MESH it certainly was easier to use. But I also mentioned, that likely all the pieces are there already in your code, I trust that you have done all the work. Now it is time to think about presentation and how to implement all that work. Just so you don't think I oversimplified MESH, please note that there was tactile feedback so when the nozzle touched the feeler gauge you could actually feel that the gauge was touched, In it's current form there is no tactile feedback so I have to print the mesh and remove it without damage from the bed so I can use my calipers to measure all the intersecting points or make visual assumptions if I cannot remove the mesh without damage.

Unless your bed is milled tool plate then flat is not a word that can be used in any 3D printer that uses a build plate of another material. Then we add heat while constraining the bed at the mounting points.

Roxy-3D, 2017-06-06:
I really don't understand all the concern.   You can generate a mesh automatically with G29 P1 and G29 P3.  You can generate a mesh with manual probing if you want with G29 P2.    You can claim you bed is flat with G29 P0 and rely on the mesh validation pattern and interactive editing (G29 P4) to get it perfect.

It really doesn't matter how you decide to get a finely tuned mesh.  The tools are in place to do that even if your machine doesn't have a Z-Probe.   And the mesh can be saved so unless your printer gets damaged or worn, you can keep using the mesh you tuned.

Personally...  I hate doing manual probes.   But the G29 P2 code works nicely and can be used by people that don't have a Z-Probe and are willing to do that.   

It looks like the only way people will be happy is if we have a LCD Panel option that says "Setup UBL perfectly with one click."   But even then, the people without LCD Panel's would be complaining there is no way for them to do one click.

bgort, 2017-06-06:
After reading all of this, what I'm seeing is that there's a desire to have a MBL-like experience but with UBL?  Is that right? 

If so, why can't we just change things so `G29 P2` on an uninitialized mesh initializes it to 0 and then proceeds with a complete mesh generation using a business card or feeler gauge or whatever people like to use, just like MBL (which I used to use)?  This wouldn't be terribly complicated, and we could make everyone who doesn't want to be bothered with learning how to use UBL as it was intended to be used, feel better.  This would basically be just like MBL, more or less, and their step-by-step instructions would be:  '`G29 P2`, manually set every mesh z point just like with MBL, `G29 S1`, `M420 S1`, `M500`, done!' .. right?  Or am I missing something?  If people don't want to use UBL's extras (which are really helpful), so be it.

(FYI, I'd love to have a 'one-click-to-perfect-mesh' leveling system too, but I don't see where that's coming from until we all have laser triangulation sensors on our printers.)

WheresWaldo, 2017-06-06:
@Roxy-3D You feel like you are getting beat up, don't you?

Not intentional, but you are solely looking at this as a coding problem and not a usability problem. We are not asking to remove any existing functionality, auto generate to your hearts content, but please don't equate ultimate end result (a properly adjusted mesh) with the process used to achieve it. You (royally) in deprecating MESH took away an easy to use, easy to implement process that provided real time feedback with a multi-part setup that does not do any of those things. Yes the final end result is the same but it certainly isn't equivalent. I am going to say something I will likely regret, but this is the essence of the issue:

> I really don't understand all the concern.

No one asked for a One-click solution. Just not to remove functionality in an effort to make it 'better'.

WheresWaldo, 2017-06-06:
+bgort too bad I can't thumbs up your post more than once.

bgort, 2017-06-06:
Pretty sure this is just a few lines worth of changes.  Let me look.

bgort, 2017-06-06:
Yeah, looks like all we need to do is set the whole bed as 'unreachable' and then G29 P2 will allow you to manually set every point.

bgort, 2017-06-06:
@WheresWaldo:

Try this and tell me if this gives you what you want, effectively:
`G29 I 999`
`G29 P2 B`

WheresWaldo, 2017-06-06:
@bgort 99% there, it duplicated the functionality perfectly, I knew it was there all along. The only thing missing is visual feedback, but that can wait until all the important things are done. Process leaves you at the info screen the entire time.

bgort, 2017-06-06:
So then really all we need to do is add a menu item named 'MANUALLY GENERATE MESH' or something, which does `G29 I <mesh points>` and then `G29 P2 B`, and have some feedback on the screen during the G29 P2 process.  I believe what you're looking for was there at one point, but maybe was removed.

WheresWaldo, 2017-06-06:
Yep, that would do it. It results in a simple to use alternative that stores mesh information in the format UBL needs.

bgort, 2017-06-06:
I'm not all that familiar with the menu system - and I think it's being worked on pretty regularly by others anyway - so I'm not the best person to do that, but I can look into the on-screen feedback, unless anyone else wants to do it.

Roxy-3D, 2017-06-06:
Besides the nozzle moving so you can see where it is relative to the bed....  What else should be seen?   Probably the current (X,Y,Z) location of the nozzle.   Anything else?

Tannoo, 2017-06-07:
Where in the menu would ya'll like `MANUALLY GENERATE MESH` placed?

This will send `G29 I999\nG29 P2 B`, correct?

Roxy-3D, 2017-06-07:
How about we put it at the very top for a few weeks just so people know it is there.  Later we can put it in a more reasonable place.

Tannoo, 2017-06-07:
I tagged a `T0` at the end for those who do want the serial output during the build.

[14 more replies]

closed 2017-11-03 by thinkyhead