MarlinFirmware/Marlin #6967: G0/G1 move: F before E not working

JuDePom, 2017-06-06:
Hi,

I'm using Marlin 1.1.2 and I encounter some troubles

If I send **G1F100E1**, the **feedrate** is set **correctly** but the **motor doesn't move**.
If I send **G1E1F100**, it **works as expected**.

If 'F' is before 'E', the program goes through (in gcode_get_destination/Marlin_main.cpp)
```
if (parser.seen('F') && parser.value_linear_units() > 0.0)
    feedrate_mm_s = MMM_TO_MMS(parser.value_feedrate());
```
in the case 'E' followed by 'F', it doesn't go there.

Is this a wanted behaviour or a bug ?

Bob-the-Kuhn, 2017-06-06:
I'm not seeing that behavior.

There was a recent change (2? days ago) that enabled the no spaces format to work properly.  Please download the latest [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x) & see if that fixes the problem.

JuDePom, 2017-06-07:
Yeah, it was using no space formats :'(
(I've removed the 1.1.2 firmware which doesn't work, so I can't be sure about that)

But now it's good with 1.1.3!

Thank you,

closed 2018-02-07 by thinkyhead