MarlinFirmware/Marlin #6974: PROBE_OFFSET_FROM_EXTRUDER ignored when print has started

xabdullahx, 2017-06-07:
disclaimer: I am not sure if this issue is related to marlin or repetier.

I've just installed a sensor as my z-axis endstop. I've finetuned the offset, so when i home in repetier, i am able to get my nozzle at exactly the height that i want. But when i start the print, the print doesnt start at Z0, it looks like my Z_PROBE_OFFSET_FROM_EXTRUDER -1.5 is ignored.

This is my start gcode:
```
G28 ; Home extruder
G29 ; Auto bed leveling
G1 Z15 F{Z_TRAVEL_SPEED}
M107 ; Turn off fan
G90 ; Absolute positioning
M82 ; Extruder in absolute mode
{IF_BED}M190 S{BED}
; Activate all used extruder
{IF_EXT0}M104 T0 S{TEMP0}
G92 E0 ; Reset extruder position
; Wait for all used extruders to reach temperature
{IF_EXT0}M109 T0 S{TEMP0}
```

The relevant settings, if you need more, tell me:
```
#define X_PROBE_OFFSET_FROM_EXTRUDER -30  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 5  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.5   // Z offset: -below +above  [the nozzle]

#define Z_CLEARANCE_DEPLOY_PROBE   5 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
```

martindb, 2017-06-07:
Doyou have initialized the eeprom values?

xabdullahx, 2017-06-07:
I probably haven't, because i don't understand what you mean

martindb, 2017-06-07:
If you have eeprom enabled, and there is a previous value stored as your z probe offset, marlin is using it instead of what you have defined in your #define line.
Do a M502, M500 and M501 to initialize the eeprom.
You can read the details here
https://github.com/MarlinFirmware/Marlin/wiki/EEPROM

xabdullahx, 2017-06-07:
Looks like eeprom is disabled by default in marlin

martindb, 2017-06-07:
Yes.

xabdullahx, 2017-06-07:
Meaning that my eeprom is disabled. Then that isnt the problem. Thanks for replying anyway

Bob-the-Kuhn, 2017-06-07:
Repetier Host isn't involved in this.  All it does is pass commands on to the printer.  

How much higher do you think it is?

Being off by 0.05mm can be the difference between a great first layer and garbage.  You simply can't get that accuracy by mechanically testing the nozzle to bed gap.  You have to adjust the offset based on test prints.

Are you using a bed leveling system?

xabdullahx, 2017-06-07:
Yes I am using the linear one.

I've uploaded two pictures, the first is when I start up repetier and, and home z, then move z to 0

The second is when I start the print, then it auto homes, then probes the bed, then it starts extruding the first layer at this height :
[image]
[image]

Bob-the-Kuhn, 2017-06-07:
Yeah - 10mm is a bit high for a first layer.

I'd like to see you config info.  Please ZIP up the two config files and a gcode/print file and drop it on your reply.

xabdullahx, 2017-06-07:
Thanks for looking into it. here are the requested files:

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1059610/Marlin.zip)

Bob-the-Kuhn, 2017-06-08:
I didn't see anything unusual in the files except that the config files are out of date.

I'm not aware of any fixes since 1.1.2 that pertain to this problem but ... the best bet right now is for you to try the latest [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x) and to migrate your machine specific settings over to the new config files.

Have you tried an X-Y move after enabling the leveling?  If I remember correctly the Z axis won't be compensated until a X-Y move is done within the bed area.  You can do all the Z only moves you want but the Z compensation isn't applied until the first XY move is done.

xabdullahx, 2017-06-08:
Thank you for your help Bob. Unfortunately, i didnt get it to work, so instead, i spent some time physically adjusting the sensor until i got it right. (and put the zoffset to 0)

I will close this issue. Thanks to everyone that replied.

closed 2017-06-08 by xabdullahx