MarlinFirmware/Marlin #6975: 1.1.3 bugfix won't compile with Extruder>1

Kaibob2, 2017-06-07:
I just downloaded the most recent bugfix-1.1.x version and when i try to compile it with Arduino IDE 1.8.3 it throws the following Error:
```
Arduino: 1.8.3 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\default\AppData\Local\Temp\arduino_build_651875\sketch\Marlin_main.cpp: In function 'void gcode_M125()':

Marlin_main.cpp:7930: error: assignment of read-only variable 'x_pos'

         if (!parser.seen('X')) x_pos += hotend_offset[X_AXIS][active_extruder];

                                      ^

Marlin_main.cpp:7931: error: assignment of read-only variable 'y_pos'

         if (!parser.seen('Y')) y_pos += hotend_offset[Y_AXIS][active_extruder];

                                      ^

Bibliothek U8glib in Version 1.19.1 im Ordner: C:\Users\default\Documents\Arduino\libraries\U8glib  wird verwendet
exit status 1
assignment of read-only variable 'x_pos'
```
I have set:
#define EXTRUDERS 2
and
//#define DUAL_X_CARRIAGE

I'm upgrading from RCBugFix.
All config Files are updated via WinMerge.

Kaibob2, 2017-06-07:
Removing the `const` from `float x_pos` and `float y_pos` in line 7917 and 7922 in Marli_main.cpp fixes this issue. I just cannot estimate what this means for the rest of the code :)

Bob-the-Kuhn, 2017-06-07:
The scope of those variables is very limited.  

I'll put this change into a pull request.  That'll get it reviewed by others as there may be a better solution.

thinkyhead, 2017-06-07:
I will have a patch for that shortly!

closed 2017-06-07 by thinkyhead

Bob-the-Kuhn, 2017-06-07:
@thinkyhead - faster than a speeding (OK, plodding along) Bob!

commit 4406fba9 (2017-06-07): Fixes #6975