MarlinFirmware/Marlin #6980: 1.1.3 Bugfix won't compile when #define CASE_LIGHT_ENABLE uncommented

Kaibob2, 2017-06-07:
Setting the case light section like this:
```
/**
 * M355 Case Light on-off / brightness
 */
#define CASE_LIGHT_ENABLE
#if ENABLED(CASE_LIGHT_ENABLE)
  #define CASE_LIGHT_PIN 11          // can be defined here or in the pins_XXX.h file for your board
                                    //  pins_XXX.h file overrides this one
  #define INVERT_CASE_LIGHT false             // set to true if case light is ON when pin is at 0
  #define CASE_LIGHT_DEFAULT_ON false          // set default power up state to on or off
  #define CASE_LIGHT_DEFAULT_BRIGHTNESS 105   // set power up brightness 0-255 ( only used if on PWM
                                              // and if CASE_LIGHT_DEFAULT is set to on
  #define MENU_ITEM_CASE_LIGHT              // Uncomment to have a Case Light entry in main menu
#endif
```
leads to the following compilation errors

```
In file included from C:\Users\Kai\AppData\Local\Temp\arduino_build_101980\sketch\Marlin_main.cpp:245:0:

C:\Users\Kai\AppData\Local\Temp\arduino_build_101980\sketch\gcode.h:123:43: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]

     static volatile bool seen(const char c) {

                                           ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:24:0,

                 from C:\Users\Kai\AppData\Local\Temp\arduino_build_101980\sketch\MarlinConfig.h:37,

                 from C:\Users\Kai\AppData\Local\Temp\arduino_build_101980\sketch\Marlin.h:36,

                 from C:\Users\Kai\AppData\Local\Temp\arduino_build_101980\sketch\Marlin_main.cpp:230:

C:\Users\Kai\AppData\Local\Temp\arduino_build_101980\sketch\Marlin_main.cpp: In function 'void gcode_M115()':

Marlin_main.cpp:7838: error: expected primary-expression before 'bool'

       bool USEABLE_HARDWARE_PWM(uint8_t pin);

       ^

C:\Users\Kai\AppData\Local\Temp\arduino_build_101980\sketch\Marlin_main.cpp: In function 'void gcode_M125()':

C:\Users\Kai\AppData\Local\Temp\arduino_build_101980\sketch\Marlin_main.cpp:7935:16: warning: unused variable 'job_running' [-Wunused-variable]

     const bool job_running = print_job_timer.isRunning();

                ^

C:\Users\Kai\AppData\Local\Temp\arduino_build_101980\sketch\Marlin_main.cpp: In function 'void update_case_light()':

C:\Users\Kai\AppData\Local\Temp\arduino_build_101980\sketch\Marlin_main.cpp:9646:13: warning: unused variable 'case_light_bright' [-Wunused-variable]

     uint8_t case_light_bright = (uint8_t)case_light_brightness;

             ^

Bibliothek U8glib in Version 1.19.1 im Ordner: C:\Users\Kai\Documents\Arduino\libraries\U8glib  wird verwendet
exit status 1
expected primary-expression before 'bool'
```

Kaibob2, 2017-06-07:
I just saw, that in fastio.h there is the line
```
#if AVR_ATmega2560_FAMILY
  #define PWM_PINS(p)  ((p >= 2 && p <= 10 ) || p ==  13 || p ==  44 || p ==  45 || p ==  46 )
```
But i always had my case light wired to PIN 11
Isn't PIN 11 also PWM capable??
Btw. i don't need a PWMable case light. On/Off would be pretty fine.
When i first implemented the case light feature #5255 i didn't expect it to become that bloated.

Bob-the-Kuhn, 2017-06-07:
So you're to blame for case light.  😉

When I was making an unrelated change I saw that it meant to allow PWM variable brightness but missed the mark.  Since then it's become a thorn in my side.  Hopefully all the user interface concerns have been settled.  Classic case of too many cooks.

The 2560 data sheet says that pin 11 is a PWM pin but Marlin is using the counter in such a way that the PWM function isn't available.  There's a write up on this in the new [wiki](http://marlinfw.org/docs/configuration/laser_spindle.html).

bgort, 2017-06-07:
Bob, I'm thinking we need to make sure that the case light comes on when the coffee maker is toggled on, so we should probably at least add a sanity check to make sure that CASE_LIGHT_ENABLE is defined when COFFEE_MAKER_ENABLE is defined.  We need light to make sure we don't want to spill the coffee.  😉 

And maybe we can make the light blink when the coffee is finished??

Bob-the-Kuhn, 2017-06-07:
Blame it on @Kaibob2 - he didn't include it in the original design specs. 😞

Kaibob2, 2017-06-08:
COFFEE_MAKER_ENABLE isn't working anymore, too. I had my automatic coffee grinder
connected to PIN12 😉
Anyway, i will change the layout of my Extension PCB so that i can use PIN 44 instead.
For now: Isn't it possible to check if the assigned PIN is PWM capable and if not just use On/Off?

Btw. a fresh copy with only #define CASE_LIGHT_ENABLE uncommented doesn't compile too and exits with `Marlin_main.cpp:7830: error: expected primary-expression before 'bool'`

Kaibob2, 2017-06-08:
> For now: Isn't it possible to check if the assigned PIN is PWM capable and if not just use On/Off?

Sorry, i just found, that this is already done in `void update_case_light() {`
So then there is just a problem in line 7830 with `bool USEABLE_HARDWARE_PWM(uint8_t pin);`

Bob-the-Kuhn, 2017-06-08:
Yep - Marlin_main.cpp line 7838 shouldn't be there.  Just delete it.

I'll do a PR to fix it in the morning.

Kaibob2, 2017-06-08:
@Bob-the-Kuhn I did a PR already. Thanks for taking a look at it.

closed 2017-06-08 by Kaibob2