MarlinFirmware/Marlin #6984: ULTRA_LCD support for Melzi?

ZirakC, 2017-06-08:
Hi,

I am enabling ULTRA_LCD for Melzi, however, I get the following error:

Arduino: 1.6.8 (Windows 10), Board: "Sanguino, ATmega1284p 16MHz"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\Zirak\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\Zirak\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Zirak\Documents\Arduino\libraries" -fqbn=marlin:avr:sanguino:cpu=atmega1284 -ide-version=10608 -build-path "C:\Users\Zirak\AppData\Local\Temp\build2a1015e1aae91b6d5a9ef5b53de5a5b6.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\Zirak\Desktop\Marlin\tags\1.1.0-RC6\Marlin\Marlin.ino"
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\Zirak\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\Zirak\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Zirak\Documents\Arduino\libraries" -fqbn=marlin:avr:sanguino:cpu=atmega1284 -ide-version=10608 -build-path "C:\Users\Zirak\AppData\Local\Temp\build2a1015e1aae91b6d5a9ef5b53de5a5b6.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\Zirak\Desktop\Marlin\tags\1.1.0-RC6\Marlin\Marlin.ino"
WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'
Warning: platform.txt from core 'Marlin AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\variants\sanguino" "C:\Users\Zirak\AppData\Local\Temp\build2a1015e1aae91b6d5a9ef5b53de5a5b6.tmp\sketch\Marlin.ino.cpp" -o "nul"
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\variants\sanguino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\libraries\LiquidCrystal\src" "C:\Users\Zirak\AppData\Local\Temp\build2a1015e1aae91b6d5a9ef5b53de5a5b6.tmp\sketch\Marlin.ino.cpp" -o "nul"
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\variants\sanguino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\libraries\LiquidCrystal\src" "C:\Program Files (x86)\Arduino\hardware\marlin\avr\libraries\LiquidCrystal\src\LiquidCrystal.cpp" -o "nul"
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\variants\sanguino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\libraries\LiquidCrystal\src" "C:\Users\Zirak\AppData\Local\Temp\build2a1015e1aae91b6d5a9ef5b53de5a5b6.tmp\sketch\M100_Free_Mem_Chk.cpp" -o "nul"
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\variants\sanguino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\libraries\LiquidCrystal\src" "C:\Users\Zirak\AppData\Local\Temp\build2a1015e1aae91b6d5a9ef5b53de5a5b6.tmp\sketch\Marlin.ino.cpp" -o "nul"
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\variants\sanguino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\libraries\LiquidCrystal\src" "C:\Users\Zirak\AppData\Local\Temp\build2a1015e1aae91b6d5a9ef5b53de5a5b6.tmp\sketch\MarlinSerial.cpp" -o "nul"
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\variants\sanguino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\libraries\LiquidCrystal\src" "C:\Users\Zirak\AppData\Local\Temp\build2a1015e1aae91b6d5a9ef5b53de5a5b6.tmp\sketch\Marlin_main.cpp" -o "nul"
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\variants\sanguino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\libraries\LiquidCrystal\src" "C:\Users\Zirak\AppData\Local\Temp\build2a1015e1aae91b6d5a9ef5b53de5a5b6.tmp\sketch\Sd2Card.cpp" -o "nul"
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\variants\sanguino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\libraries\LiquidCrystal\src" "C:\Users\Zirak\AppData\Local\Temp\build2a1015e1aae91b6d5a9ef5b53de5a5b6.tmp\sketch\SdBaseFile.cpp" -o "nul"
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\c [... cut ...]
   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7

                                                                                      ^

Using library LiquidCrystal at version 1.0.1 in folder: C:\Program Files (x86)\Arduino\hardware\marlin\avr\libraries\LiquidCrystal 
exit status 1
'LCD_PINS_RS' was not declared in this scope

It compiles when I re-comment this line, however, my LCD and the directional buttons will no longer function. Also, the printer in question is the "Afinibot A3 3D Printer".

fiveangle, 2017-06-08:
You didn't even include your config files :)

pin*.h for Melzi has the requirement for both ULTRA_LCD and NEWPANEL to be defined, or you won't get LCD_PINS_*.

```
//
// LCD / Controller
//
#if ENABLED(ULTRA_LCD) && ENABLED(NEWPANEL)
```

Good luck !

-=dave

Bob-the-Kuhn, 2017-06-08:
Usually you just enable the LCD you're using and let the system enable/disable ULTRA_LCD , NEWPANEL, DOGLCD, MAKRPANEL, ... automatically.   I think they're in the config file in case your using a LCD that isn't in the list of already supported boards.

Yes, copies of your config files would be very helpful.  Just ZIP them up and drop the ZIP on your reply.

What LCD are you using?

fiveangle, 2017-06-08:
Didn't notice he said the name - it looks like probably a standard LCD 2004 20x4 RepRap Smart Controller style:

http://www.afinibot3dprinter.com/goods/Afinibot-Prusa-I3-3D-Printer.html

ZirakC, 2017-06-08:
Sorry about that, my config files can be downloaded here:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1059938/Marlin.zip)

There are a few things I need to still fix (aside form the LCD thing) such as X/Y/Z homing, they don't work properly at the moment.

fiveangle, 2017-06-08:
you should write to whoever you bought the printer from and request the source - i see they only distribute the binary - if they have the source somewhere, you can just translate all the settings from there to the current Marlin pretty easlily.

ZirakC, 2017-06-08:
Yep already have the source of a pretty old Marlin version (I think 2014ish?) was able to convert most of it but was stuck on the LCD. The source is telling me to use RepRap keypad, problem is it uses pins that don't exist on the Melzi in the new version.

Need to sleep for work, will be able to attach the source they provided me tomorrow evening, gn all! Thanks for the help so far.

Bob-the-Kuhn, 2017-06-08:
First thing we need to do is get you on a version of Marlin that we can work with.  Please download  [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x).

I've converted your config files over to the new system.  Please do a sanity check on them and then copy them into bugfix-1.1.x.  [Configuration_1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/1060067/Configuration_1.1.x.zip)

We still have two open issues. 
- thermistor 501 - I'll need to see your thermistortables.h file in order to make an educated guess.
- We need to identify the LCD controller that you're using.  Please see if there's some identifier on it.  Most of the time the name is on the silkscreen on the LCD pwb.  We may also need to verify which pins the LCD is physically connected to.  Worst comes to worst we just try the different  sets of pin definitions in the file.

What IDE extension are you using to compile the code?

ZirakC, 2017-06-08:
I'm using the 1.1.x bugfix, however, now I'm getting this error when compiling:
Arduino: 1.6.8 (Windows 10), Board: "Sanguino, ATmega1284p 16MHz"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\Zirak\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\Zirak\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Zirak\Documents\Arduino\libraries" -fqbn=marlin:avr:sanguino:cpu=atmega1284 -ide-version=10608 -build-path "C:\Users\Zirak\AppData\Local\Temp\builda5a3a390c1d4431cef58fd4bf233f7fc.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\Zirak\Desktop\Marlin\branches\bugfix-1.1.x\Marlin\Marlin.ino"
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\Zirak\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\Zirak\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Zirak\Documents\Arduino\libraries" -fqbn=marlin:avr:sanguino:cpu=atmega1284 -ide-version=10608 -build-path "C:\Users\Zirak\AppData\Local\Temp\builda5a3a390c1d4431cef58fd4bf233f7fc.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\Zirak\Desktop\Marlin\branches\bugfix-1.1.x\Marlin\Marlin.ino"
WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'
Warning: platform.txt from core 'Marlin AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\variants\sanguino" "C:\Users\Zirak\AppData\Local\Temp\builda5a3a390c1d4431cef58fd4bf233f7fc.tmp\sketch\Marlin.ino.cpp" -o "nul"
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\variants\sanguino" "C:\Users\Zirak\AppData\Local\Temp\builda5a3a390c1d4431cef58fd4bf233f7fc.tmp\sketch\G26_Mesh_Validation_Tool.cpp" -o "nul"
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\variants\sanguino" "C:\Users\Zirak\AppData\Local\Temp\builda5a3a390c1d4431cef58fd4bf233f7fc.tmp\sketch\M100_Free_Mem_Chk.cpp" -o "nul"
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\variants\sanguino" "C:\Users\Zirak\AppData\Local\Temp\builda5a3a390c1d4431cef58fd4bf233f7fc.tmp\sketch\Marlin.ino.cpp" -o "nul"
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\marlin\avr\variants\sanguino" "C:\Users\Zirak\AppData\Local\Temp\builda5a3a390c1d4431cef58fd4bf233f7fc.tmp\sketch\MarlinSerial.cpp" -o "nul"
"C:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Users\Zirak\AppData\Local\Arduino15\packages\arduino\hardwar [... cut ...]
                 from C:\Users\Zirak\Desktop\Marlin\branches\bugfix-1.1.x\Marlin\Marlin.ino:31:

SanityCheck.h:206: error: #error "CONTROLLERFAN_PIN is now CONTROLLER_FAN_PIN, enabled with USE_CONTROLLER_FAN. Please update your Configuration_adv.h."

   #error "CONTROLLERFAN_PIN is now CONTROLLER_FAN_PIN, enabled with USE_CONTROLLER_FAN. Please update your Configuration_adv.h."

    ^

exit status 1
#error "Marlin requires C++11 support (gcc >= 4.7, Arduino IDE >= 1.6.8). Please upgrade your toolchain."

ZirakC, 2017-06-08:
I decided to get the latest Arduino version (1.8.3), but now I can't even load my boards since the new version 1.1.x doesn't have any that bundles with it. I tried using the one from 1.1.0 rc6

But I get this:
Could not find boards.txt in C:\Program Files (x86)\Arduino\hardware\Gen7-dist\cores. Is it pre-1.5?
Could not find boards.txt in C:\Program Files (x86)\Arduino\hardware\Gen7-dist\variants. Is it pre-1.5?
WARNING: Error loading hardware folder C:\Program Files (x86)\Arduino\hardware\Gen7-dist
  No valid hardware definitions found in folder Gen7-dist.
Could not find boards.txt in C:\Program Files (x86)\Arduino\hardware\Melzi\bootloaders. Is it pre-1.5?
Could not find boards.txt in C:\Program Files (x86)\Arduino\hardware\Melzi\cores. Is it pre-1.5?
Could not find boards.txt in C:\Program Files (x86)\Arduino\hardware\Melzi\variants. Is it pre-1.5?

Bob-the-Kuhn, 2017-06-09:
1.8.3, bugfix-1.1.x and the config files I supplied should give you a semi-useable image.

Here's the Sanguino IDE extension that I use:
```
/**
 * A useable Arduino IDE extension (board manager) can be found at
 * https://github.com/Lauszus/Sanguino
 *
 * This extension has been tested on Arduino 1.6.12 & 1.8.0
 *
 * Here's the JSON path:
 * https://raw.githubusercontent.com/Lauszus/Sanguino/master/package_lauszus_sanguino_index.json
 *
 * When installing select 1.0.2
 *
 * Installation instructions can be found at https://learn.sparkfun.com/pages/CustomBoardsArduino
 * Just use the above JSON URL instead of Sparkfun's JSON.
 *
 * Once installed select the SANGUINO board and then select the CPU.
 *
 */
```

Please download bugfix-1.1.x from https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x.  We fixed a couple of nasty bugs a few hours ago.

Good news on the thermistor.  The vendor files you supplies says 501 is "100k ParCan thermistor (104GT-2)" which is now called just 5.  The configuration.h file should be changed to read:
```
#define TEMP_SENSOR_0 5
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_4 0
#define TEMP_SENSOR_BED 5
```

Any info on which LCD you have?

ZirakC, 2017-06-09:
The LCD is c2004a-v2.0

Seems to closely resemble this:
http://forum.repetier.com/discussion/1105/melzi-v2-0-with-lcd2004-and-5-keys

By the way, I used git to grab the latest version of the files, so I think I should be up to date.
I'll swap the thermister to 5.

Bob-the-Kuhn, 2017-06-09:
Is this one a better match?  
http://www.ebay.com/itm/LCD-2004-controller-display-for-Makerbot-3D-Printer-control-board-Mightyboard-/311370338924?hash=item487f1e526c:g:SuUAAOSwv0tVZ-XB

Bob-the-Kuhn, 2017-06-09:
I'm not finding anything useful on the one you pointed me at.

Does yours have an SD card reader on it?

If the one I pointed you at isn't a perfect match then I think it's time that you start looking for a new LCD display.  Heck - for $40 you can get a nice display, an Arduino 2560 & a RAMPs card.  It would require changing the connectors on the exist cables.

ZirakC, 2017-06-09:
The SD card is on the actual Melzi board,
Here are some photos:
[image]
[image]

ZirakC, 2017-06-09:
Not on my computer atm but I think I might be able to get it working based off this:

http://forum.repetier.com/discussion/1918/board-anet-v1-0-display-lcd2004a-with-5-buttons

Will try it out tomorrow. Thanks @Bob-the-Kuhn for your help so far, I really appreciate the support!

Bob-the-Kuhn, 2017-06-09:
A picture of the back of the LCD will also do.

Bob-the-Kuhn, 2017-06-09:
I just compared the button positions relative to the display & the one I pointed at is NOT it.

I looked through the Marlin code & didn't find anything that supported analog buttons.  

Given enough time I'm sure that we could get the LCD portion working.  That would require a lot of try this & try that.

How functional was your original firmware?  No LCD & homing problems.

I haven't been able to compile the source code you provided.  Have you been able to?

Bob-the-Kuhn, 2017-06-09:
If you want the buttons to be functional then you'll need to stick with the old software.  Sounds like Repetier has support for them but only the old software will have them connected/functional.

ZirakC, 2017-06-09:
Will try to tinker a bit when I get home from work.

ZirakC, 2017-06-09:
Yeah I think it's this version
https://gist.github.com/erikkallen/584dd986a4e3d93265bb3ca3c60ec8e3

Still at work so can't test it yet, will let you know in 5 hours.

benlye, 2017-06-09:
This is a maintained Marlin fork for the Anet A8 printer, which uses the 5-button ADC panel:
https://github.com/SkyNet3D/Marlin/

@oderwat keeps it pretty up to date with the current Marlin code.  Some work has been done to get the ADC code ready to contribute back to the main Marlin repo, but I don't know where that has got to.  If you have the same panel you should be able to get it to work without too much headache.

Bob-the-Kuhn, 2017-06-09:
@benlye - YOU'RE WONDERFUL!!!!

I believe this fork of Marlin will do what we need.

It's missing some defines but that can be easily fixed.  The main thing is it has support for the analog pins and it appears to say that the LCD controller is software compatible with the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.

It looks like the branch we want is https://github.com/SkyNet3D/Marlin/tree/SkyNet3D-Oderwat

ZirakC, 2017-06-09:
Oh wow that sounds excellent, haven't had the chance to look at it yet, any ideas on whether the changes can be merged with the 1.1.3 bug fixes? Or are they very drastic changes?

ZirakC, 2017-06-10:
SUCCESS! Had to change a bunch of stuff to get it working using the Skynet firmware, but I was able to use the LCD screen and the buttons to navigate successfully.

It seems like I am having an issue with the heat bed, the LED seems to blink, when it should stay on solid. Because of this the heat up of the head bed is A LOT longer than it normally took.

I have attached the Skynet source for reference, I am thinking it might be some sort of setting in Configuration_ADV that I might need to make, but I am not too sure which one.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1065490/Marlin.zip)

ZirakC, 2017-06-10:
Got everything working with the Skynet firmware!!
Just a small glitch with the LCD screen, the "X", "Y" and "Z" blinks every second, or gets replaced with "?". Not sure if this issue was ever encountered before.

Bob-the-Kuhn, 2017-06-10:
YEAH!!!!

The blinking and the ? are because G28 hasn't been executed yet.,

ZirakC, 2017-06-10:
Thanks @Bob-the-Kuhn  and @benlye for the support!

For reference (if anyone ever needs it in the future) the source can be downloaded from here:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1065565/Marlin.zip)

closed 2017-06-10 by ZirakC

oderwat, 2017-06-10:
Glad it helped! Looks like I should finish the PR for the Anet board soon (tm). I am just super busy with my dayjob right now.

Bob-the-Kuhn, 2017-06-10:
Please do me a BIG favor and test this [code](https://www.dropbox.com/s/yvrufbsfukpa5m7/PR%207016%20rev%201.zip?dl=0. ).  

It's my attempt to bring your working software into Marlin via PR #7016.

oderwat, 2017-06-10:
I'd rather clone you repository than downloading zips :)

Bob-the-Kuhn, 2017-06-10:
https://github.com/Bob-the-Kuhn/RCBugfix-M43-and-spindle/tree/Skynet