MarlinFirmware/Marlin #6987: Enabling BABYSTEP_ZPROBE_OFFSET and MESH_BED_LEVELING results in compile error

FHeilmann, 2017-06-08:
```
C:\***\sketch\ultralcd.cpp: In function 'void lcd_babystep_zoffset()':

ultralcd.cpp:995: error: 'class Planner' has no member named 'abl_enabled'

             if (planner.abl_enabled)

                         ^

Using library U8glib at version 1.19.1 in folder: C:\***\libraries\U8glib 
exit status 1
'class Planner' has no member named 'abl_enabled'
```

Probably needs a sanity check to disallow BABYSTEP_ZPROBE_OFFSET when MESH_BED_LEVELING is used

Bob-the-Kuhn, 2017-06-08:
I think that's a bug.  That variable only exists if ABL is enabled.  I think that line should be made a conditional compile.

I'll put together a PR so we can get comments on how to fix it.

Bob-the-Kuhn, 2017-06-08:
The PR number is #6991.

I put "proposal" in the title so that it'd get lots of attention which should generate comments on the best way to fix this.

Bob-the-Kuhn, 2017-06-17:
I messed up when I created the PR.  

I was thinking MESH_BED_LEVELING and babystepping was throwing the compile error.  It was MESH_BED_LEVELING with BABYSTEP_ZPROBE_OFFSET that rightfully resulted in a compile error.

I've modified sanity check to provide a clear error message if this happens.

FHeilmann, 2017-07-05:
Sanity check is now in place to avoid this combo. Closing

closed 2017-07-05 by FHeilmann

fiveangle, 2017-07-08:
Aside: I've always wondered, what the heck is BABYSTEPPING anyway ?

Roxy-3D, 2017-07-08:
BabyStepping is a way to move an axis independent of Marlin 'knowing' about it.    I'm a big fan of Z-BabyStepping to help get the first layer down exactly correct.    I added the 'Double Click for Z-BabyStepping' to help accomplish just that.   When the first layer is doing the skirt, if things are not exactly right...  Double Click lets you adjust the height of the nozzle and get it exactly where it needs to be.

fiveangle, 2017-07-08:
OMG what a horribly unclear name.

So it allows one to adjust Z offset on the fly similar to the "Live Z Adjust" feature of the Prusa fork ?

With a name that implies it is related to low level Stepper driver function it's wonder nobody knows (or cares) about it.

Thanks for the tip.

fiveangle, 2017-07-08:
No lie, I imagined it might be related to an LCD menu that would allow one to adjust software to combine micro steps during high speed movements if they were encountering studdering (or some other hack like that) =p

Roxy-3D, 2017-07-08:
Well...  It does let you adjust the axis at the micro step level.   It really is good at doing that.  And for the first layer...  It is very helpful.