MarlinFirmware/Marlin #6988: MESH_BED_LEVELING exhibits weird behavior on CoreXY

FHeilmann, 2017-06-08:
I just tried to switch to MESH_BED_LEVELING for the fun of it and my CoreXY printer went batshit crazy as a result of it.

My working copy can be found here:

https://github.com/FHeilmann/Marlin/tree/bugfix-1.1.x

What I have found so far:

Homing performs a double touch, even though it is disabled. The double touch is made at full speed for Z and "minimum speed" for X and Y, even though HOMING_BUMP_DIVISOR is set to reasonable values in the configuration regardless of the bump being used or not.

Is there supposed to be no mention of leveling in the LCD anymore? I also think "Level Bed" disappeared for BILINEAR as well.

Fast movements in XY direction cause grinding steppers at speeds which are perfectly fine with BILINEAR.

This is all on a freshly synced bugfix-1.1.x from today.

Bob-the-Kuhn, 2017-06-08:
Please download [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x).  We just did a couple of changes that could solve this problem.

FHeilmann, 2017-06-12:
Synced again today, Issues are resolved as far as I can tell.

closed 2017-06-12 by FHeilmann