MarlinFirmware/Marlin #6989: G26 starts flooding the terminal

FHeilmann, 2017-06-08:
I wanted to bring up UBL, probed my bed whilst my temps were at 220/60, then issued a

`G26 B50 H200 F1.75 L0.3 N0.4`

which resulted in my terminal being flooded with temperature reports which also don't have proper line endings so they show up without any "newlines". This happens until the heaters reach the appropriate temps

bgort, 2017-06-08:
I'm aware of this one and it's on my to-do list.  I'm pretty sure I know where the problem is, but I haven't yet tried my fix.

BlackLightUK, 2017-06-08:
@FHeilmann `N0.4` isn't defined in the gcode documentation for G26 - what is it ? Nozzle diameter ?

bgort, 2017-06-08:
Nozzle size is S.  As in `S0.4` .. I don't think there's an `N`.

FHeilmann, 2017-06-08:
Then the comments on marlinfw.org should probably also change ;)

Regardless, if the parameter mixup really is the reason for the flooding it should be fixed too.

bgort, 2017-06-08:
The parameter mixup isn't the issue, as far as I know now.  I'll try to fix it later today.  Have been meaning to for a while.

FHeilmann, 2017-07-05:
@bgort can you point me towards your fix idea for this?

FHeilmann, 2017-08-17:
solved by #7515

closed 2017-08-17 by FHeilmann