MarlinFirmware/Marlin #6997: Delta G33 Auto Level, Bed not leveled

novio8, 2017-06-08:
I have latest Marlin 1.1.3 I enabeled
delta Calibration wit fixed Z probe. I run G33 P7, everything goes well
M500, but the Bed is not leveled if I check it manually, about a millimetre 
or so in the center from the corners
Am I doing something wrong or ?

Please advise
Mario

oldmcg, 2017-06-09:
Is your Z-probe in the nozzle or offset from the nozzle?  If offset from nozzle then you need pending PR #6850 from @LVD-AC, but that is for bugfix_1.1.x branch, not 1.1.3.  I recommend switching to bugfix_1.1.x branch, and if you have XY offset probe then manually apply the #6850 changes if you are able to do that.

My experience with G33 (with #6850) is that I still need UBL to compensate, but not nearly as much tilt shown in the bed in the UBL mesh due to G33 corrections.

novio8, 2017-06-09:
My Z-probe's offset is only in Y direction. I'll try and apply 6850 changes. 
If I understand you correctly you still use UBL as your  Bed Leveling procedure.
I did not try UBL yet, Is there anythin specific fo Deltas, or anything else
as I'm getting frustrated wit AutoCalibration. I'v  returned to Marlin 1.1.2 and manually
calibrated the Bed, but I would really like to have AC, but don't wanna loose another 
4 days.

LVD-AC, 2017-06-09:
Or you can use my rep https://github.com/LVD-AC/Marlin-AC; PR #6850 is already incorporated in 1.1.3 there.

novio8, 2017-06-09:
thanks, but that is just what I did. I upgraded from 1.1.2 to 1.1.3 and than my troubles started.
Maybe I made a mistake when I was editing new configuration file 1.1.3, I double checked new
configuration file ageinst the one from 1.1.2 and everything seems OK

oldmcg, 2017-06-09:
For delta and specficially UBL on delta, stick with bugfix_1.1.x branch for now.  I just added another round of UBL on delta improvements a few days ago in bugfix_1.1.x.  

To do UBL bed leveling on delta, after G33 all done and saved, run G29 P1 to auto-probe, then G29 P2 to manually probe (with nozzle and paper and dial on the LCD) the few points the probe can't reach because it is offset from nozzle, then run G29 P3.15 (not yet documented) to extrapolate all the remaining mesh points, save it to eeprom with G29 S0, then give G26 a try to print a one-layer pattern to see how well it sticks to the bed.

LVD-AC, 2017-06-09:
??? most of PR #6850 was already merged in 1.1.2-AC, 1.1.3-AC only has an extra small bug fixed in the LCD menu

novio8, 2017-06-10:
so if I just take config file from 1.1.2 to 1.1.3 it shoul work, if it works in 1.1.2

LVD-AC, 2017-06-10:
https://github.com/LVD-AC/Marlin-AC is Marlin 1.1.3 with #6850 included
Latest bugfix-1.1.x you find at https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x but you will have to merge #6850 yourself (or wait until @thinkyhead finds the time to do it)

novio8, 2017-06-26:
Hi yesterday I downloaded and installed
https://github.com/LVD-AC/Marlin-AC is Marlin 1.1.3 with #6850 included
I'v run G33 or G33 P4 C0.02, bed leveling starts, one round
and stops. Repetier report:
19:37:27.093 : Calibration OK                                    std dev:-2147483.750
19:37:27.096 : .Height:265.64    Ex:+0.00  Ey:+0.00  Ez:+0.00    Radius:163.00
19:37:27.097 : .Tower angle :    Tx:+0.00  Ty:+0.00  Tz:+0.00
19:37:27.100 : Save with M500 and/or copy to Configuration.h

So everything is 0.
What am I doing wrong again ?
Before upgrading I manually leveled the bed and I'v got some good prints.
Any suggesyions LVD-AC ?
thank you

GibaHC, 2017-07-22:
With the stable 1.1.4, is the UBL working for deltas ? The UBL documentation on the site is from 1.1.2 and could not clarify that. 
I am currently using ABL bilinear after g33 auto calibration, but I can't print large objects. The mesh can't compensate far from the middle. Maybe UBL is a step forward.

Roxy-3D, 2017-07-22:
UBL should be working in 1.1.4 for Delta's.   It should be able to compensate for a Cup & Bowl problem on a Delta bed.

GibaHC, 2017-07-24:
Today I tried the UBL. Manage to get a nice flat middle bed, but the 5cm away from the center I got some problems. Probably because I couldn't probe  manually the points that are unreachable for auto probing. I'll give another try as soon as my screen arrives. 

When configuring the UBL, I had to set 3 points in bed for eventual 3 point mesh. Nothing hard, but ... if the size of the bed is already set in config, why aren't they calculated instead of user declared?

oldmcg, 2017-07-24:
Regarding the 3-point for UBL on delta, I have changes pending to make that automatic, removing the #defines from configuration.h, but have to wait for 32-bit merge to get that in.  

Realistically I would not choose to use the 3-point leveling feature of UBL.  I use a 11x11 grid and let it auto-probe anytime I change the glass.  G29 P1 to probe, then G29 P3.14 to interpolate all the unprobeable points.  Much better than using 3-point.

Roxy-3D, 2017-07-24:
The 3-Point 'Leveling' or tilting of the mesh is included in UBL just for completeness.  3-Point was the original bed leveling.   So just from that perspective it should be included.  But if you are going to tilt the mesh, a 3x3 or 4x4 grid will give you results you can have more confidence in.  (That would be G29 J3)

oldmcg, 2017-07-24:
I don't think my G29 J3 for delta changes made it in before the freeze.

GibaHC, 2017-08-01:
I'am still validating and correcting the edges. But I found out that my effector tilt changes next to the edges. This problem lead to reading errors of the Allen key probe. In fact, all the mesh reading os the probe where shifted by the y probe offset in the wrong direction, comparing with a mesh entirely done by manually probing the bed with the nozzle and paper. 

Ex: with the nozzle at x 0 y 84 the probe is at x 0 y 65, and the z height it reads is set for the I 7 J 12 point instead of I 7 J 14 (15x15 mesh). Probing with the nozzle resulted in a bigger printable flat area in the middle.

closed 2018-02-06 by thinkyhead