MarlinFirmware/Marlin #6998: Jerks and missed steps

Redferne, 2017-06-08:
Just merged in bugfix-1.1.x tonight from 1.1.0 and finally got UBL working on my P802NR2.
However when I started printing I see very strange jerking in one corner, also when printing first layer diagonal fill there seems to be no de-acceleration at all. The printer reverses with loud bangs and eventually starts missing steps. I have no memory of having seen this behavior in 1.1.0 a month ago?

Attached configuration files:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1062545/Marlin.zip)

And video...
[Video with banging noise :-) ](https://youtu.be/1SWict6yP8E)

Can someone please help me understand how to fix this :sob:

bgort, 2017-06-08:
Grab the newest version (as of a few minutes ago) .. check #6693 for more.

Redferne, 2017-06-08:
Just my usual luck.. Should have checked the last commits before pulling my hair out.. Oh well, luckily there's a weekend coming up 😉

closed 2017-06-08 by Redferne