MarlinFirmware/Marlin #6999: error when compiling with mixing head

sniper567, 2017-06-09:
when compiling the firmware with the mixing head setting I seem to get an error
start error:
Arduino: 1.8.3 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\stepper.cpp: In static member function 'static void Stepper::isr()':

stepper.cpp:618: error: 'En_STEP_WRITE' was not declared in this scope

           if (counter_m[j] > 0) En_STEP_WRITE(j, !INVERT_E_STEP_PIN);

                                                                    ^

stepper.cpp:653: error: 'En_STEP_WRITE' was not declared in this scope

             En_STEP_WRITE(j, INVERT_E_STEP_PIN);

                                               ^

exit status 1
'En_STEP_WRITE' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
End error:

I know from traditional C++ the function has to be defined for it to be in the scope and this En_STEP_WRITE only appears in two spots but I did not see it specified? I'm still new at C++ so, I could be missing something

Bob-the-Kuhn, 2017-06-09:
Do you have EXTRUDERS set to one?  If not that would explain why you're seeing the compile error.

If that's not the problem then please ZIP up your config files & drop the ZIP onto your reply.

sniper567, 2017-06-09:
OOO OOO OOO so extruders are the hot end, not the number stepper
motors that drive filament? if you don't mind me asking is there a setting
that defines how meany stepper drivers there are for the driving of
filament or does that just come from the pins.h?

thanks

Bob-the-Kuhn, 2017-06-09:
It's defined in the MIXING_EXTRUDER section in configuration.h.

```
/**
 * "Mixing Extruder"
 *   - Adds a new code, M165, to set the current mix factors.
 *   - Extends the stepping routines to move multiple steppers in proportion to the mix.
 *   - Optional support for Repetier Firmware M163, M164, and virtual extruder.
 *   - This implementation supports only a single extruder.
 *   - Enable DIRECT_MIXING_IN_G1 for Pia Taubert's reference implementation
 */
//#define MIXING_EXTRUDER
#if ENABLED(MIXING_EXTRUDER)
  #define MIXING_STEPPERS 2        // Number of steppers in your mixing extruder
  #define MIXING_VIRTUAL_TOOLS 16  // Use the Virtual Tool method with M163 and M164
  //#define DIRECT_MIXING_IN_G1    // Allow ABCDHI mix factors in G1 movement commands
#endif
```

sniper567, 2017-06-09:
thanks that has solved my problems

psavva, 2017-06-17:
Please close this incident

closed 2017-06-18 by sniper567