MarlinFirmware/Marlin #7003: [Question] String translation

Kaibob2, 2017-06-09:
What exactly is the difference between "Initialize EEPROM"
https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/ultralcd.cpp#L2634

and "Restore failsafe"
https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/ultralcd.cpp#L2603

I'm asking because translation to german is a little tricky to not make it appear as the same function.

p3p, 2017-06-09:
Not sure if this makes it any easier to translate but the difference is that "Initialise EEPROM" loads and then saves the hardcoded defaults to the EEPROM, whereas "Restore failsafe" just loads the hardcoded defaults.

To be honest the naming isn't very good, they both perform the same function only differing in whether the EEPROM is updated.

p3p, 2017-06-09:
As an afterthought, perhaps "Initialise EEPROM" would be better described "Factory Reset" and "Restore failsafe" as "Load defaults",

Kaibob2, 2017-06-09:
Funny, i was thinking about what german words would fit best and when i looked up i saw your post.
My best ideas were: "Standardwerte laden" which is exactly "Load defaults"
and "Werkseinstellungen" which is exactly "Factory defaults"

I'm think it is absolutely necessary to change this wording in english too!

Tannoo, 2017-06-09:
MSG_STORE_EEPROM, lcd_store_settings);
MSG_LOAD_EEPROM, lcd_load_settings);
MSG_RESTORE_FAILSAFE, lcd_factory_settings);
MSG_INIT_EEPROM, PSTR("M502\nM500")); // Load factory then save.

Kaibob2, 2017-06-09:
@Tannoo Sure, but in fact the `MSG_INIT_EEPROM, PSTR("M502\nM500")); // Load factory then save.` seems to be for new users to have an easy fallback, so "Factory defaults" would fit perfect. I myself just click Load and Save if i feel like saving.
I mean `MSG_LOAD_EEPROM, lcd_load_settings);` and `MSG_RESTORE_FAILSAFE, lcd_factory_settings);` does exactly the same, doesn't it?

Kaibob2, 2017-06-09:
Oops, i think i'm wrong. The one comes from firmware and the other from EEPROM, right?

Tannoo, 2017-06-09:
>I mean MSG_LOAD_EEPROM, lcd_load_settings); and MSG_RESTORE_FAILSAFE, lcd_factory_settings); does exactly the same, doesn't it?

No.  lcd_load_settings, Loads from EEPROM.
lcd_factory_settings, loads from config files.

Tannoo, 2017-06-09:
>Oops, i think i'm wrong. The one comes from firmware and the other from EEPROM, right?

Yes.

Kaibob2, 2017-06-09:
Sorrryyy. My fault.

Tannoo, 2017-06-09:
>seems to be for new users to have an easy fallback

Yes. Many users forget to / don't know about restoring from factory. So, the Init gives them that option.
Not everyone wants to save after restoring from factory so, the `lcd_factory_settings` is still there.

Tannoo, 2017-06-09:
It is not an M code, but just a shortcut in the menu to initialize the EEPROM, especially when preparing for UBL.

Kaibob2, 2017-06-09:
To be honest i find this very confusing, and i'm absolutely not a noob. But ok, i will just translate it as it is in the language_en file.

p3p, 2017-06-09:
I think "Initialise EEPROM" is unclear, "Factory Reset" should have a translation in most languages as it is used in most commercial electronics.

Kaibob2, 2017-06-09:
@p3p absolutely!

Tannoo, 2017-06-09:
Wouldn't FACTORY RESET be misleading for new users?

Wouldn't that lead them to believe that all the configuration options would also be reset to before they changed anything?

Tannoo, 2017-06-09:
That is my only opposition to that. Otherwise, it could even be removed.

p3p, 2017-06-09:
The majority of marlin users don't compile their own firmware*, but It didn't occur to me that in a small number of cases they may think it could revert the firmware flashing, naming things is hard ^^, "Factory Reset" is literally the name for reverting a firmware to the state it was when flashed though.

Tannoo, 2017-06-09:
Of all the people that do have questions on this site, I'm sure there are five times that number that never find this site and have troubles.

Tannoo, 2017-06-09:
Change it to FACTORY RESET...what could go wrong? Seriously.... I'm not complaining.

p3p, 2017-06-09:
well when you say it like that it makes me think it will cause everyone's printers to explode, This is just us trying to clarify the meaning of some options to aid translation ^^

Tannoo, 2017-06-09:
LOFL!! Not my intention. Sorry.

Roxy-3D, 2017-06-09:
This might be too much text...   But I would know what it was trying to do if the message was:

"Restore Default Settings"

Kaibob2, 2017-06-09:
MSG_RESTORE_FAILSAFE should be "Factory settings" german: (Werkseinst.)
and
MSG_INIT_EEPROM should be "Default settings" german: (Standardeinst.)

Correct?

Kaibob2, 2017-06-09:
I would suggest it this way, as the settings in the firmware are more "factory" than the ones in the EEPROM. These can be altered, the ones in the firmware only when firmware is flashed.

p3p, 2017-06-09:
Both of these options load the default(firmware) settings, the only difference is MSG_RESTORE_FAILSAFE will not persist over a reset, this is why I differentiated calling MSG_RESTORE_FAILSAFE "Load defaults", and MSG_INIT_EEPROM "Factory reset", you need to try and convey that one is persistent

p3p, 2017-06-09:
Thinking of general users, could MSG_INIT_EEPROM be simplified to "Reset Memory", or "Clear Memory"?

Kaibob2, 2017-06-09:
I see, so the other way around
MSG_RESTORE_FAILSAFE should be "Load defaults" german: (Standardeinst.) <- default settings
and
MSG_INIT_EEPROM should be "Factory reset" german: (Zurücksetzen)<- Reset

Would it be an option to not translate these two and leave them as "Load defaults" and "Factory reset" in every language. In german there is no way to make it precise with only 20 characters. I guess there are serveral other languages which have the same problem.

Kaibob2, 2017-06-09:
Sorry, the pointer was on the Close button and the monitor went to sleep

closed 2017-07-26 by Kaibob2

BETLOG, 2018-04-11:
I also find the english menus easy to misunderstand.
Correct me if I am wrong, but this is essentially the meaning right?
Referring to the difference between eeprom (as "User settings") and firmware is probably the important distinction to be making for clarity...in all the languages.

MSG_STORE_EEPROM, lcd_store_settings);
 Save current settings to EEPROM
 "Save User Settings"

MSG_LOAD_EEPROM, lcd_load_settings);
 Load EEPROM over current settings.
 "Load User Settings"

MSG_RESTORE_FAILSAFE, lcd_factory_settings);
 Load firmware defaults over current settings.
 "Reload Firmware"

MSG_INIT_EEPROM, PSTR("M502\nM500")); // Load factory then save. A.K.A., Initialize
 Load firmware defaults and save to EEPROM
 "Save Firmware ->User Settings"

thinkyhead, 2018-04-11:
Yes, those descriptions are correct.