MarlinFirmware/Marlin #7006: U8glib compilation warnings

Kaibob2, 2017-06-09:
I upgraded from RCBugFix to 1.1.3 Bugfix and the Arduino IDE 1.8.3 with U8glib 1.19.1 throws like a million U8glib warnings during compilation. Is there a way to fix this? In the end compilation finishes and uploads fine.
Most warnings are like
`warning: unused parameter 'u8g' [-Wunused-parameter]`
[1.1.3 U8glib errors.txt](https://github.com/MarlinFirmware/Marlin/files/1064084/1.1.3.U8glib.errors.txt)

Roxy-3D, 2017-06-11:
Maybe the right answer is to ask the U8glib people to make the warnings go away?   It makes me very uncomfortable to always have U8glib warning about stuff.    Perhaps there is a way to post an 'Issue' on their GitHub page?

Andrei-Pozolotin, 2017-06-11:
@Roxy-3D can you please show "marlin-safe" correction snippet so I can PR this to U8glib? thanks.

Roxy-3D, 2017-06-11:
@Andrei-Pozolotin    I don't understand what you mean by Marlin-safe correction snippet.   

If what you mean is can I provide the corrections to get rid of the U8glib warnings, the answer is "No!"   I don't fully understand why it is complaining.     But it is very possible that showing them the error messages will help them understand what we are seeing.

olikraus, 2017-06-13:
The U8glib warnings are related to the system dependent callback function. Usually they can be ignored. Of course it probably would be better to fix this like suggested here:

https://stackoverflow.com/questions/3417837/what-is-the-best-way-to-supress-unused-variable-x-warning

In fact this problem has been fixed for u8g2, but I did not ported the solution back to u8glib.

What is the migration status for u8g2? @thinkyhead mentioned (#6965) he talks to the u8g2 people (which should be me i guess), but at least i am not aware about this.

Roxy-3D, 2017-06-13:
> What is the migration status for u8g2?

@olikraus   I don't know that anybody is looking at this.

olikraus, 2017-06-13:
Well, then... I am ready to listen. Let me know if I could do anything to support the integration of u8g2.

Roxy-3D, 2017-06-13:
Right now...  Our two biggest issues with the graphics controllers is they eat up a lot of CPU cycles and they use a lot of RAM memory.   

If you can warm over the current ultralcd.cpp code to use the u8g2 library and not significantly hurt us on the CPU cycle issue or the RAM issue...   we would love to move to the current code base.

We have been putting special characters (like thermometers and degree symbols) into PROGMEM and doing things like that to save on RAM usage.

closed 2018-02-06 by thinkyhead