MarlinFirmware/Marlin #7015: Use of inline functions cause compile to fail when enabling UBL

fiveangle, 2017-06-10:
Attempting to transition from MBL to UBL and can't get passed some compile errors...

```
"/Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/../tools/avr/bin/avr-gcc" -Os -Wl,--gc-sections,--relax  -mmcu=at90usb1286 -o "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/Marlin.ino.elf" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/G26_Mesh_Validation_Tool.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/I2CPositionEncoder.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/M100_Free_Mem_Chk.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/Marlin.ino.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/MarlinSerial.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/Marlin_main.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/Sd2Card.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/SdBaseFile.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/SdFatUtil.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/SdFile.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/SdVolume.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/blinkm.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/cardreader.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/configuration_store.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/dac_mcp4728.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/digipot_mcp4018.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/digipot_mcp4451.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/endstops.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/gcode.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/hex_print_routines.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/least_squares_fit.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/mesh_bed_leveling.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/nozzle.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/planner.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/planner_bezier.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/printcounter.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/qr_solve.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/serial.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/servo.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/stepper.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/stepper_dac.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/stepper_indirection.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/stopwatch.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/temperature.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/twibus.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/ubl.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/ubl_G29.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/ubl_motion.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/ultralcd.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/utility.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/vector_3.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/watchdog.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/libraries/Wire/Wire.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/libraries/Wire/WireKinetis.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/libraries/Wire/utility/twi.c.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/../arduino_cache_804052/core/core_teensy_avr_teensypp2_usb_serial,speed_16,keys_en-us_54b41c7cca20326e490399ec92de3489.a" "-L/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421" -lm
/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/G26_Mesh_Validation_Tool.cpp.o: In function `unified_bed_leveling::G26()':
G26_Mesh_Validation_Tool.cpp:(.text._ZN20unified_bed_leveling3G26Ev+0xb0): undefined reference to `set_current_to_destination()'
G26_Mesh_Validation_Tool.cpp:(.text._ZN20unified_bed_leveling3G26Ev+0xca): undefined reference to `sync_plan_position_e()'
/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_967421/sketch/ubl_motion.cpp.o: In function `unified_bed_leveling::line_to_destination_cartesian(float const&, unsigned char)':
ubl_motion.cpp:(.text._ZN20unified_bed_leveling29line_to_destination_cartesianERKfh+0x1044): undefined reference to `set_current_to_destination()'
collect2: error: ld returned 1 exit status
Using library Wire at version 1.0 in folder: /Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/avr/libraries/Wire 
Error compiling for board Teensy++ 2.0.
```

Aruduinio IDE 1.8.2 with Teensy 1.3.4 extensions compiling for Printrboard RevF

[Configuration.h](https://github.com/fiveangle/Marlin/blob/ubl_compile_err/Marlin/Configuration.h)
[Configuration_adv.h](https://github.com/fiveangle/Marlin/blob/ubl_compile_err/Marlin/Configuration_adv.h)

Removing _inline_ declaration from both `sync_plan_position_e()` and `set_current_to_destination()` definitions in Marlin_main.cpp to allows to compile without error, so I think someone needs to look at how these are being used in G26_Mesh_Validation_Tool.cpp and ubl_motion.cpp

Roxy-3D, 2017-06-10:
That is very strange.   I believe you!  But I've been compiling with different configurations and haven't seen that.   I pretty much always have both UBL and G26 turned on.    

And you are using bugfix-v1.1.x, right?   That is pretty much the only branch I use.

fiveangle, 2017-06-10:
Since the recent changes made to allow UBL without requiring LCD, I figured I could test and make sure it was working for everyone, since lifting the LCD requirement will probably see an influx of 128k board users looking to try it out.

I don't pretend to understand why it's happening :)

https://github.com/fiveangle/Marlin/commits/ubl_compile_err

[image]

fiveangle, 2017-06-10:
Reverting the MACROization doesn't change anyway, so it's not some macro weirdness.  I'm thinking more related to linker behavior similar to [this](https://stackoverflow.com/questions/4769479/c-inlining-class-methods-causes-undefined-reference) since these are definied in Marlin_main.cpp and not in any header ?  It is very peculiar how you're not seeing this issue.

Bob-the-Kuhn, 2017-06-10:
I was able to reproduce Dave's error and work around.

I also tried putting local copies of the two into the problem files.  It compiled but used 36 more bytes than just deleting "inline".
`inline void sync_plan_position_e() { planner.set_e_position_mm(current_position[E_AXIS]); }`
`inline void set_current_to_destination() { COPY(current_position, destination); }`

When I changed to BOARD_RAMPS_14_EFB the without "inline" saved 4 bytes.  There was not a compiler error with the "inline" in place.

To me it looks like the "inline" qualifier doesn't buy us anything.  My vote would be to just drop the "inline" qualifier.

fiveangle, 2017-06-10:
If this is just a matter of re-arranging function delcarations so the linker is happy, it probably bears a few minutes to investigate before giving up (by someone more familiar with it than I ;)

Why is it that I only seem to find the mysterious and difficult to unwind toolchain issues ? :)

fiveangle, 2017-06-11:
@Bob-the-Kuhn you may be right.  Might not be worth it, although I don't know of the potential performance implications:

```
3PT: +6b P, 0b S
  Inline:
    Sketch uses 80654 bytes (62%) of program storage space. Maximum is 130048 bytes.
    Global variables use 3105 bytes (37%) of dynamic memory, leaving 5087 bytes for local variables. Maximum is 8192 bytes.
  Not: 
    Sketch uses 80660 bytes (62%) of program storage space. Maximum is 130048 bytes.
    Global variables use 3105 bytes (37%) of dynamic memory, leaving 5087 bytes for local variables. Maximum is 8192 bytes.

BLL:  +6b P, 0b S
  Inline:
    Sketch uses 93076 bytes (71%) of program storage space. Maximum is 130048 bytes.
    Global variables use 3105 bytes (37%) of dynamic memory, leaving 5087 bytes for local variables. Maximum is 8192 bytes.
  Not:
    Sketch uses 93082 bytes (71%) of program storage space. Maximum is 130048 bytes.
    Global variables use 3105 bytes (37%) of dynamic memory, leaving 5087 bytes for local variables. Maximum is 8192 bytes.

BLB: +2b P, 0b S
  Inline:
    Sketch uses 84576 bytes (65%) of program storage space. Maximum is 130048 bytes.
    Global variables use 3259 bytes (39%) of dynamic memory, leaving 4933 bytes for local variables. Maximum is 8192 bytes.
  Not:
    Sketch uses 84578 bytes (65%) of program storage space. Maximum is 130048 bytes.
    Global variables use 3259 bytes (39%) of dynamic memory, leaving 4933 bytes for local variables. Maximum is 8192 bytes.

UBL:
  Inline:
    [Compile failure]
  Not (w/o G29):
    Sketch uses 113780 bytes (87%) of program storage space. Maximum is 130048 bytes.
    Global variables use 3559 bytes (43%) of dynamic memory, leaving 4633 bytes for local variables. Maximum is 8192 bytes.
  Not (w/G26):
    Sketch uses 123860 bytes (95%) of program storage space. Maximum is 130048 bytes.
    Global variables use 3717 bytes (45%) of dynamic memory, leaving 4475 bytes for local variables. Maximum is 8192 bytes.

MBL: +2b P, 0b S
  Inline:
    Sketch uses 81690 bytes (62%) of program storage space. Maximum is 130048 bytes.
    Global variables use 3229 bytes (39%) of dynamic memory, leaving 4963 bytes for local variables. Maximum is 8192 bytes.
  Not: 
    Sketch uses 81692 bytes (62%) of program storage space. Maximum is 130048 bytes.
  Global variables use 3229 bytes (39%) of dynamic memory, leaving 4963 bytes for local variables. Maximum is 8192 bytes.
```

Bob-the-Kuhn, 2017-06-11:
> Why is it that I only seem to find the mysterious and difficult to unwind toolchain issues ? :)

I think @Roxy-3D finds them and then maneuvers you into fixing them!  😉

Actually it's just another example of strange behavior by the Teensyduino IDE extension used by the AT90USB boards.

Roxy-3D, 2017-06-11:
Can you explain why an AT90USB board would handle inline functions differently that the AtMega boards?   I need help understanding that.

Can we move these two declarations (set_current_to_destination() and sync_plan_position_e()) to Marlin.h to resolve the issue?

Bob-the-Kuhn, 2017-06-11:
Sorry - I don't know why it's a problem for the Teensyduino IDE.

I'm going to embroider some towels so it'll be 3-5 hours before I can do any testing.

Roxy-3D, 2017-06-11:
Some silly needle point on a towel is more important than Marlin????      Geeeeze.....

Bob-the-Kuhn, 2017-06-11:
It's machine embroidery - an early form of 3D printing,  It's based on a distant fork of Marlin.

Bob-the-Kuhn, 2017-06-14:
I finally had some time to try putting set_current_to_destination() and sync_plan_position_e()) into Marlin.h.

I gave up after several hours.  Kept running into definition conflicts.  

@Roxy-3D - what do you want to do?
- delete inline
- make inline conditional compile
- put local copies into the files
- something else

Roxy-3D, 2017-06-14:
Well...   I don't know that inline has much value for G29 and G26.   Nothing in those functions is going to benefit from saving a few micro-seconds.    But set_current_to_destination() is used by a whole bunch of stuff.   So probably the safest thing to do is just make local copies of those functions inside of G26 and G29.

Bob-the-Kuhn, 2017-06-15:
I've created PR #7062 to fix this issue and another (not related to AT90USB) that popped up in tonight's testing.

FYI - the `set_current_to_destination()` problem seems to have disappeared.  I can't find any changes that would have affected it.

fiveangle, 2017-06-24:
Appears it's still present in my config:

```
Linking everything together...
"/Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/../tools/avr/bin/avr-gcc" -Os -Wl,--gc-sections,--relax  -mmcu=at90usb1286 -o "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/Marlin.ino.elf" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/G26_Mesh_Validation_Tool.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/I2CPositionEncoder.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/M100_Free_Mem_Chk.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/Marlin.ino.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/MarlinSerial.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/Marlin_main.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/Sd2Card.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/SdBaseFile.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/SdFatUtil.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/SdFile.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/SdVolume.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/blinkm.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/cardreader.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/configuration_store.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/dac_mcp4728.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/digipot_mcp4018.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/digipot_mcp4451.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/endstops.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/gcode.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/hex_print_routines.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/least_squares_fit.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/mesh_bed_leveling.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/nozzle.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/pca9632.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/planner.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/planner_bezier.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/printcounter.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/qr_solve.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/serial.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/servo.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/stepper.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/stepper_dac.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/stepper_indirection.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/stopwatch.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/temperature.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/twibus.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/ubl.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/ubl_G29.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/ubl_motion.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/ultralcd.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/utility.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/vector_3.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/watchdog.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/libraries/Wire/Wire.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/libraries/Wire/WireKinetis.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/libraries/Wire/utility/twi.c.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/../arduino_cache_735640/core/core_teensy_avr_teensypp2_usb_serial,speed_16,keys_en-us_54b41c7cca20326e490399ec92de3489.a" "-L/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629" -lm
/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/G26_Mesh_Validation_Tool.cpp.o: In function `unified_bed_leveling::G26()':
G26_Mesh_Validation_Tool.cpp:(.text._ZN20unified_bed_leveling3G26Ev+0xae): undefined reference to `set_current_to_destination()'
/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_161629/sketch/ubl_motion.cpp.o: In function `unified_bed_leveling::line_to_destination_cartesian(float const&, unsigned char)':
ubl_motion.cpp:(.text._ZN20unified_bed_leveling29line_to_destination_cartesianERKfh+0xf6a): undefined reference to `set_current_to_destination()'
collect2: error: ld returned 1 exit status
Using library Wire at version 1.0 in folder: /Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/avr/libraries/Wire 
Error compiling for board Teensy++ 2.0.
```

This was with current rcbugfix-1.1.0 as of a few min ago.  @Bob-the-Kuhn - what's different between yours and mine ?

fiveangle, 2017-07-01:
FYI, error still remains... as of https://github.com/MarlinFirmware/Marlin/commits/457489d7acf59db2fe324de3a9aac0e66df49f2e

```
Linking everything together...
"/Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/../tools/avr/bin/avr-gcc" -Os -Wl,--gc-sections,--relax  -mmcu=at90usb1286 -o "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/Marlin.ino.elf" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/G26_Mesh_Validation_Tool.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/I2CPositionEncoder.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/M100_Free_Mem_Chk.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/Marlin.ino.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/MarlinSerial.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/Marlin_main.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/Sd2Card.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/SdBaseFile.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/SdFatUtil.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/SdFile.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/SdVolume.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/blinkm.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/cardreader.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/configuration_store.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/dac_mcp4728.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/digipot_mcp4018.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/digipot_mcp4451.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/endstops.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/gcode.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/hex_print_routines.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/least_squares_fit.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/mesh_bed_leveling.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/nozzle.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/pca9632.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/planner.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/planner_bezier.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/printcounter.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/qr_solve.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/serial.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/servo.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/stepper.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/stepper_dac.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/stepper_indirection.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/stopwatch.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/temperature.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/twibus.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/ubl.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/ubl_G29.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/ubl_motion.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/ultralcd.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/utility.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/vector_3.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/watchdog.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/libraries/Wire/Wire.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/libraries/Wire/WireKinetis.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/libraries/Wire/utility/twi.c.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/core/core.a" "-L/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559" -lm
/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/G26_Mesh_Validation_Tool.cpp.o: In function `unified_bed_leveling::G26()':
G26_Mesh_Validation_Tool.cpp:(.text._ZN20unified_bed_leveling3G26Ev+0xae): undefined reference to `set_current_to_destination()'
/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_612559/sketch/ubl_motion.cpp.o: In function `unified_bed_leveling::line_to_destination_cartesian(float const&, unsigned char)':
[... cut ...] fined reference to `set_current_to_destination()'
collect2: error: ld returned 1 exit status
Using library Wire at version 1.0 in folder: /Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/avr/libraries/Wire 
Error compiling for board Teensy++ 2.0.
```

Any user can run UBL with Teensy if they do not have SD and LCD, so it is theoretically valid, but probably not a lot of users will use it, so if you want, I can just close this if we don't think it's important enough to pursue.

Bob-the-Kuhn, 2017-07-01:
Dave - I just downloaded bugfix-1.1.x, enabled TEENSYLU, EEPROM & UBL.  It compiled for me.

This issue was verified early in this thread, disappeared, came back for you & now (for me) it's OK.

I have no idea why.

I'll submit a PR & see what Scott says.

Roxy-3D, 2017-07-01:
It will be interesting to see what happens when the new folder and file organization shows up tomorrow.  

I wonder if TEENSYLU, EEPROM & UBL will cause problems there???

fiveangle, 2017-07-01:
> I wonder if TEENSYLU, EEPROM & UBL will cause problems there???

I have a handful of small simple PRs about ready to submit, but figured since they will need massaging to port into the new layout so I've held off submitting (various cleanup Printrboard_RevF pins file, add support for MINIPANEL to it, etc) and will test them all with the new layout once complete.  I can certainly do it with this one too, but I have a feeling the new layout will cause more things to break than to be automagically fixed :)

@Bob-the-Kuhn - you said you have an actual Printrboard, or ?  What exact AT90USB board do you have ?

Bob-the-Kuhn, 2017-07-01:
I have a Teensylu.

I'm hoping that Scott will be able to quickly find (or point us to) the root cause.

Agreed - I expect it'll be a week or two squashing unintended bugs with the reorg and then items like this can be addressed.  

I have no problems with you including this fix in with your PRs.

fiveangle, 2017-07-01:
@Bob-the-Kuhn - ubl_motion.cpp has the same issue, so probably best to expand #7201 to encompass it there too, so when Scott takes a look he has a fuller picture.

BTW- got the cable today and all I can say is we need to dub you the master of cyanoacrylate - I'd never considered using it in this way.  Brilliant ! :)

Bob-the-Kuhn, 2017-07-01:
I've added it to the PR.

----

I have a BIG bottle of superglue.  And a bottle of instant set.  And some parchment paper. And gunked up finger tips.

One thing I have not figured out is how to add a nice looking strain relief so the cable doesn't continue to split.

fiveangle, 2017-07-07:
looks like your PR worked around the previous 2 inline compile errors.  i'll open a new issue for the new undefined reference to ubl_lcd_map_control error...

```
"/Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/../tools/avr/bin/avr-ar" rcs "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/core/core.a" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/core/yield.cpp.o"
Archiving built core (caching) in: /var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_cache_288060/core/core_teensy_avr_teensypp2_usb_serial,speed_16,keys_en-us_54b41c7cca20326e490399ec92de3489.a
Linking everything together...
"/Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/../tools/avr/bin/avr-gcc" -Os -Wl,--gc-sections,--relax  -mmcu=at90usb1286 -o "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/Marlin.ino.elf" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/G26_Mesh_Validation_Tool.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/I2CPositionEncoder.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/M100_Free_Mem_Chk.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/Marlin.ino.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/MarlinSerial.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/Marlin_main.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/Sd2Card.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/SdBaseFile.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/SdFatUtil.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/SdFile.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/SdVolume.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/blinkm.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/cardreader.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/configuration_store.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/dac_mcp4728.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/digipot_mcp4018.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/digipot_mcp4451.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/endstops.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/gcode.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/hex_print_routines.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/least_squares_fit.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/mesh_bed_leveling.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/nozzle.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/pca9632.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/planner.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/planner_bezier.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/printcounter.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/qr_solve.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/serial.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/servo.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/stepper.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/stepper_dac.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/stepper_indirection.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/stopwatch.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/temperature.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/twibus.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/ubl.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/ubl_G29.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/ubl_motion.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/ultralcd.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/utility.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/vector_3.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/watchdog.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/libraries/Wire/Wire.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/libraries/Wire/WireKinetis.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/libraries/Wire/utility/twi.c.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/core/core.a" "-L/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324" -lm
/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/stepper.cpp.o: In function `Stepper::quick_stop()':
stepper.cpp:(.text._ZN7Stepper10quick_stopEv+0x2): undefined reference to `ubl_lcd_map_control'
collect2: error: ld returned 1 exit status
Using library Wire at version 1.0 in folder: /Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/avr/libraries/Wire 
Error compiling for board Teensy++ 2.0.
```

closed 2017-07-07 by fiveangle

Roxy-3D, 2017-07-08:
Do we need to change the 128KB entries in Arduino?  Something is configured differently compared to the ATMega entries, right?

Bob-the-Kuhn, 2017-07-08:
I haven't the faintest idea what is going on.

Roxy-3D, 2017-07-08:
The thing is...  Something is going on in UBL that is annoying the compiler.   I don't know what is different, but all of these missing function linker errors are UBL stuff.

JJJpnt, 2017-07-14:
Hi, 
I have the exact same issue compiling 1.1.4
I'm quite not very comfortable with deep inside marlin understanding and even with github...
My board is a MKS_BASE, if that could help or something...
Waiting for you wizards to do your magic
Thanks for all the great stuff

Roxy-3D, 2017-07-14:
> MKS Base V1.0 has the powerful processor ATMEGA2560 WITH 256K flash memory. It use FTDI FT232R device for USB and not ATMEGA16U2 as ARDUINO MEGA2560. In power input, we use L5970D replacing the LM1117-5.0, which make MKS Base V1.0 supporting 24V or 12V input. And XH series sockets let us easy inset the wire of step motor and end stop.

MKS Base is an Atmega-2560 based board.   It should not have any the issues that the 128KB processors have.    Can you add a .txt to the end of your Configuration. and Configuration_adv.h file and post them here?   I'll take a look and see if I can figure out why the compilation is failing.

JJJpnt, 2017-07-14:
Well... I updated arduino IDE to 1.8.3 (I had forgotten that I downgraded some time ago for some reason), and it apparently just compiled ok...
Should have checked that before sorry for the inconvenience. And thanks for the very quick reply btw.

fiveangle, 2017-07-15:
Are the compiler directives the same  or are the    AT90USB  getting something else past to the linker ?

Roxy-3D, 2017-07-15:
I'm sure somebody will jump in and correct what I say...  But each board (or platform) in Arduino has its own configuration and parameter file.   That setup is used to set compile time options and tells the linker how to locate things.   

The 128KB processors need to be linked differently than the 256KB processors (even though much of the code is compiled the same way).   

My first printer had a PrintRboard on it...  They still tell people to use Arduino v.22 to compile code for it.  I finally pulled the plug on them.   Right now I have all Atmega-2560 boards with RAMPS shields.   Soon, I'll have mostly Re-ARM boards with RAMPS shields.

fiveangle, 2017-07-15:
We all know how much you hate PrintrbotCo @Roxy-3D, that wasn't in question — only the linker differences between the AT90USB and Atmega2560 was 😉