MarlinFirmware/Marlin #7021: repeated G29 T1 (or G29 T0) gives different results...

lrpirlet, 2017-06-10:
**Background**

I am using the latest bugfix-1.1.x (0b6b7c9) to build my firmware for my Prusa i3 printer... 

Configuration.h is the only customized file and is accessible from https://github.com/lrpirlet/Marlin/tree/lrp-2-jun-2017 under https://github.com/lrpirlet/Marlin/blob/lrp-2-jun-2017/Marlin/Configuration.h

I issue an M133 S30 to avoid "echo:busy: processing" in the middle of mu disered output...
I Issue G28, G29 P0 followed by G29 P1...
I issue G29 W to log the G29 characteristics...
All the above without any problem...

**Issue**
I send G29 T1 several time and got different results... or rather I DO MISS some values at DIFFERENT places in each of the outputs.

extracted from https://github.com/lrpirlet/Marlin/blob/lrp-2-jun-2017/logs/Repetier_log.txt please find here the evidence for G29 T1... G29 T0 also missbehave...
```C++
14:49:39.459 : N21 G29T1*68
14:49:39.679 : ,0.148
14:49:39.819 : 0.030,0.163,0.201,0.249,0.276,0.251,0.274,0.062,0.118
14:49:39.991 : -0.015,0.126,0.175,0.158,0.228,0.230,0.272,0.198,0.148
14:49:40.147 : -0.093,-0.031,0.144,0.213,0.223,0.263,0.268,0.232,0.116
14:49:40.303 : -0.117,-0.013,0.120,0.194,0.236,0.230,0.269,0.233,0.143
14:49:40.428 : -0.124,-0.044,0.097,0.162,0.229,0.246,0.270,0.165,0.190
14:49:40.569 : -0.086,0.063,0.141,0.208,0.226,0.255,0.272,0.214,0.214
14:49:40.709 : ,0.248,0.244,0.212
14:49:40.850 : 0.004,-0.042,0.194,0.218,0.238,0.290,0.297,0.224,0.204
14:49:40.865 : ok
14:49:51.766 : N22 G29T1*71
14:49:51.970 : 0.048,0.189,0.227,0.252,0.220,0.228,0.229,0.180,0.148
14:49:52.110 : 0.030,0.163,0.201,0.249,0.276,0.251,0.274,0.062,0.118
14:49:52.252 : -0.015,0.126,0.175,0.158,0.228,0.230,0.272,0.198,0.148
14:49:52.393 : -0.093,-0.031,0.144,0.213,0.223,0.263,0.268,0.232,0.116
14:49:52.580 : -0.117,-0.013,0.120,0.194,0.236,0.230,0.269,0.233,0.143
14:49:52.735 : ,0.246,0.270,0.165,0.190
14:49:52.876 : -0.086,0.063,0.141,0.208,0.226,0.255,0.272,0.214,0.214
14:49:53.017 : -0.067,-0.034,0.177,0.215,0.233,0.284,0.248,0.244,0.212
14:49:53.157 : 0.004,-0.042,0.194,0.218,0.238,0.290,0.297,0.224,0.204
14:49:53.173 : ok
14:49:56.896 : N23 G29T1*70
14:49:57.083 : 0.048,0.189,0.227,0.252,0.220,0.228,0.229,0.180,0.148
14:49:57.254 : 0.030,0.163,0.201,0.249,0.276,0.251,0.274,0.062,0.118
14:49:57.380 : -0.015,0.126,0.175,0.158,0.228,0.230,0.272,0.198,0.148
14:49:57.583 : -0.093,-0.031,0.144,0.213,0.223,0.263,0.268,0.232,0.116
14:49:57.723 : ,0.269,0.233,0.143
14:49:57.864 : -0.124,-0.044,0.097,0.162,0.229,0.246,0.270,0.165,0.190
14:49:58.005 : -0.086,0.063,0.141,0.208,0.226,0.255,0.272,0.214,0.214
14:49:58.145 : -0.067,-0.034,0.177,0.215,0.233,0.284,0.248,0.244,0.212
14:49:58.294 : 0.004,-0.042,0.194,0.218,0.238,0.290,0.297,0.224,0.204
14:49:58.306 : ok
14:50:06.686 : N24 G29T1*65
14:50:06.890 : 0.048,0.189,0.227,0.252,0.220,0.228,0.229,0.180,0.148
14:50:07.029 : 0.030,0.163,0.201,0.249,0.276,0.251,0.274,0.062,0.118
14:50:07.170 : -0.015,0.126,0.175,0.158,0.228,0.230,0.272,0.198,0.148
14:50:07.311 : -0.093,-0.031,0.144,0.213,0.223,0.263,0.268,0.232,0.116
14:50:07.452 : -0.117,-0.013,0.120,0.194,0.236,0.230,0.269,0.233,0.143
14:50:07.593 : -0.124,-0.044,0.097,0.162,0.229,0.246,0.270,0.165,0.190
14:50:07.733 : ,0.214,0.214
14:50:07.874 : -0.067,-0.034,0.177,0.215,0.233,0.284,0.248,0.244,0.212
14:50:08.016 : 0.004,-0.042,0.194,0.218,0.238,0.290,0.297,0.224,0.204
14:50:08.032 : ok
```

**Questions**

Does anyone else see that behavior???

Since the difference between the 3 commands G29 T1 is just a different time, could we see some interrupt misshandling... I am completely at lost, and would appreciate some help... 

**Last but not least,**

Thanks in advance.

Roxy-3D, 2017-06-10:
@lrpirlet      I don't know what a M133 command is.   But the G29 W in the second link says you have a 9 x 9 mesh.   I looked at the various G29 T commands you gave it.    I do not think the mesh is changing.   I think data is getting lost as it is being transmitted to Repetier Host.     

I use PronterFace and do not see a problem.   There are a couple of things we can try to get the map to print out better and not lose data.     First you can try changing in Configuration_adv.h to have
```cpp
#define TX_BUFFER_SIZE 32
```
because that will cause 
```cpp
  void unified_bed_leveling::display_map(const int map_type) {
...
        #if TX_BUFFER_SIZE > 0
          MYSERIAL.flushTX();
        #endif
        safe_delay(15);
```
The other thing you can do is change that safe_delay(15) to safe_delay(50)  in the display_map() function to slow the map generation down enough you don't lose data.

Please let me know what the results are.   OK?

lrpirlet, 2017-06-10:
Hi @Roxy-3D, **thanks a lot**... Your point of view triggered another approach... 

I used the Arduino imbedded tool called serial monitor (well in french, that tool is named "moniteur série")...

I get good data, see bellow... (Accessorily, M133 S<seconds> sets the time interval for the keep alive message, setting 30 seconds does clean up the output from the many _echo:busy: processing_ issued every 2 seconds by default... Note that Repetier timeout is 40 seconds so 30 second is quite enough to keep Repetier happy...)

So what is different between passive Arduino serial monitor and Repetier? 

- Repetier issues M115 every 3 seconds
- Repetier processes every received message
- More???

The baud rate I use is 115200, the serial monitor seems slower than that (I can see the line progressing) meaning that there is some sort of hand checking possible between the printer and the host...
Next action will be to read the Repetier doc and play with the safe_delay in Marlin...

I doubt that Marlin is buggy, I think it may be over-optimized for Repetier...

There is no echo in the Arduino serial monitor, find the command I typed in front of the result...

I'll keep reporting when more results available... again, thanks.

G28
```
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
X:0.00 Y:0.00 Z:7.79 E:0.00 Count X:0 Y:0 Z:31160
ok
```
M113 S30
```
ok
```
G28
```
X:0.00 Y:0.00 Z:7.79 E:0.00 Count X:0 Y:0 Z:31160
ok
```
G29 W
```
Unified Bed Leveling System v1.01 inactive.
No Mesh Loaded.
UBL object count: 1
planner.z_fade_height : 10.0000
zprobe_zoffset: -4.7899999
UBL_MESH_MIN_X  (0 + 1)=1
UBL_MESH_MIN_Y  (0 + 1)=1
UBL_MESH_MAX_X  (180 - (1))=179
UBL_MESH_MAX_Y  (180 - (1))=179
GRID_MAX_POINTS_X  9
GRID_MAX_POINTS_Y  9
MESH_X_DIST  22.25
MESH_Y_DIST  22.25
X-Axis Mesh Points at: 1.000  23.250  45.500  67.750  90.000  112.250  134.500  156.750  179.000  
Y-Axis Mesh Points at: 1.000  23.250  45.500  67.750  90.000  112.250  134.500  156.750  179.000  
Kill pin on :41  state:1

ubl_state_at_invocation :0

ubl_state_recursion_chk :0

Meshes go from 0x0248 to 0x0F7F
sizeof(ubl) :  1

z_value[][] size: 324

EEPROM free for UBL: 0x0D37
EEPROM can hold 10 meshes.

Unified Bed Leveling sanity checks passed.
ok
```
G29 P1
```
Default storage slot 0 selected.
Mesh invalidated. Probing mesh.
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
ok
```
G29 T0
```
Bed Topography Report:

(0,8)                                                           (8,8)
(1,179)                                                        (179,179)
  0.025    0.161    0.188    0.229    0.219    0.223    0.213    0.179    0.168  
 
 -0.014    0.126    0.166    0.236    0.256    0.208    0.281    0.119    0.069  
 
 -0.059    0.097    0.132    0.093    0.157    0.218    0.251    0.210    0.058  
 
 -0.106   -0.046    0.111    0.154    0.196    0.195    0.229    0.194   -0.012  
 
 -0.089   -0.042    0.078    0.145    0.209    0.229    0.252    0.214    0.147  
 
 -0.098    0.027    0.077    0.175    0.210    0.228    0.242    0.223    0.160  
 
 -0.110   -0.025    0.057    0.173    0.190    0.244    0.243    0.148    0.172  
 
 -0.036    0.003    0.159    0.204    0.207    0.260    0.268    0.183    0.183  
 
[ 0.007]  -0.011    0.167    0.220    0.224    0.255    0.257    0.184    0.174  
(1,1)                                                            (179,1)
(0,0)                                                             (8,0)
ok
```
G29 T1
```
0.025,0.161,0.188,0.229,0.219,0.223,0.213,0.179,0.168
-0.014,0.126,0.166,0.236,0.256,0.208,0.281,0.119,0.069
-0.059,0.097,0.132,0.093,0.157,0.218,0.251,0.210,0.058
-0.106,-0.046,0.111,0.154,0.196,0.195,0.229,0.194,-0.012
-0.089,-0.042,0.078,0.145,0.209,0.229,0.252,0.214,0.147
-0.098,0.027,0.077,0.175,0.210,0.228,0.242,0.223,0.160
-0.110,-0.025,0.057,0.173,0.190,0.244,0.243,0.148,0.172
-0.036,0.003,0.159,0.204,0.207,0.260,0.268,0.183,0.183
0.007,-0.011,0.167,0.220,0.224,0.255,0.257,0.184,0.174
ok
```

Roxy-3D, 2017-06-11:
Good!  You can get a nice looking Topology Map now!    Now you can see what things look like.  Your bed is very good!  But there is a slight 'fold' in it at the center (X-Axis).   Front to back is about as good as you can hope for.

You don't need to do this...  But you can  mess with #define PROPORTIONAL_FONT_RATIO 1.5 in Configuration_adv.h to push those columns titles over.   

Have you done G26 so you know which mesh points still need tweaking?

lrpirlet, 2017-06-12:
@Roxy-3D 
Ok, I think I understand what is going on... 
I have autotemp enabled
Looking at the Repetier log shows that Marlin sends a temperature report every second... this DOES interfere with the output of the command G29 T

@repetier 
I have no idea if what I see is intended or not. I have at the very begin (during what I call the initial setup) a M155 S1 command so marlin generates a temp report every seconds... that is NOT coherent with the printer parameter where a temperature report should be queried every 3 seconds ( or M105 should be issued every 3 seconds)

**Confirmation**
I did issue M155 S5... I could get a clean G29 T0 about every time, but NOT ALWAYS...

**MY interpretation**
It seems that G29 T0 is preventing M105 command to be executed... so without autotemps we serialize the commands output...
NOW, IF autotemp report is enabled, then chances are that we get temp report in the middle of a long output such as the G29 T output...

**Proposed resolution**
Would it be difficult to suspend AUTOTEMP while long output are generated?? Sorry, my skills with C and/or C++ are such that I may come (or not) with an implementation in about 2-3 months...

repetier, 2017-06-13:
Yes, 1s temperature report was intendet. 3s is more because it adds extra traffic on sending commands and with autotemp there is no interrupt of normal command flow any more.

From host side it should be no problem to suspend report - it just gets no updates to show but that was the same before as G29 would have blocked M105 the same way.

I guess you just cache the autotemp flag, disable it and at the end of G29 set it back to cached value.

lrpirlet, 2017-06-13:
@Roxy-3D 
Thanks for the topology map... I have not tried G26 yet but will soon after I re-calibrate speed, acceleration and jerk... 

@repetier 
Ok, thanks, this would NOT have been MY choice...

I use my 3D printer to print N-scale object (N-scale is 1:160), so I need high accuracy in both the cartesian axis and the extruder axis... That implies, for me, using both mesh levelling (UBL is great) AND LIN-Advance... The 8 bit controller is limit in terms of speed... 

To increase the traffic, without GUI control, is IMHO, a step backward...
I guess I need to edit my startup script to include a M155 S3... or market around to find a program that gives me control AND do not make things under my feet with a GUI that is only a gimmick (that is how I feel)...

AGAIN, please no offences, I just am somewhat frustrated because I feel I do not have full control over the host software... On the other hand, I like the repetier host I chose because of the integration of the slicers programs...

I carefully reduce as much as I can the interrupts in the controller by reducing event output occurrences (M113 S30 is an example)...
I find this AUTOTEMP replying 30 characters every second or 90 characters every 3 seconds instead of sending about 7 characters( M105*<checksum>) plus 30 characters answer... that makes 90 VS 37 interrupts for absolutely no REAL benefit (ever see PHYSICAL temp of the printer changing significantly in ONE second?)

BTW, thanks a lot for pointing out that SENDING M105 is what synchronize the commands... I guess my proposed solution now become to queue an equivalent of M105 at every AUTOTEMP time... (ok, I still need to read and understand the code to verify if that approach makes sense...)

lrpirlet, 2017-06-15:
Ok, please find here a picture of what is going on...
I have AUTOTEMP enabled (configuration_adv.h)... 
This output a temperature report every M155 <seconds>...
Please note that this is **TOTALY ASYNCHRONEOUS** to the command flow... ANY command sending several lines of console output run the risk to be _CORRUPTED_ by this....

In my case, Repetier default activate a temperature report every seconds... So any console output taking more than one second will see an imbedded temp report at least once and sometime twice...

I have examined the code... It would be extremely costly in term of resource to submit a M105 command at every M155 <second>...
I guess the only acceptable solution would be to have an interlock over the console resource... I imagine that implemented with a bool over any console output that need protected
```C++
...
# define lock-console false
...
lock-console true;

lock-console false;

```
09:11:59.890 : start
09:11:59.890 : echo: External Reset
09:11:59.890 : Marlin bugfix-1.1.x
09:11:59.890 : echo: Last Updated: 2017-05-04 12:00 | Author: LRP
09:11:59.906 : Compiled: Jun 10 2017
09:11:59.906 : echo: Free Memory: 2765  PlannerBufferBytes: 1232
09:11:59.921 : echo:V38 stored settings retrieved (453 bytes; crc -15873)
09:11:59.921 : Unified Bed Leveling System v1.01 inactive.
09:11:59.968 : Unified Bed Leveling initialized.
09:11:59.968 : UBL System reset()
09:11:59.968 : echo:  G21    ; Units in mm
09:11:59.968 : echo:  M149 C ; Units in Celsius
09:11:59.984 : echo:Filament settings: Disabled
09:11:59.984 : echo:  M200 D3.00
09:11:59.984 : echo:  M200 D0
09:11:59.984 : echo:Steps per unit:
09:11:59.984 : echo:  M92 X80.00 Y80.00 Z4000.00 E402.00
09:12:00.000 : echo:Maximum feedrates (units/s):
09:12:00.000 : echo:  M203 X300.00 Y300.00 Z3.00 E25.00
09:12:00.000 : echo:Maximum Acceleration (units/s2):
09:12:00.000 : echo:  M201 X1000 Y1000 Z50 E5000
09:12:00.015 : echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
09:12:00.015 : echo:  M204 P1500.00 R2000.00 T1500.00
09:12:00.015 : echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
09:12:00.031 : echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
09:12:00.031 : echo:Home offset:
09:12:00.031 : echo:  M206 X0.00 Y0.00 Z0.00
09:12:00.031 : N1 M110*34
09:12:00.031 : N2 M115*36
09:12:00.031 : N3 M105*36
09:12:00.031 : N4 M114*35
09:12:00.031 : N5 M111 S6*98
09:12:00.031 : echo:Unified Bed Leveling:
09:12:00.031 : echo:  M420 S0 Z10.00
09:12:00.031 : Unified Bed Leveling System v1.01 inactive.
09:12:00.093 : Active Mesh Slot: -1
09:12:00.093 : z_offset: 0.000000
09:12:00.093 : EEPROM can hold 10 meshes.
09:12:00.093 : echo:Material heatup parameters:
09:12:00.093 : echo:  M145 S0 H180 B70 F0
09:12:00.093 : M145 S1 H240 B110 F0
09:12:00.109 : echo:PID settings:
09:12:00.109 : echo:  M301 P21.90 I1.43 D83.96
09:12:00.109 : echo:Z-Probe Offset (mm):
09:12:00.109 : echo:  M851 Z-4.79
09:12:05.534 : echo:SD init fail
09:12:05.534 : ok
09:12:05.534 : N6 T0*60
09:12:05.566 : FIRMWARE_NAME:Marlin bugfix-1.1.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Pi3 3D LabteK EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
09:12:05.566 : Cap:EEPROM:1
09:12:05.566 : Cap:AUTOREPORT_TEMP:1
09:12:05.566 : Cap:PROGRESS:0
[... 161 lines cut ...]
09:13:26.784 : (1,179)                                                        (179,179)
09:13:26.987 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:26.987 : 
09:13:27.128 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:27.128 : 
09:13:27.333 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:27.333 : 
09:13:27.473 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:27.473 : 
09:13:27.614 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:27.614 : 
09:13:27.769 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:27.769 : 
09:13:27.910 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:27.910 : 
09:13:28.050 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:28.050 : 
09:13:28.191 : [ 0.000]   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:28.222 : (1,1)                                                            (179,1)
09:13:28.238 : (0,0)                                                             (8,0)
09:13:28.254 : ok
09:13:29.177 : T:21.95 /0 B:19.96 /0 @:0 B@:0
09:13:32.179 : T:21.87 /0 B:19.92 /0 @:0 B@:0
09:13:35.182 : T:21.95 /0 B:19.92 /0 @:0 B@:0
```

lrpirlet, 2017-06-15:
@Roxy-3D 
Ok, please find here a picture of what is going on...
I have AUTOTEMP enabled (configuration_adv.h)...
 
So Marlin outputs a temperature report every M155 <seconds>...
Please note that this is **TOTALY ASYNCHRONEOUS** to the command flow... ANY command sending several lines of console output run the risk to be _CORRUPTED_ by this...
The implementation of AUTOTEMP is at least incomplete...

In my case, Repetier default activates a temperature report every seconds... So any console output taking more than one second will see an imbedded temp report at least once and sometime twice... see below the log. 
Note that this is NOT TIED with Repetier...

I have examined the code... It would be extremely costly in term of resource to submit a M105 command at every M155 <second>...
I guess the only acceptable solution would be to have an interlock over the console resource... I imagine that implemented with a bool over any console output that needs be protected
```C++
...
# define lock-console false // in some .h file
...
// Then in the code
...
lock-console true;
// console print to be protected such as G29 T0 
lock-console false;
...
  inline void auto_report_temperatures() {
    if (auto_report_temp_interval && ELAPSED(millis(), next_temp_report_ms)) {
      next_temp_report_ms = millis() + 1000UL * auto_report_temp_interval;
      if !lock-console {
        print_heaterstates();
        SERIAL_EOL();
      }
    }
  }
```

What would be the risk???

To be honest, if that takes too much time or if it needs a more complex solution, I may revert and switch off the feature AUTOTEMP... I really fear intermittent, hard to trace failures... 

**The log from repetier**
```
09:11:59.890 : start
09:11:59.890 : echo: External Reset
09:11:59.890 : Marlin bugfix-1.1.x
09:11:59.890 : echo: Last Updated: 2017-05-04 12:00 | Author: LRP
09:11:59.906 : Compiled: Jun 10 2017
09:11:59.906 : echo: Free Memory: 2765  PlannerBufferBytes: 1232
09:11:59.921 : echo:V38 stored settings retrieved (453 bytes; crc -15873)
09:11:59.921 : Unified Bed Leveling System v1.01 inactive.
09:11:59.968 : Unified Bed Leveling initialized.
09:11:59.968 : UBL System reset()
09:11:59.968 : echo:  G21    ; Units in mm
09:11:59.968 : echo:  M149 C ; Units in Celsius
09:11:59.984 : echo:Filament settings: Disabled
09:11:59.984 : echo:  M200 D3.00
09:11:59.984 : echo:  M200 D0
09:11:59.984 : echo:Steps per unit:
09:11:59.984 : echo:  M92 X80.00 Y80.00 Z4000.00 E402.00
09:12:00.000 : echo:Maximum feedrates (units/s):
09:12:00.000 : echo:  M203 X300.00 Y300.00 Z3.00 E25.00
09:12:00.000 : echo:Maximum Acceleration (units/s2):
09:12:00.000 : echo:  M201 X1000 Y1000 Z50 E5000
09:12:00.015 : echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
09:12:00.015 : echo:  M204 P1500.00 R2000.00 T1500.00
09:12:00.015 : echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
09:12:00.031 : echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
09:12:00.031 : echo:Home offset:
09:12:00.031 : echo:  M206 X0.00 Y0.00 Z0.00
09:12:00.031 : N1 M110*34
09:12:00.031 : N2 M115*36
09:12:00.031 : N3 M105*36
[... 183 lines cut ...]
09:13:26.784 : (1,179)                                                        (179,179)
09:13:26.987 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:26.987 : 
09:13:27.128 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:27.128 : 
09:13:27.333 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:27.333 : 
09:13:27.473 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:27.473 : 
09:13:27.614 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:27.614 : 
09:13:27.769 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:27.769 : 
09:13:27.910 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:27.910 : 
09:13:28.050 : 0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:28.050 : 
09:13:28.191 : [ 0.000]   0.000    0.000    0.000    0.000    0.000    0.000    0.000    0.000
09:13:28.222 : (1,1)                                                            (179,1)
09:13:28.238 : (0,0)                                                             (8,0)
09:13:28.254 : ok
09:13:29.177 : T:21.95 /0 B:19.96 /0 @:0 B@:0
09:13:32.179 : T:21.87 /0 B:19.92 /0 @:0 B@:0
09:13:35.182 : T:21.95 /0 B:19.92 /0 @:0 B@:0
```

lrpirlet, 2017-06-15:
@thinkyhead 
Could I have your advise too... Git blame reports you to be the submitter, gitk report you are the author...

Thanks in advance.

lrpirlet, 2017-07-05:
@thinkyhead @Roxy-3D @repetier

**I have solved that bad console output using the following workaround that prevent AUTOTEMP output for a while... 
M155 S15
G29 T0
M155 S3 (that is MY setup, repetier host sends M155 S1...)**

I have also proven that a lock mechanism does work, but I have NOT been able to complete ALL the modifications in the Marlin code (I am so slow that I will spend more time to keep current than to make progress).

I really think that some console print serialization mechanism could be invented. Various solutions could be used:
- To modify the various macro so that they set or reset the bool... the code has NO common way to enter or exit a console print...
- To find each console print in the code and set the bool at entry point then reset the bool at exit... Asynchronous console print excepted... That is ok today cause today only AUTOTEMP sends asynchronous console print.. but will fail as soon as another asynchronous console print is introduce... (no doubt it will happens, if one occurrence exists, several will)
- To stop AUTOTEMP or any other asynchronous console print before issuing non atomic console output such as 'G29 T0'...

Now I really wonder why we need a multi characters protocol to synchronize Marlin and the host... I know that GCODE protocol impose the 'ok'... Probably because a human should be able to read what is going on and eventually react... 

I guess we could propose an alternate protocol that would be a _**ONE CHARACTER ASCII**_ instead of a multi characters set that could be intermixed because asynchronously produced. This control character could be periodically repeated till the condition goes away.
- control the command flow so that the buffer is constantly nearly full (XON XOFF or rather **DC1 DC3)
- control the priority of the command (ESC-command so that, for example, we can flush the command buffer and kill the bot OR request immediate status OR request immediate temp reading or ...)
- tell the host that the received command is garbled (NAK, please send it again)
- provide the host with some current command status ( BEL bot is still busy working)
- request the host more information ( ENQ for bot waiting for user input...)
- more?  (DC2 DC3 can be defined at will)

ref: https://en.wikipedia.org/wiki/ASCII

 Of course this implies to engage meeting with the host software authors. Besides, I have not the skill (or speed) to write it.
Please do consider this protocol proposal as a request for improvement....

Roxy-3D, 2017-07-05:
I believe we need a much more robust protocol for 3D Printing.  It needs better error detection and synchronization.   For starters...   I want at least a 16-bit CRC to help validate the contents of the command line that has been sent.

apelly, 2017-07-06:
Crazy talk! The USB and serial protocols already have error handling built in.

@lrpirlet is actually asking for a separate console and terminal. All that is showing up here is conflicting display data. This is the nature of the environment.

There is a clue earlier in the issue:
> I doubt that Marlin is buggy, I think it may be over-optimized for Repetier...

I'm as busy as usual, but Marlin must have one routine to output data to the console. The solution is to send it complete lines where interruption is undesirable. Then it is up to the host to handle unwanted messages (lines) during data collection, say, by disabling or filtering them.

WRT keep alive messages; why are they being sent during a busy output time? They are not needed then?

Roxy-3D, 2017-07-06:
> Crazy talk! The USB and serial protocols already have error handling built in.

Yes.  But it is possible to lose characters.   They may not be getting lost on the fake serial port implemented  on USB.   But I assure you...   characters can get lost.

davidelang, 2017-07-06:
The USB protocol has error handling built in, but does the serial protocol? most 
arduino boards don't have RTS/CTS hooked up.

The default settings for the arduino serial library only has a 64 byte buffer, 
if you have a high speed computer and/or the arduino gets busy (sending pulses 
to motors for example :-) it is possible to overrun this buffer.

We are fighting this problem with the Maslog CNC router project.

Roxy-3D, 2017-07-06:
I believe what you say is true.    But I've seen a lot of characters get lost when Marlin is transmitting to the host.   (The opposite direction!)    I had to insert a lot of safe_delay()'s in so the host's don't lose characters.

https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/ubl.cpp#L157-L165

davidelang, 2017-07-06:
again we get to the small buffer and lack of RTS/CTS. If there is other stuff on 
the USB bus, the USB/serial on the arduino isn't going to be able to send the 
data as fast as the arduino tries, and so some data will be lost.

This should be far less common than problems in the other direction. But if you 
are using a Pi where the network is on USB, you can easily run into this sort of 
problem.

lrpirlet, 2017-07-07:
@davidelang @Roxy-3D

Well, XON/XOFF is nothing but a software equivalent of RTS/CTS to synchronize the data flow over a serial line... This is a very old and known terminal protocol implemented in ASCII.

BUT to use it implies to have BOTH SIDES to comply... in other terms it means defining a protocol (that goes a little beyond just the data flow control... We also need that the bot can tell the host: I wait for data or user intervention needed or I am busy working on a long command or etc...)

That is exactly what I propose, using the MINIMUM resource (one character signal rather than multi-characters messages)... So that also we can free the CPU for more important tasks such as LIN-ADVANCE...

jbrazio, 2017-07-07:
ATmega family does not support hardware flow control, we would require to build a custom implementation using two additional pins for it. It's doable but this will only add more strain into the uC.

davidelang, 2017-07-07:
a protocol that changes all existing software seems like a really bad idea.

Roxy-3D, 2017-07-07:
> a protocol that changes all existing software seems like a really bad idea.

Relax...   It would be a negotiated mode.   Unless both sides supported it, it wouldn't happen.

davidelang, 2017-07-07:
On Fri, 7 Jul 2017, Roxy-3D wrote:

>> a protocol that changes all existing software seems like a really bad idea.
>
> Relax...   It would be a negotiated mode.   Unless both sides supported it, it wouldn't happen.

Ok, I have no problem with that. I've just seen too many projects break lots of 
external stuff because the developer(s) decided to 'improve' the protocol and 
ignored backwards compatibility.

lrpirlet, 2017-07-16:
@roxy-3D
I never updated that entry (forgot about it...). I did run ``` G26 P B105  H215 F2.79 L0.2 N0.4 ``` and got this nice result..

I have NO problem, I just want to take this opportunity to underline your nice work... Thanks.

[image]

closed 2017-07-16 by lrpirlet

Roxy-3D, 2017-07-16:
Wow!  That is a nice Mesh Validation Pattern!   I can't even see where it started the print.   I need to bookmark that picture for when people ask "What is the mesh validation pattern supposed to look like?"

PS.  Nozzle size is now controlled with the S parameter.   But the default is .4mm so you did get the same setup as you wanted.