MarlinFirmware/Marlin #7023: M420 State not stored into eeprom

sarf2k4, 2017-06-10:
Hi

I was trying to enable the M420. From my understanding to use this, would require me to do the following order

1. g28
2. g29
3. m500
4. m501
5. m420 s1
6. m500

Lastly would be of course, verify that the settings are stored properly. When I did the above steps, issuing m501 for m420 state, it just stays at S0 instead of S1, while m503 says it is S1.

I re-connected the printer's board again via pronterface, checked on both m501 and m503, both states m420 S0

Right now it is M420 S0 Z10.00

fiveangle, 2017-06-11:
Mine sticks.

sarf2k4, 2017-06-11:
I am using ramps 1.4, single nozzle 2 extruder setup, bilinear leveling and
babystepping enabled.

I wonder why it doesnt stick

fiveangle, 2017-07-03:
I think I was confused which hex I had on the bot when I tested, sorry.  Tested again with a fresh compile with this option (no, your step 6 does not save state upon restart).  Reviewing further, G28 resets bed leveling state when BILINEAR is active, so I don't see how M420 state could ever be saved / restored from EEPROM.  Sorry for the misinformation.  After G28 one would have to execute M420 S1 to enable BL.  I believe UBL has this feature (which has met some critisizm since it is a departure from historical expectations.)

Scott recently [updated docs here](http://marlinfw.org/docs/gcode/G029-abl.html) and it states only that M420 S1 is required to use the mesh, but doesn't mention sticking across reboots.

thinkyhead, 2017-07-04:
As the machine boots into unknown state, you have to `G28` before activating bed leveling. Then you can use `M420 S1` to use any data that was saved to EEPROM with `M500`. The currently-active state of leveling is not saved by `M500`, but the mesh data is.

The actual effect of the "sticky" behavior is that once you enable leveling it continues to stay on after `G28`. But historically, `G28` is specified to disable your bed leveling, so this would be an exception. So, I believe the behavior should be the historical behavior normally, unless you enable an option specifying that it should persist through `G28`.

```cpp
//#define RESTORE_LEVELING_AFTER_G28 // If bed leveling was on before G28, re-enable it
```

thinkyhead, 2017-07-04:
> steps 1-3 to save a mesh to EEPROM

Yes. Of course, after editing the mesh, you can `M500` to save again…

> Steps 1 and 5 to use it

Yes. Home first, then enable leveling with saved data.

> m501, m420 s1, m500

`M501` isn't much needed. Its effect is to throw away recent changes and go back to whatever was saved on EEPROM. So unless you've modified settings and want to throw them out, you can skip `M501`.

And of course, no step 6 needed. Use `M500` only after `G29` to save the mesh.

jbnarch25, 2017-08-08:
Hi all,

Would it make sense to put the m420 s1 in start code after g28? I have a delta with a fixed z probe for bed leveling that i disconnect for printing so I cant run g29 before each print in start code.

thinkyhead, 2017-08-27:
> Would it make sense to put the m420 s1 in start code after g28?

Yes, that is the recommended thing to do. I recently added it to all my slicers' start-gcode.

boelle, 2019-02-19:
@sarf2k4 do you still have an issue with latest bugfix 2.0?

sarf2k4, 2019-02-25:
Yes, this issue still present in bugfix 2.x. Here is the procedure that I issued the command to the printer

```
g28
g29
m500
m501
m420 s1
m500
m501
```

However, the m420 state stored in temporary settings only instead of eeprom. I could say that m420 state were ignored in m501 command.

```
m420 s1
m503
```
Will show that m420 is enabled, S1 via m503, I even tried m501 then m503, the state will be disabled

thinkyhead, 2019-02-25:
Only UBL preserves the leveling state in the EEPROM. The other leveling systems require `M420 S1` to be used after a reload of the EEPROM.

sarf2k4, 2019-02-25:
May I have a clarification of m420, g29 and g28?

From what I understand
G28 homes all of the axis and assumes that all of the plane are flat
G29 do bed probing 
M420 enables bed leveling as well as z fade

Usually I only put G28 and G29 in my start gcode. If I don't put m420 s1 in my start gcode, does that mean marlin will not do bed compensation data from that G29 as long as M420 is disabled? I am using bilinear bed leveling method

thinkyhead, 2019-03-03:
> May I have a clarification of m420, g29 and g28?

- http://marlinfw.org/docs/gcode/G028.html
- http://marlinfw.org/docs/gcode/G029-abl.html
- http://marlinfw.org/docs/gcode/M420.html

boelle, 2019-03-16:
@sarf2k4 did what thinkyhead wrote help you?

boelle, 2019-03-16:
correct order should be

g28
g29
m500
m420 s1

G28 homes
G29 probes
M500 saves it
M420 S1 loads from EEprom

and then before each print
G28
M420 S1

but i will test this later tonight

ManuelMcLure, 2019-03-16:
> M420 S1 loads from EEprom

Actually, you want
``M420 L0``
(or ``L1``... depending on which slot you saved the mesh into with ``G29 S...``) to load the mesh.

boelle, 2019-03-16:
@ManuelMcLure i would see if i could confirm this

but can you see any of the problems @sarf2k4 has?

boelle, 2019-03-16:
tried it

M420 L1 did not switch bed level on so i must be doing it wrong

g28
g29
m500
m420 s1

or 

g28
g29 S1
m500

ManuelMcLure, 2019-03-17:
What version of automatic bed leveling are you using? Bilinear, UBL, manual mesh leveling?

boelle, 2019-03-17:
Bilinear

i'm just trying to see if i can reproduce what @sarf2k4 says

ManuelMcLure, 2019-03-17:
The thing is, the behavior of ``M420`` and such changes depending on the type of bed leveling. ``M420 L`` seems to be only available with UBL, and ``G29 S`` has a completely different effect with Linear/Bilinear (Set the XY travel speed between probe points) than with UBL (Save the mesh to EEPROM in the given slot). So knowing the version of bed leveling in use is vital to determine what commands to use.

boelle, 2019-03-17:
> Bilinear

in my case, but yes we need to see what @sarf2k4 uses, he forgot to attach his 2 config files as a zip file

sarf2k4, 2019-03-17:
Hi,

In the gcode explanation about those 3, didn't state that I must use M420 S1 in order to use the bed leveling and mesh data. That would lead me to conclude that G28 then G29 is sufficient  as per older marlin behaves. If M420 S1 is mandatory after G29, then that would explain why I kept getting results of marlin behaves abnormally even with the auto leveling such as left = low, right = high behavior. As of now I didn't have proof over the behavior between before putting M420 S1 and after putting M420 S1. 

I tend not to drag the argument as it may lead to negative conversation between each other.

I have another machine that runs the same version as the config below, with slight changes is the bed size, manual mesh leveling, and mechanical settings. Right now I have RESTORE_LEVELING_AFTER_G28 enabled and at the start of gcode I have these inserted.

```
G28
G29
M420 S1
```
In the lcd settings, bed leveling state was on, though I checked through octoprint terminal, M420 S0 was displayed

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2974836/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/2974837/Configuration_adv.txt)

boelle, 2019-03-17:
maybe the people who knows a bit more about how it works can chime in and tell

question, are you running  the latest bugfix 2.0? i guess it has nothing to do with this but always good to run the latest where bugs are fixed daily (almost)

sarf2k4, 2019-03-17:
> 
> 
> maybe the people who knows a bit more about how it works can chime in and tell
> 
> question, are you running the latest bugfix 2.0? i guess it has nothing to do with this but always good to run the latest where bugs are fixed daily (almost)

If I remember correctly the 2.0.x bugfix was about 4 weeks ago https://github.com/MarlinFirmware/Marlin/issues/7023#issuecomment-465289942, The time you asking for a follow up. I doubt that this will be fixed in instant as this issue 'not really critical' I think. 

Changing printer's firmware daily is a tiring thing to me, not that I complaint, it is just that to configure marlin, has to compare the previous setting and bring them over to the new config to avoid issues if we were to overwrite using older config files. I usually update the firmware when I have the drive to do so and also if there's any changes to the critical feature that I need, in this case the bed leveling stuffs.

boelle, 2019-03-17:
i agree, it was just a thought :-D

but let's see what happens during the day or in a few days

either @ManuelMcLure or maybe even @Roxy-3D could give a clue as to what goes wrong for you

sarf2k4, 2019-03-17:
i have long history of problematic mesh leveling way back right before UBL were officially implemented, Which is always not corrected  the proper way, left perceived low, right perceived high by marlin but the reality it was the opposite. M421 came and it somewhat solved the problem where I have to dial in the offset value for every point from the probed value.

ManuelMcLure, 2019-03-17:
> left perceived low, right perceived high

This sort of behavior is often caused by a mechanical issue, specifically by a twisted X axis. Unfortunately this is very common with the the i3-style X axis with two parallel linear rods my Anet A8 has this which is why I've moved to using UBL and essentially used the G26 test pattern to manually create a mesh that I saved and just reuse every time I print.

closed 2019-05-16 by thinkyhead