MarlinFirmware/Marlin #7032: Issue with steppers DAC

emartinez167, 2017-06-11:
I get this error with the latest 1.1.x bugfix:

Arduino: 1.6.13 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

/var/folders/7s/_12bnnk90y72chfwxb19sj_80000gn/T//ccqkAfgD.ltrans11.ltrans.o: In function `dac_driver_commit()':
ccqkAfgD.ltrans11.o:(.text+0x14): undefined reference to `dac_current_set_percents(int*)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

fiveangle, 2017-06-11:
Can't really help without more info:

http://marlinfw.org/docs/basics/reporting_bugs.html

emartinez167, 2017-06-12:
It appears it might have been broken in this commit: (6643d55)

https://github.com/MarlinFirmware/Marlin/commit/6643d553f6764fbb44cd8aa8ab6de041674a641f

In function 'dac_driver_commit()'
Undefined reference to 'dac_current_set_percents(int*)'

Does anyone know what has changed? I am running a RigidBotv2.

Regards,

Ernesto.

p3p, 2017-06-13:
line 98 of stepper_dac_cpp
`void dac_current_set_percents(const int8_t pct[XYZE]) {`

probably should be 
`void dac_current_set_percents(const int16_t pct[XYZE]) {`

emartinez167, 2017-06-13:
I'll check tonight and report back (in Australia).

Regards,

Ernesto.

Bob-the-Kuhn, 2017-06-13:
I swear this bug was fixed last week but I can't find any proof.

The solution is to make the two declarations of `dac_current_set_percents` and  driverPercent match the array this info is stored in which is:
 `uint8_t dac_channel_pct[XYZE] = DAC_MOTOR_CURRENT_DEFAULT;`

I'll start a PR in the morning.

Bob-the-Kuhn, 2017-06-13:
Must be too late at night. I can't get this to fail on a Windows system. 

I downloaded the latest bugfix-1.1.x, selected RIGIDBOARD_V2 and uncommented DAC_MOTOR_CURRENT_DEFAULT.  I verified the int8_t and int16_t defines were still there.

emartinez167, 2017-06-13:
That is interesting. I'll try again tonight with a fresh download. I must have other settings enabled.

Regards,

Ernesto.

p3p, 2017-06-13:
@Bob-the-Kuhn sorry I wasn't clear in my last comment the function prototype and definition don't match
line 98 of stepper_dac.cpp
`void dac_current_set_percents(const int8_t pct[XYZE]) {`
line 50 stepper_dac.h
`void dac_current_set_percents(int16_t pct[XYZE]);`

and the only time the function is called:
`void dac_driver_commit() { dac_current_set_percents(driverPercent); }`
its passed a variable of type int16_t
`int16_t driverPercent[XYZE];`

so I presumed the definition was wrong and should be int16_t.

emartinez167, 2017-06-13:
I did a bunch of tests gradually applying my own Configuration.h and Configuration_adv.h lines and found that the issue only appears when #define RIGIDBOT_PANEL is uncommented, so obviously something was missed. Whatever changes were done, they were only applied to other LCD screens but were left out for this type of LCD panel.

Regards,

Ernesto.

Bob-the-Kuhn, 2017-06-14:
I've created PR #7056 to fix this.  

Please see if it actually changes the percents via the menu.

emartinez167, 2017-06-14:
I'll test it tonight after work. Thanks!

Saludos,

Ernesto.

closed 2018-02-07 by thinkyhead