MarlinFirmware/Marlin #7034: Moving steppers individually with Z_DUAL_STEPPER_DRIVERS

040medien, 2017-06-11:
Is there any way I can manually move the two stepper drivers to correct a slight X axis imbalance?

I know the preferred way for an i3 is to max out on the Z axis until both steppers grind and I know about having two z endstops with a manual offset. Ideally I would however prefer to use my Z probe to home Z a few times with the probe alternating side until the axis is level to the bed.

In absence of this feature I was thinking to do this manually by probing the two spots and then moving one side's stepper to correct. Is this possible today?

Bob-the-Kuhn, 2017-06-11:
The method you describe is not supported today.

What I do is enable the dual Z endstops, move the Z_MIN endstop to the top and then home in the + direction.  The important defines for that mode are:
```
   #define Z_DUAL_ENDSTOPS
    #if ENABLED(Z_DUAL_ENDSTOPS)
      #define Z2_USE_ENDSTOP _ZMIN_
      #define Z_DUAL_ENDSTOPS_ADJUSTMENT -11  // use M666 command to determine/test this value
    #endif
#define Z_MIN_PROBE_ENDSTOP
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MAX_POS 200   // play with this until Z0 is 5-10 mm above the bed
#define Z_HOME_DIR +1
```

fiveangle, 2017-06-11:
I saw some people asking this before and having experimented with printed linear bushings and this, I can say with 100% confidence that while it sounds like a great idea, in reality the linear bearings are designed for zero thrust angle.  The strive should be to manually adjust the Z endstops so that after homeing, the X carriage is orthogonal to the 2 Z guides.  The orthogonality of the Z guides to the X carriage Z bearings needs to be the driver that determines Z stepper sync, not the X carriage nozzle-to-bed distance across the range of motion.

I've thought about this a bit and conceptualized an X carriage where the Z bearing guides were connected to the X gantry via a pivot that constrained the movement to within the plane of the two Z guides but allowed free pivot within that plane.  The result would allow what you want, but would also add additional width to the X gantry (with the added pivot on each side) along with whatever increase in backlash that might add.

Sounds good but doesn't look so good on paper without some clear design changes :)

fiveangle, 2017-06-11:
If I were Stratsys engineer tasked with designing a relatively "cheap" industrial desktop printer based on the i3 design, to reduce precision maching costs (not an issue in their flaship printers), I would incorporate thrust roller bearings on each side of the X-carriage Z-rod linear bearing holders (in front and behind when looking at a typical i3 design) so they could pivot through an axis orthogonal to both the Z-carriage and the Z-rods and going through the centerline of the Z-rod on each side, then augment UBL with the feature you're describing.  It would work a treat !

If this isn't making sense, imagine the following is the top-down view is one side of the X gantry with the linear bearing holder with the bearing (the "o") connected through a rod that pivots and connects to the rest of the X gantry:
```
X     TOP VIEW
|
|    ||  -  ||
|<- =||=|o|=||= ------new pivot axis------>
|    ||  -  ||
|     \\   //
|      \\_//
|       \ /
|       | |
|
Z-------------------- Y
```

fiveangle, 2017-06-11:
FYI - on this topic, on the Prusa Original i3 MK2 they lower stepper DAC current then have the user run the Z lead screws up together until they both phyically hit the end brackets and miss steps (I cring every time I have to do it :).  Then they home Z and calibrate the bed mesh. This technically requires the top Z brackets to be within the bisector of a circle that is tangential to each Z rod.  If no, there will be a (small) thrust angle that works against the linear bearings as they move up and down.

I've used little strips of kapton tape to adjust the "topmost" postion of the X carriange on each side so at the top the X carriage as orthogonal to the Z rods as possible.  This keeps the Z axis linear bearings riding as close to within the Z rod movement axis as possible (driven by the Z rod geometry, not the bed plane).  While you dont notice the thrust angle with the cheap linear bearings that come with the MK2, my near-zero-tolerance hand-lapped printed linear bushings will bind at the slightest thrust angle.

dot-bob, 2017-06-12:
On my large 500mm x 500mm x 600mm printer I am too impatient to wait for it to ascend to the top for limit switches so I have some L brackets clamped under the end of my linear rails.  My print head is able to travel Y -10 and clear the bed so I can bottom the left and right linear carriages on the brackets with the nozzle slightly under the bed.  I usually do this by hand though as I don't like the idea of relying on steppers crashing.  The z lift then moves the head up for the probe clearance then moves the x and y axis.  Perhaps one of these days I will put limit switches there and code up a gcode command to run a short script to automate this process.

closed 2018-02-07 by thinkyhead