MarlinFirmware/Marlin #7035: Point me to the right direction, i3 Prusa with BLTOUCH, how ?

jfxmedia, 2017-06-11:
Help! I've completed my i3 diy build, installed an e3d hotend, things are good, but I've purchased a BLTOUCH sensor and do not know where to start. Mounting is no issue, installing via firmware and where to plug it on the board is the issue I'm having... When trying to upload the rc8 firmware to my ramps1.4 I get no display.. I would donate via paypal for the proper help on this one..

Roxy-3D, 2017-06-11:
You probably should be using https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x and not RC8.

You can turn on #define BLTOUCH in configuration.h   And plug your probe into the first servo connector on the board.   And then try things suggested here:  https://github.com/MarlinFirmware/Marlin/issues/7024

jfxmedia, 2017-06-11:
Ok thank you, so I finally got that bugfix version working, but it's apparently a process. I have successfully finally got the LCD to turn on by defining the right LCD controller in the config file. I'm missing the heatbed temp totally, just the nozzle temp is showing. Also auto home makes the motors freak out, as well as the knob left and right is incorrect... I'm surprised at how difficult this is...

Tannoo, 2017-06-12:
If you have a heated bed, you need to define the thermistor type other than 0.
There is a menu direction to define to reverse the operation.
Motors freaking out... dunno. One person's freakout is another person dance.

Make sure the travel rates and accels are set correctly

Roxy-3D, 2017-06-12:
You should bring up PronterFace on your PC.   That lets you type in and send commands like M119 and M114 to check the endstops and the current position.   And it also lets you click on the target to move a given axis 1mm left or right or forward or back or up or down.  It is very quick to know if you have the motor directions correct.

closed 2017-10-31 by thinkyhead