MarlinFirmware/Marlin #7036: Make possible to exit from Filament Change Menu

magnuspub, 2017-06-11:
If a print is stopped (extruder is shutting down) and try to make a filament change you cannot exit because it wait for extuder temperature

Video:
[VID_20170611_204953.zip](https://github.com/MarlinFirmware/Marlin/files/1066689/VID_20170611_204953.zip)

.

tcm0116, 2017-06-12:
This should have been fixed with #6963. Please update to the latest bugfix-1.1.x and verify that the problem is resolved.

magnuspub, 2017-06-12:
I'm already with the latest bugfix-1.1.x

tcm0116, 2017-06-12:
The reason that it's stuck is that the target temperature is set to 0 (in your video you can see the 33/0, where 33 is the actual temperature and 0 is the target temperature). Since the printer can't actually get the nozzle down to 0 degrees, it gets stuck. 

#6963 should have prevented this from occurring. Can you answer these questions to help me diagnose this?

1. Is **PREVENT_COLD_EXTRUSION** enabled in your Configuration.h?
2. Had you sent an `M302` command to the printer?
3. Can you walk me though the sequence (and GCode commands, if possible) you took from the time you powered on the printer and when it got stuck?

magnuspub, 2017-06-12:
I understand that it's just a problem for me, that sometimes it can enter the wrong menu, but it's not nice to force the printer reset to get out of a menu.
As it was done for ubl manual calibration, it could be functional if you held down the button for a few seconds you could exit

1. yes i have PREVENT_COLD_EXTRUSION enabled:
   #define PREVENT_COLD_EXTRUSION
   #define EXTRUDE_MINTEMP 170
2. No, I print from sdcard
3. Power the print, select a file from sdcard, nozzle and bed start heating, let print a few layer, than stop print (for example for a mistake), temperature start to decrease, go to filament change while the nozze is over 170 (>170), once entered in filament change menu and your temp in the meantime is <170, you have to reset your printer to exit from a menu.  (if not connected to computer)

tcm0116, 2017-06-12:
Ideally, `M600` shouldn't get stuck, so there should be no need to have to abort. However, it appears that there is some path by which `M600` can get stuck, and I'd like to get that fixed. Adding the ability to abort `M600` is a new feature that could be then added in the future.

What do you have **FILAMENT_CHANGE_UNLOAD_LENGTH** defined as in Configuration.h?

magnuspub, 2017-06-12:
#define FILAMENT_CHANGE_UNLOAD_LENGTH 100

magnuspub, 2017-06-12:
Did you think about this?

> Or, more simply, make sure that you can not enter the filament change menu if the "current temperature" is less than EXTRUDE_MINTEMP, OR "set temperature" is less than EXTRUDE_MINTEMP

tcm0116, 2017-06-12:
After `M600` processes any arguments, it calls `pause_print`. The first lines of `pause_print` are:

```
    if (!DEBUGGING(DRYRUN) && unload_length != 0) {
      #if ENABLED(PREVENT_COLD_EXTRUSION)
        if (!thermalManager.allow_cold_extrude && 
            thermalManager.degTargetHotend(active_extruder) < thermalManager.extrude_min_temp) {
          SERIAL_ERROR_START();
          SERIAL_ERRORLNPGM(MSG_TOO_COLD_FOR_M600);
          return false;
        }
      #endif

      ensure_safe_temperature(); // wait for extruder to heat up before unloading
    }
```
As you can see, the first thing it does is verify that the target temperature for the active hotend (or "set temperature" as you called it) is above the `extrude_min_temp` (which defaults to **EXTRUDE_MINTEMP** if you haven't changed it via `M302`), and aborts if not.

`ensure_safe_temperature` is then called, which waits for the active hotend to get within 3 degrees of the target temperature.

In your case, it somehow got passed the first check, which I'm not really sure how that's possible since you have **PREVENT_COLD_EXTRUSION** enabled and the `unload_length` should be `-100` since you have **FILAMENT_CHANGE_UNLOAD_LENGTH** set to `100`.

I hate to sound redundant, but could you verify one more time that you're using bugfix-1.1.x newer than about 3 days ago? You can check `pause_print` in Marlin_main.cpp to see if it starts with the code block above.

magnuspub, 2017-06-12:
I flashed the last bugfix1.1.x firmware yesterday before making the video and opening the issue, however I try to update it now and retry.
And yes pause print looks like your.

magnuspub, 2017-06-12:
ah, please, can you check you have read all my message above?

magnuspub, 2017-06-12:
I flashed the last bugfix1.1.x and tried again. I can not get out of the menu.
Or maybe I can enter the menu equally. The problem is not M600 (which I have never tried) the problem is the user interface (the Change filament menu). Remember, I use only the display not the serial interface.
Thanks for your patience.

magnuspub, 2017-06-12:
If it helps, I tried the M600 and it works. Is the menu that does not work as it should.

tcm0116, 2017-06-12:
> ultralcd.cpp line 2210 - 2216

I think this is the problem. `lcd_enqueue_filament_change` is only checking the actual temperature of the hotend, which is good, and then displays the **ADVANCED_PAUSE_MESSAGE_INIT** screen. However, `pause_print` ends up aborting because the target temperature is not high enough, so the LCD gets stuck on the **ADVANCED_PAUSE_MESSAGE_INIT**.

I'll see if I can get this fixed tonight. My printer is working on an 18 hour print that won't finish until the early morning, but I should be able to put something together for you to try, if you don't mind.

magnuspub, 2017-06-12:
I try something like this but doesn't work. 
``
#if ENABLED(ADVANCED_PAUSE_FEATURE)

    void lcd_enqueue_filament_change() {
      if (!DEBUGGING(DRYRUN) && thermalManager.tooColdToExtrude(active_extruder) && (thermalManager.degTargetHotend(active_extruder) < thermalManager.extrude_min_temp)) {
        lcd_save_previous_screen();
        lcd_goto_screen(lcd_advanced_pause_toocold_menu);
        return;
      }
      lcd_advanced_pause_show_message(ADVANCED_PAUSE_MESSAGE_INIT);
      enqueue_and_echo_commands_P(PSTR("M600 B0"));
    }
``

tcm0116, 2017-06-13:
I just created #7048, which should resolve this issue.

magnuspub, 2017-06-13:
Ok. I've tested #7048 an it fixed the issue!

closed 2017-11-03 by thinkyhead