MarlinFirmware/Marlin #7041: Compilation ERROR : 'lcd' was not declared in this scope

Bergerac56, 2017-06-12:
When #define DOGLCD  and #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and #define STATUS_MESSAGE_SCROLLING are active, the last bugfix 1.1.x of today does not compile (at least at my side). I get a 'lcd' was not declared in this scope message

(No other modifications to configuration files)

benlye, 2017-06-12:
Copy and paste problem from the HD44780 code?  On lines 424 and 428, replace `lcd.print` with `u8g.print`.

I also had to add `const bool blink = lcd_blink();` at line 409 to avoid a different error about 'blink' not being declared.

Bergerac56, 2017-06-12:
Thanks,

It works.

Bergerac56, 2017-06-14:
Just to mention that the correction proposed by benlye is still not yet implemented. Need to do the changes manually on each new version of bugfix.

benlye, 2017-06-16:
499d5c3b2456aeb79161ff34e10d4b7a1f02676a nearly fixes this by replacing `lcd.print` with `u8g.print`, but I still get an error when compiling because blink isn't declared:

```In file included from sketch\ultralcd.cpp:75:0:
sketch\ultralcd_impl_DOGM.h: In function 'void lcd_implementation_status_message()':
ultralcd_impl_DOGM.h:434: error: 'blink' was not declared in this scope
       if (last_blink != blink) {
                         ^
exit status 1
'blink' was not declared in this scope```

Bergerac56, 2017-06-16:
I confirm. "const bool blink = lcd_blink(); " still missing and generating a compile error.

Bergerac56, 2017-06-17:
Thanks thinkyhead and all. It compiles fine now

closed 2017-06-17 by Bergerac56