MarlinFirmware/Marlin #7043: Heated bed temperature rises during printing

adasko1, 2017-06-12:
Hi, I recently changed my board to arduino mega clone + ramps 1.4. I flashed marlin 1.1 bugfix from marlin site. I would say nearly everything works, but I have one problem. 

When I preheat for example temps for PLA it heats up and stays on demanded temp. If I than set the temp to 0, bed cools down just fine. 
But when Im trying to print the temp of the bed starts continously  rising over the seted temp. Even setting bed temp during print to 0 doesnt change anything. Only way to stop bed temp from rising is resetting the board.
[image]

adasko1, 2017-06-25:
Ok, I figured out that the bed start heating above set temp, when i set the cooling fan on 100%, anybody know what can be wrong?

Bob-the-Kuhn, 2017-06-25:
Could the fan shorted to the bed?

ss89, 2017-07-04:
I do have the same problem with Marlin 1.1.1 stable, will try to just unplug the fans and report back

ss89, 2017-07-04:
Last thing i did was replacing the arduino and the ramps board.
Since i just tested that the arduino is fine, i will now replace the ramps board with the one that was inside that machine before.

ss89, 2017-07-04:
@adasko1 Please try to replace your ramps board. I had just replaced mine and after switching back the problem was gone. I had mine from here: https://www.aliexpress.com/item/F401-1pcs-RAMPS-1-4-3D-printer-control-panel-printer-Control-Reprap-MendelPrusa/1519223336.html?spm=2114.13010608.0.0.XHORbf
So you better NOT buy from that seller.

Bob-the-Kuhn, 2017-07-04:
Please ZIP up both your config files & drop on your reply.

With that I can see if there's something wrong in your setup.

closed 2018-02-09 by thinkyhead