MarlinFirmware/Marlin #7049: Strange behavior at Ultimaker Original when homing - need assistance

Drayson80, 2017-06-13:
Dont know if it is a bug or just an Error80 (so my fault) but I could not manage to port my config.h and .adv settings form a 2014 Marlin to the new 1.1.3
Issues: 
when homing, it ignores the endstops
when moving (independend of 0,1 or 1 or 10mm) it makes a few mm and then stops
moving into "negaitve" direction is not possible, also when proir moving (a few mm) into "positive" direction.

I assume, it's a fault in settings so if anybody could take a look, I would be gratefull.

Ultimaker setup with dual extrusion, self built heat bed and Silent Step Stick drivers TMC2100 at all 5 steppers.
Attached also my 1.1.3 config files
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1070745/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/1070746/Configuration_adv.h.txt)

raguera, 2017-06-13:
Hello,

Have you tried to invert your endstops logic?

I have dual extrusion in my printer and endstops work fine.

Drayson80, 2017-06-13:
thank you for the hint - will doublecheck it...

Drayson80, 2017-06-14:
seems I inverted the endstops twice... so that was the issue. Thanks for the hint.
now start testing...

closed 2018-02-07 by thinkyhead