MarlinFirmware/Marlin #7050: Printer doesn't print with DUAL_X_CARRIAGE enabled

raguera, 2017-06-13:
Dear Marlin Community,

I am trying to print with my 3D printer but it doesn't print after Repetier/Cura starts print.
After print command, printer makes X Y Z homing and warm hotends and bed up, but once temperature is reached, host software sends commands properly and it gets 'ok' response from firmware constantly but without moving any axis, so "supposedly" printing finishes in just few seconds, when it should take one hour.
It just happens if I enable DUAL_X_CARRIAGE, otherwise it prints well if it's disabled.

I'm using a Rumba board and last version of Marlin

Anyone knows what could happen?

Appreciate any help.

oysteinkrog, 2017-06-13:
Duplicate of issue https://github.com/MarlinFirmware/Marlin/issues/6956
I confirm this issue, works in 1.1.1
Pinging @thinkyhead as I think this is a pretty serious regression.

raguera, 2017-06-13:
You are right.
Thanks a lot!

oysteinkrog, 2017-08-04:
Hmm, for some reason my merged PR did not correctly close this, please close it now.

PR #7386 (2017-08-01): Fix DUAL_X_CARRIAGE movement
Broken by 91841d75c9a473e15932bd4589d64ee59efe947e (between 1.1.1 and 1.1.2).
Search for DUAL_X_CARRIAGE in that commit and the error is obvious (removed line in the #if chain).
The old code was:
```cpp
  #if IS_KINEMATIC
    if (prepare_kinematic_move_to(destination)) return;
  #else
    #if ENABLED(DUAL_X_CARRIAGE)
      if (prepare_move_to_destination_dualx()) return;
    #endif
    if (prepare_move_to_destination_cartesian()) return;
  #endif
```
…and was changed, in error, to…
```cpp
   if (
     #if UBL_DELTA // Also works for CARTESIAN (smaller segments foll [... cut]
commit e948f77c (2017-08-01): Fix DUAL_X_CARRIAGE not moving at all

Axis would only move when homing.
Bug introduced in 91841d75c9a473e15932bd4589d64ee59efe947e.
Fixes #6956, fixes #7050 and fixes #7291
closed 2017-08-07 by raguera