MarlinFirmware/Marlin #7052: Require homing when issuing M600.

JimBrown, 2017-06-13:
I like to use M600 to change filaments prior to starting a print (I have a bowden type extruder.) I have configured M600 to move the hot end to the back corner when changing filaments during a print. However, if I'm changing the filament prior to a print and have not yet homed the system, the hot end movement starts from what ever position it thinks it's at, causing the bed and hot end to crash the end stops.

I suggest that a check that all axis have been homed be added to the M600 command, similar to what's done for G29.

I have tried this:

```c++
  inline void gcode_M600() {

    // Don't allow auto-leveling without homing first
    if (axis_unhomed_error()) return;
```

And it seems to work, but I don't know enough to know if that will have any other effects.

Thanks.

Roxy-3D, 2017-06-13:
The homing operation is already a nuisance to the user.   If we are not going to allow M600 to function without being homed...  Let's have M600 home the machine if needed:
```cpp
  inline void gcode_M600() {
    // Don't allow filament change without homing first
    if (axis_unhomed_error()) {
        home_all_axes();
    }
```

JimBrown, 2017-06-13:
That would work too!

May I assume that this might show up in the code at some point?

Thanks.

Roxy-3D, 2017-06-13:
Make a Pull Request...   I'll merge the above code.

JimBrown, 2017-06-13:
Quick question on pull requests... (Haven't done one before.)

I have a fork with a number of modifications. Can I limit the pull request to a specific modification?

Thanks.

JimBrown, 2017-06-13:
My other thought is to do a new fork, make the modification, do a pull request, and once accepted, delete the new fork...

Roxy-3D, 2017-06-13:
I'm not a Git expert.    But the Git experts would say:   Create a new branch off of the bugfix-v1.1.x branch.  Make sure it is up to date.   Make the changes and test them.  (Although this is so simple...  there isn't much to test.)   And generate the Pull Request from that.

FHeilmann, 2017-06-13:
if you have a set of commits that only implements a certain  modification then you can do the following:

from a synced bugfix-1.1.x branch, assuming "origin" is your remote git repo

git checkout -b "my_new_feature"
git cherry pick <commit_hash1> <commit_hash2>

git push -u origin "my_new_feature"

it probably won't work well if your modifications depend on each other though.

Tannoo, 2017-06-14:
So, will this home in the middle of a print during filament change?
I would think that homing might be needed only if it's not printing.

JimBrown, 2017-06-14:
It should not, since homing has already been done prior to starting the print.

Tannoo, 2017-06-14:
I will try it out.

thinkyhead, 2017-06-16:
Applying changes from #7058 over at #7071…

closed 2017-06-16 by thinkyhead