MarlinFirmware/Marlin #7055: Marlin serial connect to another arduino.

Zipstar, 2017-06-13:
Hello 
I have Arduino mega with ramps 1.4 and i try to connect another Arduino to send G Code live through the serial port .
When i open serial connection and send the G Code through the computer everything work fine and i see the messages and can send the code but when i try to do so with another Arduino that will send the G Code to the Mega its not working and i don't see any  message.
i will be more than happy if someone can help me with that issue.
Thanks

Tannoo, 2017-06-14:
RX and TX not correct?
Transmitting to the incorrect Mega serial port?

Zipstar, 2017-06-14:
RX TX correct ... its like the Marlin waiting to some code to send to start working

Tannoo, 2017-06-14:
That's what it sounds like. So, serial port selection, wiring, or speed issues.

Zipstar, 2017-06-14:
but its not ..
i think Marlin firmware is waiting to some command to start the serial communication .
when i plug the Mega to the computer and "listen" with another arduino connect to the 0/1 (Rx Tx) i can see the communication between them .

Tannoo, 2017-06-14:
Aren't ports 1 and 2 for Serial0? Same port as the usb?

Tannoo, 2017-06-14:
Well, ports 0 and 1.

Zipstar, 2017-06-14:
yup , usb connection is serial

Tannoo, 2017-06-14:
Search for serial here and see what you can find in pas discussions. You can also search arduino forums. Maybe even 3D printer forums.

Tannoo, 2017-06-14:
*past

Tannoo, 2017-06-14:
Yes, but USB is  special. Lol

Tannoo, 2017-06-14:
I believe it also using RS232 which does not support more than 2 devices talking to each other.

jimovonz, 2017-07-11:
Are you using the correct EOL char/s?

closed 2018-02-09 by thinkyhead