MarlinFirmware/Marlin #7057: Problem with ADVANCED_PAUSE_FEATURE on 1.1.3

JanVelleman, 2017-06-14:
Just put 1.1.3 on my delta that previously had 1.1.0-1.

The filament change routine worked previously but now then pressing "resume print" the head returns to the previous print position but to immediately return to the "park position" and continue printing from there... ?

Any Ideas?

This is my config:

`  #define PAUSE_PARK_X_POS 0                  // X position of hotend
  #define PAUSE_PARK_Y_POS 0                  // Y position of hotend
  #define PAUSE_PARK_Z_ADD 50                 // Z addition of hotend (lift)
  #define PAUSE_PARK_XY_FEEDRATE 120          // X and Y axes feedrate in mm/s (also used for delta printers Z axis)
  #define PAUSE_PARK_Z_FEEDRATE 120             // Z axis feedrate in mm/s (not used for delta printers)
  #define PAUSE_PARK_RETRACT_FEEDRATE 120      // Initial retract feedrate in mm/s
  #define PAUSE_PARK_RETRACT_LENGTH 6         // Initial retract in mm
                                              // It is a short retract used immediately after print interrupt before move to filament exchange position
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE 25  // Unload filament feedrate in mm/s - filament unloading can be fast
  #define FILAMENT_CHANGE_UNLOAD_LENGTH 800   // Unload filament length from hotend in mm
                                              // Longer length for bowden printers to unload filament from whole bowden tube,
                                              // shorter length for printers without bowden to unload filament from extruder only,
                                              // 0 to disable unloading for manual unloading
  #define FILAMENT_CHANGE_LOAD_FEEDRATE 25     // Load filament feedrate in mm/s - filament loading into the bowden tube can be fast
  #define FILAMENT_CHANGE_LOAD_LENGTH 640       // Load filament length over hotend in mm
                                              // Longer length for bowden printers to fast load filament into whole bowden tube over the hotend,
                                              // Short or zero length for printers without bowden where loading is not used
  #define ADVANCED_PAUSE_EXTRUDE_FEEDRATE 3   // Extrude filament feedrate in mm/s - must be slower than load feedrate
  #define ADVANCED_PAUSE_EXTRUDE_LENGTH 50    // Extrude filament length in mm after filament is loaded over the hotend,
                                              // 0 to disable for manual extrusion
                                              // Filament can be extruded repeatedly from the filament exchange menu to fill the hotend,
                                              // or until outcoming filament color is not clear for filament color change
  #define PAUSE_PARK_NOZZLE_TIMEOUT 120       // Turn off nozzle if user doesn't change filament within this time limit in seconds
  #define FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS 5 // Number of alert beeps before printer goes quiet
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT       // Enable to have stepper motors hold position during filament change
                                              // even if it takes longer than DEFAULT_STEPPER_DEACTIVE_TIME.
  #define PARK_HEAD_ON_PAUSE                // Go to filament change position on pause, return to print position on resume`

JanVelleman, 2017-06-14:
I found if I comment the "set_current_to_destination();" in line 6084 in the Marlin_main.cpp.
This behavior is resolved. I cross checked version 1.1.0-1 with 1.1.3 and this line seemed to be added but does not need to be here??

So I'm happy as my 1.1.3 version is working (filament change and pause print) as I want it know but does anybody have an idea if this line was an error or needs to be here??

sincerely

tcm0116, 2017-06-15:
Hmm. When I combined M125 and M600, that line got added because it was in `move_back_on_resume()`, which was used by M125. However, the similar portions of the M600 code did not call `set_current_to_destination()`.

I know calling `set_current_to_destination()` after `planner.buffer_line_kinematic()` works for kinematic machines, so maybe the best option would be to move it to within the `#if IS_KINEMATIC` section right after the call to `planner.buffer_line_kinematic()`.

I'm not that familiar with how all of the planner movement commands work, so maybe @thinkyhead or @Roxy-3D can chime in with their thoughts.

thinkyhead, 2017-06-18:
The current (and perhaps previous) version of `M125`/`M600` don't alter the `current_position` when they do their moves, and I guess this is by design. Note that if bed compensation is active, these moves _will_ include bed compensation.

As long as none of the code involved with `M125` and `M600` ever alters `current_position`, and as long as `resume_position` is the same as the starting position, then there's no need to set `current_position` from `destination`.

However, it may be better in the long run to alter the code so it does update `current_position` after each move, so that these functions can be made cancellable. The problem is, if you decide to cancel a filament change or the active print job in the middle of `M125` or `M600`, the `current_position` will be lost if it's not updated after each move.

Alternatively, it should be safe to use `set_current_from_steppers_for_axis(ALL_AXES)` after any cancellation. This function should now be very reliable — though maybe not fully for UBL yet, as it is still not integrated with `Planner::unapply_leveling` AFAIK.

tcm0116, 2017-06-19:
> Note that if bed compensation is active, these moves will include bed compensation.

I would think that this is what we want to ensure the nozzle doesn't crash into the print while moving to the park position. The nozzle is lifted some amount before moving, but I think having the leveling applied is good. 

> As long as none of the code involved with M125 and M600 ever alters current_position, and as long as resume_position is the same as the starting position, then there's no need to set current_position from destination.

For the purposes of fixing this issue I'm the short term, it seems like just removing the call to `set_current_to_destination()` is the best option. We could then revisit this after the code rework. Work for you?

Roxy-3D, 2017-06-19:
> For the purposes of fixing this issue I'm the short term, it seems like just removing the call to set_current_to_destination() is the best option. We could then revisit this after the code rework. Work for you?

I will load up any Pull Request you do and provide test information on a Cartesian printer with a BL-Touch probe.   I'll help any way I can.

thinkyhead, 2017-06-19:
> just removing the call to set_current_to_destination() is the best option.

Ok, I will add that change to my other PR.

thinkyhead, 2017-06-19:
The change has been merged. If it resolves the issue with no side-effects, then we can close this issue.

closed 2017-06-19 by thinkyhead