MarlinFirmware/Marlin #7061: 1.1.3 Custom Bootup Logo Screen // RepRap FullGFX Discount LCD

bolsoncerrado, 2017-06-14:
So I've googled past and present issues with RepRap LCDs and I did generate a logo to customize our printers, created the _Bootscreen.h as adviced and posted it on the Marlin folder. Everything compiles fine, but the bootup logo screen doesnot change at all.

Is it defined at dogm_bitmaps.h now, no matter which LCD you have? Do I have to change all the WIDTH/HEIGHT shown there?

Or what am I missing here??

Thanks!

tcm0116, 2017-06-15:
You might try adding:

```
#if ENABLED(SHOW_BOOTSCREEN) && ENABLED(SHOW_CUSTOM_BOOTSCREEN)
  #include "_Bootscreen.h"
#endif
```

after

```
#define SHOW_CUSTOM_BOOTSCREEN
```

in Configuration.h

Roxy-3D, 2017-06-15:
If you post your boot screen source code...   I can turn it into a text file....  So you can see it without hoping it is hooked into the LCD Panel code right.

bolsoncerrado, 2017-06-15:
Yeah but main question is: where is the width/height of the LCD being used defined? Is the dogm_bitmaps.h file used always anyway?

closed 2017-06-15 by bolsoncerrado

HoangHai1411047, 2018-02-19:
128 for the width and 64 for the height