MarlinFirmware/Marlin #7063: Feature request? Make it possible to skip the bootscreen logo's with encoder button

JanVelleman, 2017-06-15:
Hi,

I have used the custom bootscreen functionality (I have added 3 custom screens) and I was thinking if this would be a nice/handy feature to be able to skip the custom bootscreen with just a press of the encoder button?

Sincerely

boelle, 2019-02-19:
it would not make sense to me, if i dont want the boot screen then i would disable it

and i dont think you save much time either. sure if you reboot your machine A LOT then it would make sense

AnHardt, 2020-07-08:
Does not make sense to me.
At first adding 3 additional custom bootscreens - then asking how to skip them.
The original Marlin bootscreen can be switched off - but that does not save time. It is displayed that long as the initialization of the rest of the hardware takes.

closed 2020-07-08 by ellensp