MarlinFirmware/Marlin #7064: M907 code file for Mini-RAMBo?

VladDracul, 2017-06-15:
I'm at my wits end on this. 

I'm towards the end of a Mini-Rambo based Wanhao Di3, and everything works, sans the Y-axis motor not getting enough current and getting real jerky print bed movement, causing shifted layers. Belt is nice and tight, as tight as my Ultimaker belts. 

Only thing I could get working for uploading firmware was following this: http://manual.prusa3d.com/Guide/Upgrading+firmware/66#_ga=2.59758831.313801453.1497351488-1113437191.1485173819 Every effort I tried to talk to the printer has failed: Repetier, Slic3r, and Arduino IDE, just doesn't work. Repetier, even though it connects to the printer, any commands just go into "waiting" and eventually restarts the Rambo with no results (and calibration works perfectly). Arduino IDE keeps getting random and different errors in the coding (I'm just using the suggested Marlin FW). And these programs keep having these problems despite reinstalling them, and running them as administrator and under compatability. I tried everything. 

Only thing left I could try to do is run a M907 code via SD card: 

`M907 X200 Z100 E100 ;Max 255`

But, I have no idea how to actually make this work. I tried just pasting that in a Notepad file and saving it as a .gcode, but nothing happens, the board just says printing the file and just sits there. Same thing happened when I tried editing a PID tune .gcode file, which I added the M907 line to it. I tried searching online for .gcode examples for the M907 command, but I haven't any luck at all. I tried to search creating my own .gcode, but that kept on bringing up slicer programs instead of writing .gcode manually. 

Can someone make a M907.gcode for me? Or offer any other help for this situation? 

Thank you, any help would be greatly appreciated!

Tannoo, 2017-06-15:
Open text editor. Notepad works fine for this.
Type in your commands.
Pess enter to start a new line.
 Click 'File' then 'Save As' and type in a filename using "gco" for the extention. (filename.gco).

Now copy that to an SD card an try it.

Bob-the-Kuhn, 2017-06-15:
You may need to shorten the extension to just "g".  

----

It sounds like you have two problems:
1. can't talk to the printer via the USB port
2. need to modify the DAC current.

USB com problems are usually either baud rate or port selection issues.
- baud rate - look in your configuration.h file for the baud rate 
- port selection - Remove the USB cable from the controller, In Repetier Host select Config -> Printer Settings and then select the Connection tab.  Pull down the Port selection/menu and note what ports are listed.  Close out of that pop-up.  Plug your controller into the computer.  Navigate back to the Port selection/menu and note which port has appeared.  That's the port you need to use.

VladDracul, 2017-06-16:
Alright, running the .gcode/.gco/.g didn't work. Tried it with all those extensions.

But I did manage to get the printer finally talking with Repetier. Oddly enough, I had to change baudrate settings in the Windows Device Manager (and changed the COM port to COM 3, was COM 5, for good measure). Which I haven't found a single instance of in all my searching.... 

So with things working with Repetier, like motor and position control, naturally I tried to send over the M907 command to the printer, but that didn't seem to fix anything. 

Arduino IDE still isn't working for me, so, I think I'm going to start testing out on older laptops that will need a good dusting...

Bob-the-Kuhn, 2017-06-16:
Sounds like your Windows drivers need to be updated.  Try downloading the Arduino auto installer (.msi) and see if that fixes it.  It'll automatically install the right drivers if needed.

Over at Teensyduino on the download page they also have a Windows driver install utility.

But even with those, operation isn't guaranteed.  My W10 laptop will talk to Mega2560, Teensylu and Sanguino but my W10 desktop won't talk to the Sanguino.

VladDracul, 2017-06-16:
I've tried just that numerous times on my laptop and desktop already... 

So, I dusted off an older laptop, Win7, and everything was actually running fine, but when I clicked upload, the LCD screen on my printer froze. I turned off the printer and turned it back on, only to get two white bars on the LCD screen. 

Any ideas? I tried both reset buttons (LCD screen and board), and the factory resetting techniques supplied by Prusa, but those didn't work.

Bob-the-Kuhn, 2017-06-16:
Sounds like the upload only partially completed.  The bootloader should still be intact.  

Have you tried uploading again?

Worst case is you get a hardware programmer (such as USBtinyISP [$3.38 China ship](http://www.ebay.com/itm/USBTiny-USBtinyISP-AVR-ISP-Programmer-6-10-Pin-Bootloader-for-Arduino-UNO-MEGA-/191821167214?hash=item2ca96eba6e:g:hUgAAOSwTA9X4Oxq) or [$6.75 USA ship]( http://www.ebay.com/itm/USBTiny-USBtinyISP-AVR-ISP-programmer-for-Arduino-bootloader-Meag2560-uno-r3-/192010044348?hash=item2cb4b0c3bc:g:CO8AAOSwHsRYEXJ7)) and re-write the bootloader.  Arduino and your board are setup to make this an easy thing to do.

VladDracul, 2017-06-16:
Laptop doesn't even recognize it and Arduino doesn't recognize it. (sorry, I should have posted this info as well in my last one...). I tried upload anyway, but it didn't work either. 

I'll definitely order that USBtinyISP, is there any instructions out there on how to use it? Thanks!

Bob-the-Kuhn, 2017-06-17:
The X18 6 pin ICSP header is what the USBtiny ISP connects to when programming the 2560.   The ICSP-1 X17 is for the USB interface 32u2.

There's a jumper on the board.  When installed it connects the USB cable's +5V to the main board.  I expect you have an external supply connected to the mini-RAMBO so the jumper should be removed.

When you plug a USB cable into the USBtinyISP it will NOT show up as a com port on the PC.

To burn the bootloader (and erase the board) 
1. Connect the USBtinyISP to the board.  On my 2560 plugging it in backwards didn't do any harm.
2. Connect a USB cable to the USBtinyISP
3. In Arduino - select the board 
4. In Arduino - select the programmer
5. In Arduino - click on Burn Bootloader

Burning the Arduino generated image via the USBtinyISP is more complicated.  Here's what I do:
- Copy the image, the AVRDUDE program and my handi-dandi crib sheet into one directory.
- Open up a command window and navigate to the directory
- Hook up the USBtinyISP
- Consult the crib sheet on what commands to paste into the command window.

To get the image 
- Compile via Arduino as you normally would but do NOT close Arduino.  You can use either the VERIFY or the UPLOAD option.
- Navigate to C:\Users\your_name\AppData\Local\Temp
- In there you'll see some arduino_build_xxxxx directories.  Open the one with the time stamp from the compile.
- In that directory there are usually two .HEX files.  The xxxxxx.ino.with_bootloader.hex is preferred if it's available.  The xxxxxx part is the name of the .INO file used in the compile.
- Once you've copied the .HEX file to the destination directory you can close Arduino if you want.

What I normally do is get the fuse bytes first and then include them in the burn command.

Here's the files you'll need.
[mega 2560.zip](https://github.com/MarlinFirmware/Marlin/files/1082359/mega.2560.zip)

VladDracul, 2017-06-25:
Alright, 

I got the USBTinyISP in, followed your directions for burning the boot loader, and I get this message from Arduino when trying to burn the bootloader:

Error while burning bootloader: missing 'bootloader.tool' configuration parameter

Is there supposed to be that file in with Marlin?

Bob-the-Kuhn, 2017-06-25:
Arduino should be setup this way to burn the bootloader.

[image]

If that's the way you set it up then there's something wrong with the Arduino setup.  You'll need to re-install Arduino.

Actually, you might be better off trying to see if there is any life in the 2560.  Do the "Burning the Arduino generated image via the USBtinyISP " section but only do the` avrdude -p atmega2560 -c usbtiny -v` script.  That'll tell you if there's any life in the CPU.

VladDracul, 2017-06-26:
Oh, that!  I've been selecting RAMBo as the board. I will try it later tonight, then!

VladDracul, 2017-06-26:
Alright, tried that set up, and got this error:

avrdude: verification error, first mismatch at byte 0x3e000
         0xff != 0x0d
avrdude: verification error; content mismatch
Error while burning bootloader.

Uhhh, I didn't remove any jumper, is that the cause? I can't find the jumper on any schematics or the mini-RAMBo wiki.

Bob-the-Kuhn, 2017-06-27:
As long as the controller has power then the USBtinyISP should be able to program the bootloader.

Did you try the AVRDUDE method to see if there's any life in it?

VladDracul, 2017-06-27:
This what I got from doing that (considering I did that right):
[image]

Bob-the-Kuhn, 2017-06-27:
Yes, you did it correctly.  Those are the expected values.

There is some life in the CPU.   Try burning an image & see what happens.

VladDracul, 2017-06-27:
Success? What now? 

[image]

Bob-the-Kuhn, 2017-06-28:
I take it that the controller still doesn't work.

It say's that the FLASH failed to verify at the very first byte.

I'm afraid it's time to throw the controller in the trash.

closed 2018-02-09 by thinkyhead