MarlinFirmware/Marlin #7066: Compiler error when activate the LCD_BED_LEVELING

fsadak, 2017-06-15:
sketch\ultralcd.cpp: In function 'void lcd_bed_leveling()':

ultralcd.cpp:1774: error: 'lcd_refresh_zprobe_zoffset' was not declared in this scope

         MENU_ITEM_EDIT_CALLBACK(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX, lcd_refresh_zprobe_zoffset);

                                                                                                                                   ^

sketch\ultralcd.cpp:245:30: note: in definition of macro '_MENU_ITEM_PART_2'

         menu_action_ ## TYPE(__VA_ARGS__); \

                              ^

sketch\ultralcd.cpp:297:53: note: in expansion of macro 'MENU_ITEM'

   #define MENU_ITEM_EDIT_CALLBACK(type, label, ...) MENU_ITEM(setting_edit_callback_ ## type, label, PSTR(label), ## __VA_ARGS__)

                                                     ^

sketch\ultralcd.cpp:1774:9: note: in expansion of macro 'MENU_ITEM_EDIT_CALLBACK'

         MENU_ITEM_EDIT_CALLBACK(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX, lcd_refresh_zprobe_zoffset);

         ^

ultralcd.cpp:1774: error: 'lcd_refresh_zprobe_zoffset' was not declared in this scope

         MENU_ITEM_EDIT_CALLBACK(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX, lcd_refresh_zprobe_zoffset);
                                                                                                                                   ^

sketch\ultralcd.cpp:249:102: note: in definition of macro '_MENU_ITEM_PART_2'

         lcd_implementation_drawmenu_ ## TYPE(encoderLine == _thisItemNr, _lcdLineNr, PSTR(LABEL), ## __VA_ARGS__); \
                                                                                                      ^

sketch\ultralcd.cpp:297:53: note: in expansion of macro 'MENU_ITEM'

   #define MENU_ITEM_EDIT_CALLBACK(type, label, ...) MENU_ITEM(setting_edit_callback_ ## type, label, PSTR(label), ## __VA_ARGS__)
                                                     ^

sketch\ultralcd.cpp:1774:9: note: in expansion of macro 'MENU_ITEM_EDIT_CALLBACK'

         MENU_ITEM_EDIT_CALLBACK(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX, lcd_refresh_zprobe_zoffset);
         ^

exit status 1
'lcd_refresh_zprobe_zoffset' was not declared in this scope

Bob-the-Kuhn, 2017-06-15:
I can't reproduce the problem.

Please ZIP up your config files and drop it on your reply.  That gives me the best chance of reproducing the problem.

fsadak, 2017-06-15:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1078799/Configuration.zip)

Thank you...

Bob-the-Kuhn, 2017-06-15:
Try this [file](https://www.dropbox.com/s/x30y2wccxda92zb/ultralcd.zip?dl=0).

I've moved the definition of 'lcd_refresh_zprobe_zoffset' to before the first place it's used.  It compiled without errors after that.

fsadak, 2017-06-16:
Compiler error changed for me like this;

sketch\ultralcd.cpp: In function 'void lcd_info_printer_menu()':

ultralcd.cpp:3368: error: 'MSG_MESH_LEVELING' was not declared in this scope

Bob-the-Kuhn, 2017-06-16:
I'm still not getting a compile error.

What version of Marlin are you using?

fsadak, 2017-06-16:
I am using 1.1.3...

Bob-the-Kuhn, 2017-06-16:
Looks like they've fixed it in bugfix-1.1.x.

I've created PR #7072 to implement the changes in the file I posted.

Bob-the-Kuhn, 2017-06-16:
I think I know what the remaining problem is.  I gave you a file based on bugfix-1.1.x.  It should have been based on 1.1.3.

Try this [file](https://www.dropbox.com/s/o768eb24fac0i59/ultralcd%201.1.3.zip?dl=0).  It's from 1.1.3 with the mods added to it.

closed 2018-02-07 by thinkyhead