MarlinFirmware/Marlin #7067: RFE: Z Proximity Probe Auto Touch in UBL

Andrei-Pozolotin, 2017-06-15:
for automatic z-offset measurement in UBL:
this is a request to back port into UBL the "Active leveling - Touching the bed"
https://ultimaker.com/en/community/23463-inside-the-ultimaker-3-day-6-active-leveling
[image]

Roxy-3D, 2017-06-15:
I read the full text and all the comments.   I'm a little bit confused.  It appears they are probing the bed at three points (two in the front and one in the back).   And then they are using those three points to physically adjust the height of the bed.  Is that correct?

Because if that is correct, there are already tools that do the same thing (but better) in UBL.   For example, G29 P1 U   will probe the outside edges of the bed until you tell it to stop.   You can then use that information to physically adjust the bed, and do it again to see how much more correction is needed.

Andrei-Pozolotin, 2017-06-15:
UBL is much better then U3 in terms of leveling. 

Main point in this RFE is to add support for "auto-kiss" 
head-to-plate-touch-based approach to measure the z-offset as part of UBL.

I works like this:
1) slowly move plate(Z) towards head(E1) (blue line above), expecting the touch
2) monitor analog output of inductive proximity sensor (red line above), expecting the touch
3) detect sensor output pattern change (red line, right side), claim measured Z  (blue line, right side), stop

Roxy-3D, 2017-06-15:
Both Ulti-Maker and Prusa have analog sensors to make bed measurements now.   It would be good to get out in front and figure out how to embrace this technology.

boelle, 2019-02-19:
do prusa use analog sensors? as far i know they just use an inductive probe that is either on or off

and i have recommended support for analog sensors and at least it would require 32 bit to be fast enough to make sense. 

and so far the only analog sensor i know of is the IR ones and those need a matt black surface to work optimal. 

i have an mk42 clone bed from spain and the white silkscreen print made it very dificult to use ir

sooo.... if anything i would vote this down at the veryy bottom of the list if there is even still interest