MarlinFirmware/Marlin #7069: [Support Question] 32-Bit Marlin status

mperdue, 2017-06-15:
OK, i lost a controller board in the printer today and I really don't want to spend money on another 8-bit controller since I plan to switch to 32-bit across the board. So, again, I'm wondering where things stand with the 32 bit port of Marlin. Which files are the most recent version? What version is is based on? What major features are missing? What controllers are supported?

I currently have an Arduino Due that I could put a RADDS board on and set things up with it. Or I could temporarily switch to a different firmware and come back to Marlin on that printer down the line when the 32-bit port is more mature.

What do you guys think?

p3p, 2017-06-15:
I have been keeping the HAL (32-bit version) up to date with bugfix since January and hopefully soon it will be merged (see #7028 some discussion). There is a DUE Platform in the HAL currently but I haven't used it myself so can't say how functional it is, @teemuatlut has created the Teensy (3.5-3.6) Platform, and I'm currently working on the Re-Arm (LPC1768) Platform. I can't really say what the best option for you is, but I think @Roxy-3D and @thinkyhead are quite excited about the Re-Arm so that will probably end up being well supported, but this is all going to be very much in development (and probably unstable) for a while yet.

thinkyhead, 2017-06-15:
I'll try to ensure that we have at least one 32-bit platform supported for the 1.2.0 release. That's coming soon, and will also have code split up into more sub-folders!

mperdue, 2017-06-15:
The Re-Arm to the X5 from Panucatt Devices both look like good options but the failing board (that put me in this bind) is an X3 Pro. Ive had 3 out of four fail in the last year. Lets just say I'm weighing my options. Unfortunately, I need to do something pretty quickly as I'm in the middle of a big design and build project.

Roxy-3D, 2017-06-15:
> I'll try to ensure that we have at least one 32-bit platform supported for the 1.2.0 release. 

If we can only twist your arm into doing one... please consider making it the Re-ARM board.

p3p, 2017-06-15:
>  Ive had 3 out of four fail in the last year

Sounds like there could be a fault with some other part of your electronics causing the controller to fail, how are they failing?

mperdue, 2017-06-15:
I would agree if they were failing in the same device and/or in the same way but they aren't. The first one had a problem when I received it and Roy replaced it. The second one had a USB port failure. The most recent one, this morning, doesn't power up. However, in typing this it occurs to me that I can use the one with the bad USB port and print from the microSD card. I'll have to burn the code with a programmer. Will that work? Will Marlin run if loaded with a programmer?

Bob-the-Kuhn, 2017-06-15:
Most definitely Marlin will run if a programmer is used to update the FLASH.  I've used the USBtinyISP for this many times.

mperdue, 2017-06-15:
Well, this may buy me some time and get me printing on both printers again. :-)

p3p, 2017-06-15:
If it's just the FTDI chip that's gone then you could get a USB FTDI cable and connect it to the hardware serial pins on the ATmega 2560 to bypass it.

mperdue, 2017-06-15:
And the plot thickens... It looks like maybe the problem is with the two heated beds on my printer. The specified resistance is 1.3 ohms which would have them peaking at about 9.25 amps each or 18.5 amps total. However, over time the resistance of the beds seems to have changed a little, dropping to 1.1 ohms. This puts them at almost 11 amps each peak. As there is only one sensor it's logical to assume they will both hit that peak at the same time and that will exceed the 20amp capacity of the fuse protecting the circuit. I'll probably need to move the beds off the board and use a couple of SCRs to control them.

mperdue, 2017-06-16:
Ok, parts will be here tomorrow to resolve this problem so I'm good to hold off on 32-bit for now. No need to leave this open.

closed 2017-06-16 by mperdue