MarlinFirmware/Marlin #7073: Strange behavior with UBL G29 P4

dot-bob, 2017-06-16:
I am having a strange issue when fine tuning the UBL mesh.
A little background first, my printer has a 500mm x 500mm bed, BL Touch Smart with a -2mm z_offset.  Right now I have a 45mm mesh inset from the bed edges, and running the latest bugfix-1.1.x as of the morning of 06-16-17.

I can issue a G29 P1 and get the following mesh:
0,9)                                                                   (9,9)
(45,455)                                                                (455,455)
 -0.554   -0.459   -0.106   -0.046    0.028   -0.021   -0.197   -0.087   -0.141   -0.356

 -0.612   -0.498   -0.201   -0.108   -0.022   -0.131   -0.317   -0.214   -0.291   -0.502

 -0.656   -0.545   -0.235   -0.186    0.059   -0.226   -0.404   -0.266   -0.396   -0.628

 -0.715   -0.622   -0.244   -0.229   -0.171   -0.256   -0.451   -0.307   -0.388   -0.691

 -0.734   -0.606   -0.321   -0.259  [-0.180]  -0.198   -0.444   -0.202   -0.459   -0.674

 -0.746   -0.574   -0.349   -0.241   -0.144   -0.219   -0.419   -0.288   -0.446   -0.695

 -0.738   -0.614   -0.246   -0.197   -0.064   -0.136   -0.228   -0.255   -0.360   -0.639

 -0.664   -0.619   -0.294   -0.136   -0.047   -0.138   -0.219   -0.159   -0.294   -0.571

 -0.596   -0.522   -0.198   -0.114    0.021   -0.068   -0.176   -0.010   -0.211   -0.495

 -0.552   -0.412   -0.102   -0.051    0.110    0.036   -0.108    0.051   -0.109   -0.406
(45,45)                                                                    (455,45)
(0,0)                                                                     (9,0)

I can run a G26 just fine and get the expected print pattern, and print models just fine.

Now when I want to fine tune the mesh using G29 P4 T it places the nozzle 3mm or more above the bed and displays the value of the closest mesh point on the display.

For example if I have the nozzle in the center of the bed and issue "G29 P4 T".  It moves to the nearest point in the mesh above [-0.180] displays that value on the screen
and the nozzle more than 3mm above the build plate.  If I move the nozzle to the bed and press the encoder wheel to save I get the following mesh:

0,9)                                                                   (9,9)
(45,455)                                                                (455,455)
 -0.554   -0.459   -0.106   -0.046    0.028   -0.021   -0.197   -0.087   -0.141   -0.356

 -0.612   -0.498   -0.201   -0.108   -0.022   -0.131   -0.317   -0.214   -0.291   -0.502

 -0.656   -0.545   -0.235   -0.186    0.059   -0.226   -0.404   -0.266   -0.396   -0.628

 -0.715   -0.622   -0.244   -0.229   -0.171   -0.256   -0.451   -0.307   -0.388   -0.691

 -0.734   -0.606   -0.321   -0.259  [-3.345]  -0.198   -0.444   -0.202   -0.459   -0.674

 -0.746   -0.574   -0.349   -0.241   -0.144   -0.219   -0.419   -0.288   -0.446   -0.695

 -0.738   -0.614   -0.246   -0.197   -0.064   -0.136   -0.228   -0.255   -0.360   -0.639

 -0.664   -0.619   -0.294   -0.136   -0.047   -0.138   -0.219   -0.159   -0.294   -0.571

 -0.596   -0.522   -0.198   -0.114    0.021   -0.068   -0.176   -0.010   -0.211   -0.495

 -0.552   -0.412   -0.102   -0.051    0.110    0.036   -0.108    0.051   -0.109   -0.406
(45,45)                                                                    (455,45)
(0,0)                                                                    (9,0)

Has anyone seen this behavior?

Roxy-3D, 2017-06-16:
The nozzle movement is not for the purpose of calibrating the nozzle's position.    You should be using the number on the LCD display to adjust a point.    Remember....   the nozzle is position well above the bed during the interactive editing.   If you move the nozzle the opposite direction, you can get the mesh point to be +10.000mm above the bed if you want.

dot-bob, 2017-06-16:
Thanks,  I assumed it worked similar to a G29 P2.  Maybe as a option have a flag to calibrate by placing the nozzle to the bed like a G29 P2.  Then you wouldn't need to nessisarliy need to run a G26  if you knew you had a trouble spot in a certain area.  So G29 P4 X?.??? and it places the nozzle close to the bed within 0.3mm (or they could specify the specific height),  Then you would could move dial to have the nozzle just touch the bed like in a G29 P2.  Then it would lift the head 5mm+ between points to prevent crashing into the bed or the G26 pattern that may be printed on the bed.

dot-bob, 2017-06-17:
@Roxy-3D,

From what you have described the comments in the ubl_G29.cpp file should probably be updated:
```
   *   P4    Phase 4    Fine tune the Mesh. The Delta Mesh Compensation System assume the existence of
   *                    an LCD Panel. It is possible to fine tune the mesh without the use of an LCD Panel using
   *                    G42 and M421; see the UBL documentation for further details.
   *
   *                    The System will search for the closest Mesh Point to the nozzle. It will move the
   *                    nozzle to this location. The user can use the LCD Panel to carefully adjust the nozzle
   *                    so it is just barely touching the bed. When the user clicks the control, the System
   *                    will lock in that height for that point in the Mesh Compensation System.
```

Or perhaps add an option 'S' to fine tune with surface of the bed.  So add something like the following to unified_bed_leveling::fine_tune_mesh(...)

```
       if (!parser.seen('R'))    // fine_tune_mesh() is special. If no repetition count flag is specified
        g29_repetition_cnt = 1;   // do exactly one mesh location. Otherwise use what the parser decided.
     
>      #ifdef UBL_MESH_EDIT_MOVES_Z
>        bool is_surface = parser.seen('S');
>      #endif
      
      mesh_index_pair location;
```

and down lower

```
        do {
          new_z = lcd_mesh_edit();
>          #ifdef UBL_MESH_EDIT_MOVES_Z
>            if(is_surface)
>            	do_blocking_move_to_z(new_z);
>            else
>              do_blocking_move_to_z(Z_CLEARANCE_BETWEEN_PROBES + new_z);  // Move the nozzle as the point is edited
>          #endif
          idle();
        } while (!ubl_lcd_clicked());
```

Tannoo, 2017-06-23:
Can't N be used to specify the edit height you want?

Roxy-3D, 2017-06-23:
> Can't N be used to specify the edit height you want?

Not without provoking the GCode Nazi's...

Tannoo, 2017-06-23:
Lofl!

Tannoo, 2017-06-23:
I am sorry... I meant H... not N. I must have been tired.

Roxy-3D, 2017-06-23:
> Or perhaps add an option 'S' to fine tune with surface of the bed. So add something like the following to unified_bed_leveling::fine_tune_mesh(...)

@dot-bob   So are you thinking with the 'S' option that the nozzle starts at a 0.000mm height?   There does need to be some height between the nozzle and the bed.

dot-bob, 2017-06-23:
> There does need to be some height between the nozzle and the bed.
@Roxy-3D I was thinking starting at the mesh height.  So if their mesh point is 0.000mm then yes.  If they did actually perform a G26 their nozzle was already at a height of the mesh + layer height.  The idea would be to place the nozzle on the surface of the bed like the current comments for G29 P4 suggest.  If we determine we need a bit of breathing room place it at the mesh height + 0.2mm like G26 for layer height.

I did submit a pull request last night #7124 perhaps change the S to an L and use it like the L in G26 where you can specify layer height and if ho height is specified default to 0.2mm.

Roxy-3D, 2017-06-24:
> I did submit a pull request last night #7124 perhaps change the S to an L and use it like the L in G26 where you can specify layer height and if no height is specified default to 0.2mm.

Yes.  I saw it and looked at it.  But I need a little help understanding the purpose.  Why do you want to edit a mesh point with the nozzle so close to the bed?  What is the benefit of doing that?

When I am setting up a mesh for a new piece of glass, I move the nozzle to an area of the G26 Mesh Validation Pattern that needs adjustment, and just start editing points.  I don't even look at the LCD Display.   I just turn the dial more or less and one way or the other.  

Why would I want the nozzle very close to the surface of the bed and moving up or down so it can crash into the bed?

dot-bob, 2017-06-24:
The idea is to be able to use the bed or a feeler gauge as a reference.  Think of it as a way to probe any point on the mesh manually like P2.  You wouldn't necessarily need to use G26 to tune the mesh.  If you knew you had a bad area you could issue the command then pull out your feeler gauge and fix a couple of points.

When the nozzle is floating in the air above the bed it is hard to gauge if I am setting the correct height other then looking at the G26 print and making a best guess.

Roxy-3D, 2017-06-24:
OK...  But help me understand...  That works if the mesh is a little bit too high.  But what happens if the mesh point is a little bit too low.   Won't that drive the nozzle into the bed?

Bob-the-Kuhn, 2017-06-24:
The G26 results are the ultimate truth.  

During the fine tuning you'll be making adjustments on 0.05mm or less.  You just can't reliably eyeball it or use a shim to gauge that fine of a distance.

dot-bob, 2017-06-24:
> OK... But help me understand... That works if the mesh is a little bit too high. But what happens if the mesh point is a little bit too low. Won't that drive the nozzle into the bed?

By that argument wouldn't G26 be ramming into the bed?  If that is the case then specifying a starting height is needed, but the end result would be to save the mesh point when the nozzle in on the surface of the bed or a shim.  P4 should only be used for fine tuning the nozzle height.  If crashing into the bed is a problem or the mesh is off by more than 0.5mm they need to address probe, probe offset, and or mechanical issues and start over at P0-P1. 

I am not suggesting doing away with the current method just have an alternate method that behaves like P2 and the current code comments for P4 suggest.

Roxy-3D, 2017-06-24:
> I am not suggesting doing away with the current method just have an alternate method that behaves like P2 and the current code comments for P4 suggest.

Yes.  I understand.  But there are several problems with putting the nozzle right next the bed at the start of the edit cycle for a mesh point.   First, if there is any error anywhere in the machine's configuration, there is no margin left to keep the nozzle from hitting.  But the other problem is you can only go up at that point.  You can't go down.   

If you really are comfortable operating with the nozzle that close to the bed when you are trying to edit a mesh point, the capability is already there to do that.   The nozzle is moved to the various mesh points as they are edited with:
```cpp
    do_blocking_move_to_z(Z_CLEARANCE_BETWEEN_PROBES);
    do_blocking_move_to_xy(lx, ly);
    do {
      location = find_closest_mesh_point_of_type(SET_IN_BITMAP, lx, ly, USE_NOZZLE_AS_REFERENCE, not_done, false);
```
You can set your Z_CLEARANCE_BETWEEN_PROBES to 1 if that really makes sense.   If you really want to be closer than that (.25 mm ????) we can change this to be a float:
```cpp
#define Z_CLEARANCE_BETWEEN_PROBES  3 // Z Clearance between probe points
```

fiveangle, 2017-06-25:
> During the fine tuning you'll be making adjustments on 0.05mm or less. You just can't reliably eyeball it or use a shim to gauge that fine of a distance.

A bit off topic from this thread, but I actually found a hacky way but ingenious way to make shim measurements reasonablly accurate down to approx 0.02 using custom controls in Octoprint... 

The controls move the nozzle up/down 100, 10, 5, and 1 micron increments:

[image]

Then when you get down to where you've found the 5µm limit, you move up 5µm, move up 1µm, then move back up/down/up/down 5µm while feeling shim.  In this way, you can find nearly down to the 1µm level where it drags differently because you are comparing +5µm to the point you are looking for (instead of +/-1µm which is impossible by human touch.

It's works pretty damn well.

Custom controls config for Octoprint yaml config:

[abl_cust_controls.yaml.txt](https://github.com/MarlinFirmware/Marlin/files/1100112/abl_cust_controls.yaml.txt)

Roxy-3D, 2017-06-25:
Wow!  Good thinking @fiveangle !   Please let me think about this for a few days.   I suspect I'll be twisting @Tannoo 's arm for a couple of small changes....

dot-bob, 2017-06-25:
> You can set your Z_CLEARANCE_BETWEEN_PROBES to 1 if that really makes sense. If you really want to be closer than that (.25 mm ????) we can change this to be a float:

The problem with Z_CLEARANCE_BETWEEN_PROBES it is used elsewhere.  If I was to reduce it much below 5 my nozzle may crash into my bed clips.

I have updated the pull request #7124.  I changes 'S' to 'H' and now use H to specify an offset like Z_CLEARANCE_BETWEEN_PROBES.  I have the allowable range for offset set from 0-10mm.

closed 2017-06-27 by dot-bob