MarlinFirmware/Marlin #7074: Which motherboard need to chose in configuration for MKS SMELZI V1.0?

gapx15, 2017-06-16:
Dear,

does anyone have MKS SMELZI V1.0 board? I do not know which motherboard I need to chose in Marlin  configuration to flash new version of Marlin on my board. 

BOARD_MELZI or BOARD_MELZI_MAKR3D?

Please help me guys.

Thanks in advance.

Bob-the-Kuhn, 2017-06-16:
Your best bet is to choose BOARD_MELZI.

----

Did I mention that I hate MKS?  Almost no technical documentation.

Seriously - the Melzi type boards just don't have the FLASH space to support the latest Marlin options and an LCD.  If you get seriously into 3D printing you'll want the options and the convenience of an LCD.

Your best course of action is to ~dump all MKS products into the trash &~ get a Mega2560 & RAMPS & RepRap Discount full graphics LCD package.  More FLASH, more IO and a lot more support for only a few dollars more.

Maybe your vendor will allow you to exchange the Smelzi system for a RAMPs system.

Did I mention that I hate MKS?  Almost no technical documentation.

.

closed 2017-06-16 by gapx15

Bob-the-Kuhn, 2017-06-16:
If you have trouble with the SMELZI you can try some of the M43 tools to figure out how it's physical I/O pins map to logical pins.