MarlinFirmware/Marlin #7075: [Feature Request] Printhead-Dock tool changer support

marcossfernandez, 2017-06-16:
Hi all:

I'm trying to do a 3 independent printhead 3Dprinter. I succesfully made the tool change hardware and host script  to do the tool change when needed (two dock example):

https://goo.gl/photos/yJVhjMbWb518j42j6

This printer work well with the tool change script from Simplify3D software.
Accuracy and repeatibility are great, having less moving mass and one x-y-z-offset for all heads. No blobs, dripping or contamination problems affected. The printed parts (head support, dock) will be made on CNC Mill for better matting accuracy.

I'll like to do a 3+ fixed docks and 3+ detachable heads as shown in the video. 

It will be desirable to implement to Marlin fw the necesary function to know where the tool is docked, know what tool need to park and where, and who to catch at tool switch operation. For now I depend on scripts based on host software, and a better solutions would be being hardcoded.

I think it will be a good implementation to multi print head printers. Ultimaker forumdeveloped a system similar to this aproach: Ultimaker-mark2 (Mark2 is an asymmetrical multi-extrusion upgrade for the Ultimaker 2) but with a fixed head and aditional one in tandem. The software develop to integrate it's very good.

Any help is welcomed to achieve the goal. Thank you!

- Sorry, English is not my native language.-

Marcos

mperdue, 2017-06-17:
OK, take my money - I want this...

marcossfernandez, 2017-06-27:
Thanks mperdue!

Still need some help to do the code to integrate in Marlin Fw, because I'm not programmer. Simplify3D or Cura can integrate some tool change scripts but would be better to do it host independent.

marcossfernandez, 2018-04-24:
We already achieved to get the firmware to control the docks and the printheads. Its slicing software independent and it can do any pre or post tool change scripting, or independent XYZ printhead offsets. Still developing some tweaks and sanitize the code but actually is working flawlessly in our CoreXY multi tool 3d printer. E3D guy's showed a similar idea few days ago.
[https://www.youtube.com/watch?v=sLgGDq47K1c](url)

When we finish these code will be released as a fork to Marlin.

kevinmce, 2018-07-12:
I am interested in this feature also, @marcossfernandez is there are fork already available that could be tested?
Or maybe @thinkyhead there was some progress on your side?