MarlinFirmware/Marlin #7076: Halt for Code Rework, Marlin 2.0

thinkyhead, 2017-06-16:
For a long while, due to updates in the Arduino toolchain we've no longer had to use a flat file hierarchy, but it seemed proper to retain Marlin 1.1.x in a flat arrangement to keep it the "spiritual successor" to the Grbl/Sprinter-based Marlin and make it easier to compare to Marlin 1.0.x and follow changes.

Now that Marlin 1.1.x is out in the wild, this is a good time to halt new development to Marlin and switch to the hierarchical file layout, incorporating the HAL at the same time, and set up the basis for future Marlin development.

It's not easy to stop the flow of issues, and we want to keep fixing any bugs that exist in 1.1, of course. So I propose that we take care of the current crop of obvious bugs —and only bugs— do a 1.1.4 release very soon, and one more release —1.1.5— just before we release 1.2.x, the new hierarchical + HAL version of Marlin.

- Stop all new development.
- Focus on bugs in the issue queue.
- Release 1.1.4 in the next day or two to address 1.1.3 issues.
- After 5-10 days, release 1.1.5… aim to clear all obvious bugs.
- Collaborate around #7028 to get the HAL in place.
- At the same time, @thinkyhead will put together the new 1.2.x branch with hierarchical layout.
- Next, we work on the hierarchical layout to make sure it serves user/developer needs.
- Release 1.2 (or Marlin 2.0?) with the hierarchical layout, support for some 32-bit boards.

Roxy-3D, 2017-06-16:
> It's not easy to stop the flow of issues, and we want to keep fixing any bugs that exist in 1.1, of course. So I propose that we take care of the current crop of obvious bugs —and only bugs— do a [...]

@Tannoo and I have some clean up work going for the LCD header files.   As it turns out...  The .h files can only be included by one .cpp file because the .h files generate code.   While this is not a 'bug fix' it is an important item from a 'code cleanliness' perspective.

We have the Graphical LCD Display code clean up done.   The 20x4 LCD Display clean up should progress quickly.     https://github.com/Tannoo/Marlin/tree/ultralcd_impl_DOGM.cpp

Tannoo, 2017-06-16:
https://github.com/Tannoo/Marlin/tree/LCD_Interactive_Mesh_Edit_Refinements

teemuatlut, 2017-06-16:
> Release 1.2 (or Marlin 2.0?) with the hierarchical layout, support for some 32-bit boards.

Go with v2.0.
Support for 32bit platforms will indicate a new era for Marlin and for sure will warrant a tick in the major version count.

Roxy-3D, 2017-06-16:
> We have the Graphical LCD Display code clean up done. The 20x4 LCD Display clean up should progress quickly. https://github.com/Tannoo/Marlin/tree/ultralcd_impl_DOGM.cpp

Both the Graphical LCD Panel and the 20x4 LCD Panel .h files are cleaned up in this branch...    

Tannoo is going to get the indentation and white space cleaned up....

thinkyhead, 2017-06-16:
> The .h files can only be included by one .cpp file because the .h files generate code.

That is correct. The LCD-oriented header files may only be included once, and only in `ultralcd.cpp`. They are essentially just `.cpp` files, but given `.h` extensions so that they don't compile outside of `ultralcd.cpp`.

thinkyhead, 2017-06-17:
> The .h files can only be included by one .cpp file because the .h files generate code.

That is correct. The LCD-oriented header files may only be included once, and only in `ultralcd.cpp`. They are essentially just `.cpp` files, but given `.h` extensions so that they don't compile outside of `ultralcd.cpp`. We could make them into proper `.cpp` files and fix the unorthodox usage, without any negative affect on performance.

ImplementOfWar, 2017-06-17:
Is the new file structure going to allow Marlin to use a single file for Configuration?  Much like Smoothieware?  Is that one of the immediate benefits of doing this,  apart from 32 bit and HAL
?

thinkyhead, 2017-06-17:
@ImplementOfWar That has not been discussed yet. Smoothieware is able to do certain things by limiting itself to 32-bit boards which have far greater resources. When and if we implement a configuration system in emulation of Smoothieware, it will be for 32-bit boards only.

Roxy-3D, 2017-06-17:
@thinkyhead   What are you thinking for a time table to lock down the code?   

I thought I could get the ultralcd_impl_HD44780.h seperated into declarations and code generation...  But it is turning out to be worse than I expected.   We have the DOGM stuff seperated, so that is a start.  But the 20x4 LCD code is especially bad.    It is very very ugly...

So, we could just go with what we have...   Or... You can take a look at where we are at:  https://github.com/Tannoo/Marlin/tree/ultralcd_impl_DOGM.cpp  and see if you can figure out the issues?

Tannoo, 2017-06-17:
>You can take a look at where we are at: https://github.com/Tannoo/Marlin/tree/ultralcd_impl_DOGM.cpp

https://github.com/Tannoo/Marlin/tree/LCD_Interactive_Mesh_Edit_Refinements

Tannoo, 2017-06-17:
I should delete the ultralcd_impl_DOGM.cpp branch. It was created as a test bed for the splitting of the DOGM.h file.

petrzjunior, 2017-06-18:
We should probably ping the whole @MarlinFirmware/language-team to complete translations.

fixoid, 2017-06-18:
Do I understand correctly that the new features will not be added before the release of version 2.0 (1.2)?
If so, when can we expect 2.0?
What about the PRs that were created before the decision was made?

Roxy-3D, 2017-06-18:
> What about the PRs that were created before the decision was made?

In general...  The Pull Requests generate for the current organization that do not make it into the new file layout should not be that difficult to regenerate.   The bulk of the files are not going to change.  What may change are things like paths to header files or stuff like that.   If  your Pull Request doesn't make it to the new layout automatically...  It will be very straight forward to regenerate it.

thinkyhead, 2017-06-19:
>  thought I could get the ultralcd_impl_HD44780.h separated into declarations and code generation... But it is turning out to be worse than I expected.

The `ultralcd_impl_*` files were never designed to be standalone, but are really just split out parts of `ultralcd.cpp`. It would be best to wait till the ultralcd code is converted to a singleton class, and rearrange anything needing rearranging at that time. It will make more sense at that time, as `ultralcd_impl_*` may end up as two instances of a helper class. Or there may be other separation of functionality into more classes. It's something I want to work on more comprehensively, immediately after rearranging the code. And I'm pretty sure the menu code can be simplified somewhat also.

thinkyhead, 2017-06-19:
> Do I understand correctly that the new features will not be added before the release of version 2.0 (1.2)? If so, when can we expect 2.0? What about PRs that were created before the decision was made?

There's nothing to be done about it. There will never be some point in time where every PR will be merged, so it's best to stop the flow of new PRs and work on the PRs that we already have in front of us. I have been trying to get as many merged as possible, but more keep coming in. And so, we just have to go for it and it's just bad timing for any PRs that will need to be reworked.

As it is, it will be me who has to re-work all the existing PRs into the new layout, because I have the mojo to make that happen as quickly as possible, and I will have all contingencies freshly in mind.

thinkyhead, 2017-06-19:
> What are you thinking for a time table to lock down the code?

Let's keep fixing bugs all the live long day. Simply don't merge new stuff, and put off new features. If any new features come in while we're working on the transition, we'll simply put them off. Most contributions will be straightforward enough that it will be easy to re-make them for the new layout. Some we may have to work on within the old layout first, then when ready also bring them to the new layout.

We should continue to fix bugs in the 1.1.x branch and I will make sure they are brought into the new layout as well.

As for a timetable, I think we need to go in two stages. First, since we have so much work being done to make the 32-bit code ready, perhaps we should get that into minimal working shape first. Then I can include a genuine HAL as part of the initial 2.0.x branch.

The layout conversion itself isn't hard. The first conversion will take me less than a day to complete, and then we can go over it and see where it needs tweaking.

I'd like to focus on going over the open issues for the next day or two, patch up as many clear bugs as possible, and merge as many of the lingering PRs as possible. One of PRs contains the 32-bit HAL rebased on the latest `bugfix-1.1.x` and that will also need to be brought up to date — though not merged with any Marlin 1.1.x.

Once the HAL is working and we're happy with it, then I'll rebase it onto the latest code and use that as the starting-point for Marlin 2.0 with the new file layout.

I think it's a good idea to take the next few days and close as many issues as we can so we have a fresh start.

Yes, it's possible that we could have a 2.0 layout and initial release by the end of the week… but let's loosen up the schedule so we can clear as many issues as possible and just aim to have it ready by July 1. How does that sound?

Roxy-3D, 2017-06-19:
> Yes, it's possible that we could have a 2.0 layout and initial release by the end of the week… but let's loosen up the schedule so we can clear as many issues as possible and just aim to have it ready by July 1. How does that sound?

**_# July 1st sounds fine...   Especially if there is a sample 32-Bit HAL limping along and able to do move the nozzle around!!!_**

p3p, 2017-06-19:
@Roxy-3D I'm pretty sure all the platforms currently in the HAL can do that, if the board supports the Arduino framework it's fairly trivial to implement basic features, the main stumbling block for non Arduino framework devices is the extra features needing Arduino libraries or Classes. I should be able to print tonight on my Re-ARM but the HAL needs further extended to strip all Arduino based features out of the main code and behind an API this will need to be done incrementally once we get the HAL merged.

Roxy-3D, 2017-06-19:
> I should be able to print tonight on my Re-ARM but the HAL needs further extended to strip all Arduino based features out of the main code and behind an API this will need to be done incrementally once we get the HAL merged.

What classes and functions are you missing?   Would it be possible to just pull in the Arduino libraries so we don't have duplicated code complicating things?

p3p, 2017-06-19:
Things like Print and Wire (as two random examples), just generally anything that is from the Arduino framework will need to be abstracted, this wont lead to code duplication and should lead to much cleaner and more understandable code, anything that can use the Arduino framework still can, although sometimes it may turn out it wasn't even necessary just convenient when it was implemented.

In some cases it may be difficult without introducing some indirection overhead, but anything performance critical already avoids the Arduino framework. It may be a good idea to have folders in the HAL (arduino_framework, mbed_framework) devoted to anything that can be reused among platforms that use that framework, but that's probably discussion for another day.

Roxy-3D, 2017-06-19:
> It may be a good idea to have folders in the HAL (arduino_framework, mbed_framework) devoted to anything that can be reused among platforms that use that framework, but that's probably discussion for another day.

@thinkyhead  @p3p Well...   Right now is a good time to express opinions and recommendations.  It may not be too hard to get the HAL folders setup logically from the very start.   But any subsequent, improved layout and organization for the HAL's will be welcomed also.

p3p, 2017-06-19:
> Would it be possible to just pull in the Arduino libraries so we don't have duplicated code complicating things

In the case of Wire all Arduino based platforms could still use it, but the calls would be through a standard API such as i2c_begin(channel, address), i2c_send_byte(channel, data), that other platforms would implement themselves or through another framework.

tcm0116, 2017-06-20:
What can I do to help with this effort? I've got a bit of free time in the evenings this week, so I'm happy to help out if I can.

Roxy-3D, 2017-06-20:
I think it is a little too early to do anything.  We need the new organization to happen and be in place.   But as soon as that happens...   Several important items are on the list:
- Simple, step by step directions for getting the firmware onto a Re-ARM board.  (Using Cygwin in a Windows environment would be preferable, but if we have to bring up a Linux machine, that would be tolerable.)
-- Any extra configuration required.
-- Steps to compile and build image outside of Arduino
-- Steps to flash including notes on using the 2nd USB port
-- Etc.    
- Simple, step by step directions for getting the firmware onto other 32-Bit boards.
- It maybe good to have a thread showing how far the boot process is working for each 32-bit board type.   And what sub-systems of Marlin still need attention for the board.
- It will also be good to have the 32-Bit developers look at the new layout and suggest changes (and submit Pull Requests) to organize things better for themselves.

I believe there is a lot of pent up demand for an official 32-Bit version of Marlin.  If we can just get one board up and working with the new release, and clear instructions so anybody can duplicate the success...   We should see the a lot of activity.  And my guess is a number of different 32-Bit boards will start to have people using them.

p3p, 2017-06-20:
> Simple, step by step directions for getting the firmware onto a Re-ARM board.

1: Clone  https://github.com/p3p/Marlin.git and checkout branch 32bit-bugfix-1.1.x-LPC1768
2: Install PlatformIO UI
3: Open project in PlatformIO and build for Re-ARM (click bottom left of screen and scroll to bottom of options)
4: Copy firmware (<Marlin Folder>/.pioenvs/Re-ARM/firmware.bin) to an sd card and put into Re-Arm

I've maintained compatibility with the Arduino pin layout so the same pins files should work. 

This is a work in progress so once Marlin is flashed to the Re-ARM you lose access over usb to serial and the micro sd card, you will need another means of accessing the sdcard and an ftdi cable to access the 6 pin hardware serial port on the Re-ARM.

As of today I'm printing with Re-ARM, steppers, heaters, ADC, SD card and graphical controller all work (character displays probably will not, they are more closely integrating with Arduino) using my RAMPS14, so basic functionality is at least out of the way.

Roxy-3D, 2017-06-20:
@p3p  THANK YOU SO MUCH!   I'm going to start messing with getting some simple firmware compiled and loaded into a Re-ARM board using your steps...   I'll at least be in a position to start messing with different things when the time comes!

Roxy-3D, 2017-06-20:
@p3p  Up above you say to 'build for Re-ARM'.    I can't find that in the list.  But I did find Platform NXPLPC NXP mbed LPC1768.    Is that the correct selction for the Re-ARM board?

I'm also having trouble getting the project to open up in PlatformIO.   But I did get something to happen when I dragged and dropped the platformio.ini file onto my Atom icon on the desk top.   A little more info on what I do to open the project would be helpful too!

Grogyan, 2017-06-20:
I too am having trouble.
Though we might need to generate a platform specifically for the Re-ARM
http://docs.platformio.org/en/latest/platforms/creating_board.html

p3p, 2017-06-20:
When you open Atom it should open onto a Home screen one of the Quick Access options is to "Open Project" if you browse to the folder containing platformio.ini and click open, in the left panel you should get the Marlin project file listing (.github, buildroot, ect) from here you should be able to click the PIO build button in the bottom corner and scroll down to Re-ARM

p3p, 2017-06-20:
> Though we might need to generate a platform specifically for the Re-ARM

Using the LPC1768 board with a modified linker script for compatibility with the smoothieware bootloader works well for now but we could do that in the future. The Re-ARM build is defined in the platformio.ini file so as long as you have correctly checked out the branch it should be there.

Grogyan, 2017-06-20:
Thanks Chris,
Posting a couple of screenies to help others

[image: Inline image 3]

[image: Inline image 4]

>
>

Roxy-3D, 2017-06-20:
Ooooops!   I managed to do a Re-ARM Upload.   But I had my Atmega2560 with a RAMPS board plugged in....   My LCD screen went crazy.

THANKS p3p!!!

p3p, 2017-06-20:
Just don't expect too much when (if) you get it on the hardware I only got the heaters working this afternoon ^^ and if you change the config it will probably explode .. I also just noticed an issue with corruption in the serial data during a test so that's my task tonight (he says at 23:30)

Roxy-3D, 2017-06-20:
Yeah...  Don't worry about that!   I'm just trying to get up the learning curve because my guess is when we get the new file layout, progress is going to be made fast.    I don't want to be left behind.

So...   I'm going to be playing with a lot of stuff.   Mostly simple stuff.   But I want to understand all the steps and how the development environment works.   And then...  Who knows, maybe I can help with some of the simpler stuff like getting the 20x4 LCD Displays working.   

The point is, when 32-Bit progress can be made against the current code base, eventually....   everything will be working even though the code base kept evolving as that was happening.   And if people can run the current code base on 32-Bit platforms...   More and more people will jump on-board.

paulusjacobus, 2017-06-21:
Hey @Roxy-3D is that Ray Cortes kickstarter board?
https://www.kickstarter.com/projects/1245051645/re-arm-for-ramps-simple-32-bit-upgrade
I did order it for $48 for my 3D printer and it's gathering dust at the
moment. I would be really interested to see this getting of the ground. I'm
doing a kickstarter for my Gerbil K40 laser controller at the end of the
month so I cannot contribute much but I'm eager to learn how to port code
to this environment. Where can I follow it in detail?

Roxy-3D, 2017-06-21:
It is not Ray's board.  I have that on my desk too.   But I'm trying to learn how to generate code with the PlatformIO IDE.

@p3p  I obviously don't understand something...   I thought I built the firmware OK, and put it on the SD Memory card.  But nothing happened on the Graphical LCD Panel when I started up the board.   So I'm not sure what I'm fighting and decided to just try to make an LED blink.  I added this code to Marlin_main.cpp
```cpp

void pinMode(int pin, int mode);
void digitalWrite(int pin, int pin_status);
bool digitalRead(int pin);
void analogWrite(int pin, int pin_status);
uint16_t analogRead(int adc_pin);

void setup_LEDs() {
int p, r, t, s;

    for(p=1; p<54; p++) {
        pinMode(p, OUTPUT);
	for(r=0; r<10; r++) {
    		digitalWrite(p, LOW);  // drive it down
		s = 0;
		for(t=0; t<5000; t++) {
			s=s+digitalRead(20)&0x01;
	    	}
    		digitalWrite(p, HIGH);  // drive it up
		for(t=0; t<5000; t++) {
			s=s+digitalRead(20)&0x01;
	    	}
    		digitalWrite(p, LOW);  // drive it down 
	}
        pinMode(p, INPUT);
    }
}
```
and then in the setup() function I called it:
```cpp
void setup() {

  #ifdef DISABLE_JTAG
    // Disable JTAG on AT90USB chips to free up pins for IO
    MCUCR = 0x80;
    MCUCR = 0x80;
  #endif

  #if ENABLED(FILAMENT_RUNOUT_SENSOR)
    setup_filrunoutpin();
  #endif

  setup_LEDs();
```
I have LED's plugged into different places on the Re-ARM board.   (Right now, pins 1,2 and 3) and I do see some changes in them when I reset the board.  But I don't get on and off blinks like I expect.   Obviously, I'm missing something.   Can you change the setup_LEDs() so it blinks an LED for me?  With that I'll be able to make a lot more progress.

p3p, 2017-06-21:
Without the serial connection it will be hard to know what went wrong, can you confirm the firmware was written correctly, the Re-ARM renames "firmware.bin" to "firmware.cur" after flashing. Assuming that's correct you should at least get serial output (and the logo if the config is compatible with your setup).

Roxy-3D, 2017-06-21:
> Without the serial connection it will be hard to know what went wrong, can you confirm the firmware was written correctly, the Re-ARM renames "firmware.bin" to "firmware.cur" after flashing. 

Yes...   It is renaming the file.

> Assuming that's correct you should at least get serial output (and the logo if the config is compatible with your setup).

I do not have the board connected up to any endstops or stepper motors yet.  And I maybe fighting the problem where some graphical displays have reversed cables.  I don't know what I'm fighting.   That is why I thought I would start small and try to blink an LED.

It appears you re-mapped the ReARM connector locations to match the RAMPS board locations...

p3p, 2017-06-21:
The functions (digitalWrite, pinMode ect) that I wrote make the pin mapping transparent, you just use them as you would on Arduino Mega, not all pins have been mapped yet (or can be) pinmap_re_arm.h controls that. 

With nothing plugged into the Re-ARM it should still boot up into Marlin and you can control it via the serial port, without more information about what is being output I can't really help, Displays and controllers I wasn't expecting just to work unless they are identical to reprap full graphic smart controller. 

If you don't have a ftdi cable you can forward the serial through an Arduino but I recommend getting one they come in handy.

p3p, 2017-06-21:
I just realised you didn't say if the LCD display was powering up or that you used the adaptor cable from http://www.panucatt.com/product_p/glcd-idc.htm to connect it, all that cable does is split pin 1 off so you can connect it to a 5v pin on the Re-ARM.

If you do want to run code on the Re-ARM without any of Marlin getting in the way, there is a main function in HAL.cpp that calls setup and loop (this isn't Arduino anymore everything is done yourself)

Roxy-3D, 2017-06-21:
I do get the Graphics LCD Screen to light up.   (RepRap Discount Full Graphic Smart Controller)
I have the split cable and apparently did get the +5 volt wire to work.    I just don't have anything being displayed on it (yet).   I'll keep plugging away.

thinkyhead, 2017-06-22:
I'll try to apply the new file layout incorporating the HAL this weekend and post a 2.0.x branch that we can patch up for release. We might still have too many open issues needing fixing before we jump ahead. I haven't counted them lately, but I will do an audit and try to produce a list of lingering issues that we should fix ahead of the new layout.

Roxy-3D, 2017-06-22:
> We might still have too many open issues needing fixing before we jump ahead, but at least this will provide a starting point. I haven't counted open issues lately, but I will do an audit and try to produce a list of lingering bugs and issues that we should fix ahead of the new layout.

@thinkyhead  From reading all the new issues and posts to them...   There doesn't seem to be anything critical.    We have a few questions about BL-Touch probes not working.   We have a couple of bed leveling issues across the spectrum of different bed leveling schemes.  But nothing dire. 

My suggestion would be, we cut and run.    Until the new folder and file layout is complete, people can just continue with today's bugfix-v1.1.x .

Roxy-3D, 2017-06-22:
@thinkyhead  @p3p    I moved the LPC1768 stuff for the Re-ARM board over to 32-Bit-RCBugFix-new just to make sure they don't get forgotten!   We need p3p's LPC1768 stuff in the new file & folder layout.

thinkyhead, 2017-07-27:
Now that the `bugfix-2.0.x` branch has been created, we can try to bring over the elements from oh-so-older `32-Bit-RCBugFix-new` branch, including your new changes in #7121, if they're ready for inclusion and/or consistent with the HAL stuff from @teemuatlut and @p3p.

p3p, 2017-07-27:
@thinkyhead The new branch was based on 32-Bit-RCBugFix-new so I'm relatively sure there is nothing missing that was in there (although I seem to remember there was some issue rebasing @teemuatlut's PIN DEBUG code), I was back-porting anything that was general HAL from the LPC1768 branch into the PR but I do have to finish the new SPI integration (I have to implement it for the other platforms), then I would like to continue moving the HAL features into the HAL::Feature naming convention (HAL::ADC , HAL::I2C, ect) with general headers.

teemuatlut, 2017-07-27:
You can leave the pins debug for Teensies in whatever state. I'll go over it once everything has settled down.

oysteinkrog, 2017-07-28:
@thinkyhead You said in the first post that the aim for 1.1.5 was to fix all obvious bugs.. well dual X carriage has been broken since 1.1.1. See #6956 #7050 #7291.

teemuatlut, 2017-07-28:
I may be interested in finding a fix for this as I'm building a dual X machine myself, but I can only get to it next week so if it's not solved by then I'll take a crack at it. Further discussion should be made on #7291

oysteinkrog, 2017-07-31:
@teemuatlut I've tracked down the problem and I have submitted a PR that fixes it. It's great to have more people with DUAL_X_CARRIAGE though as I'm pretty sure there are still bugs related to it.

teemuatlut, 2017-07-31:
Good to hear.

bruce356, 2017-08-10:
Hi, I have dual X carriage printers with RADDS / Due + RADDS LCD Display with SD card.
What LCD option do I select in Marlin2
Thanks - bruce

klcjr89, 2017-08-10:
Is Marlin 2.0 compatible or intended to be compatible with the Azteeq x5 mini? (LPC1769) 
https://www.matterhackers.com/store/printer-accessories/azteeg-x5-mini-32bit-all-in-one-controller-v1.1

Roxy-3D, 2017-08-10:
> Is Marlin 2.0 compatible or intended to be compatible with the Azteeq x5 mini? (LPC1769)

Right now, the bulk of the 32-bit activity is on the Panucatt Re-ARM board.  That uses a LPC-1768 which is very similar to the LPC-1769.    The way the code is being done, the current work should move over to that board with a minimum of effort.   If pins are assigned differently, they would just get moved around in mapping table.   And the LPC-1769 is faster so some timer speeds would shift.    

Incidentally, Panucatt supplied us with enough Re-ARM boards to get critical mass on the development side.   It kind of feels like they want the X5 (and the X5 mini) to support Marlin so that will probably happen as soon as the Re-ARM board is solid.

mperdue, 2017-08-10:
I'm very happy to hear this. The X5 is the board I want to use. I may have to drop Roy an email to get things rolling on purchasing a couple of them.

Roxy-3D, 2017-08-10:
If you want to order a single X5 board and get it connected up to a printer...   People in this thread should be able to tell you what to edit on the Re-ARM code to make it work with the X5:  https://github.com/MarlinFirmware/Marlin/pull/7390

mperdue, 2017-08-10:
Sounds like a good plan. I'll see about getting one on order shortly. I might have to grab some spare parts and put together a functional 3D printer setup for testing.

Roxy-3D, 2017-08-10:
And probably...   At least from my perspective...   I think it is likely the Re-ARM HAL goes away and we have a single LPC-1768(9) HAL.    Probably, the changes to the Re-ARM code is very minor to make it work on the X5 board.    BUT...   That remains to be seen.

p3p, 2017-08-10:
In theory the firmware will run as is on an LPC1769 they are binary compatible, but the pins will more than likely be completely wrong.

I did take into account that there would be multiple boards running on the LPC17xx HAL (mostly), there are a few things that will need written for the X5 but are already split out in a way that should make it pretty simple. An internal pin-mapping file will be needed (pinmap_re_arm.h), and the Marlin pin-mapping (pins_RAMPS_RE_ARM.h), the linker script should be fine as should the startup assembly but the System init function will need to detect it's running on an LPC1769 and set the clock appropriately for full support.

[1634 more replies]

closed 2019-06-21 by thinkyhead