MarlinFirmware/Marlin #7081: G28 Z-Homing Broken for Me with 1.1.x Release Branch

riverratdc, 2017-06-17:
My printer is a Prusa i3 clone utilizing MKS GEN 1.2 electronics (functional RAMPS 1.3 clone) with an inductive sensor for Z homing and Linear ABL. I've used this configuration for some time with great results running 1.1 RC7 & RC8. However, any code in the 1.1.x release branch (tried 1.1.0, 1.1.0-1, 1.1.1, 1,1,2 & 1.1.3) has a problem when running G28 to home all axes. Using the release code, the G28 appears to execute correctly, but upon completion, after the nozzle/probe raises to 10mm + my M851 offset, the LCD properly displays this Z height while Repetier Host 1.6.2 only shows the Z height to be the M851 offset without the 10mm safety raise. The LCD also displays the message "Endstops Z" at the end of the G28. After this, there is nothing I can do to synchronize the Z-height in Repetier to the actual Z-height properly displayed on the LCD.  As a result, I'm unable to manually move the nozzle below 10mm in Repetier using the 1.1.x release branch. I've spent a lot of time trying to isolate the problem without success. If I revert to RC8, all is once again well.

I enabled debugging for autoleveling in both RC8 and 1.1.2 and I've attached the debug output of a G28 command for each version, as well as a copy of my Configuration.h. My settings are always manually crossed-over between versions to avoid any variable name conflicts and I can't for the life of me see any configuration setting(s) that would lead to this behavior.

Any help would be greatly appreciated

Thanks,
Kevin

[1-1-1 G28 Debug.txt](https://github.com/MarlinFirmware/Marlin/files/1082450/1-1-1.G28.Debug.txt)
[1-1 RC8 G28 Debug.txt](https://github.com/MarlinFirmware/Marlin/files/1082451/1-1.RC8.G28.Debug.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1082455/Configuration.h.txt)

.

FHeilmann, 2017-06-19:
Try a fresh bugfix-1.1.x.

The "Endstops Z" display was something I also had back when there were some memory corruption issues, which were recently fixed.

riverratdc, 2017-06-19:
Thanks for the suggestion, Florian. Tried last night's 1.1.x bugfix (6/18) and the symptoms remain the same as described, including the "Endstops Z" message.

riverratdc, 2017-07-05:
Wanted to update this issue as I just finished confirming that it still exists in the just released Marlin 1.1.4.

Since I originally posted the issue, I've reconfigured my z-probe several times for test purposes. Since the MKS board has hardwired pull-up resistors on the endstop connectors that can't be disabled by configuration directive, I originally implemented my inductive sensor using pin D44 on the MKS board for the signal wire. This pin is unused by other hardware in my configuration and has no pull-up resistor. I set Z_MIN_PIN 44 in pins_ramps.h and have been using this configuration without trouble in RC7 & RC8. For test purposes, I've moved the probe signal wire to pin D4 and then D5 with no change in the original issue. These pins also work without problem in RC8.

One thing I have noticed since my original post, is that after homing all axes to put the printer into the problem state, if I subsequently home just the X or Y axes individually, the Z height in Repetier will jump to the correct value of 11mm and I can then manually lower the nozzle below 10mm. A final thing that I neglected to mention in my original post is that printer prints properly under host control (though it generates the "Endstops Z" message during the print job's G28 & G29 commands) using the 1.1.x release branch. The host's drip-fed gcode is able to lower the nozzle to the bed and print properly despite the initial disagreement in the Z height between the LCD and host.

After the work I've done, I'm now firmly convinced that this behavior is a bug introduced between RC8 and the 1.1.x release branch, somewhere in the probe/ABL code that was so extensively reworked between RC8 and the 1.1.x release branch.

RalphSch, 2017-07-25:
Is the behavior the same when disabling workspace offsets in Configutration_adv.h? I had/have similar problems if workspace offsets are activated.
I think it was no good idea to activate the workspace offsets by default because it leads most people into problems when upgrading from Marlin 1.0.x.

/**
 * Enable this option for a leaner build of Marlin that removes all
 * workspace offsets, simplifying coordinate transformations, leveling, etc.
 *
 *  - M206 and M428 are disabled.
 *  - G92 will revert to its behavior from Marlin 1.0.
 */
#define NO_WORKSPACE_OFFSETS

riverratdc, 2017-07-25:
NO_WORKSPACE_OFFSETS was disabled in my copy of 1.1.4. I enabled it for test purposes. It did result in a slightly smaller sketch (133K vs. 137K) but exhibits the originally described behavior. The state of NO_WORKSPACE_OFFSETS does not affect my problem. Interesting to note that NO_WORKSPACE_OFFSETS is not an available configuration setting in RC8, which I've once again reverted to. 

Thanks,
Kevin

riverratdc, 2018-01-01:
The "Endstops Z" message after homing on my printer was finally corrected with Marlin 1.1.7 or later.

closed 2018-01-01 by riverratdc