MarlinFirmware/Marlin #7082: Biliear correction not applied

Sungod3000, 2017-06-17:
Hi,

Im using Marlin 1.1.3 with bilinear autolvel on a cartesian machine with a fixed mounted probe. 
The probe triggers correctly and G29 finishes without issue, eeprom is actived and M500 confirms saved settings.
When i plot the G29 data I can clearly see that the bed shape is correctly captured.
Whats not happening however is any adjustment. The Z axis doesnt move at, so im not sure if I misconfigured anything especially because it worked in previous version on my other delta.

Any ideas?

Cheers

mperdue, 2017-06-17:
Have you enabled it with M420 S1 ? The leveling information is restored on boot up but G28 disables it.

Sungod3000, 2017-06-17:
I see, I had used M420 S1 before, but it came before the G28 in my start script. 
Seems to work now, thanks!

mperdue, 2017-06-17:
Even I can get it right on occasion... :-)

closed 2017-06-19 by thinkyhead